motioneye-project/motioneyeos #2882: Odroid-c1 refuses connection

NorthBaandee, 2022-02-09:
I am not able to reach port 80. Connection refused.
I am able to ssh into the system but the connection is closed after a few minutes. No logs can be found.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-odroidc1-20200606.img.xz

### Board Model

I am using the following board/model: odroid-c1

### Camera

I am using the following type of camera: simple mjpeg camera

My camera model is: Kodak dvc325

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 Monitor (no display can be seen)

### Log Files

I consider the following log files relevant to this issue:

 No logs appear to exist!
 
#

starbasessd, 2022-02-09:
Did you try 20200203? There isn't a version for dev20201026, and 20200606 has issues on most of the Pi Hardware.  When you are ssh'd in, look in the /var/log folder for motioneye.log, boot.log & messages. you could scp the files from another machine.

NorthBaandee, 2022-02-10:
I've tried 20200203 and also 20180627. On this last version I can see a few logs. From /data/log/boot.log I can see that the system is constantly rebooting because motioneye fails to start. There is no motioneye.log so obviously it bombs out before it can create that file. I can see that proftpd smbd and ntpd start correctly.

starbasessd, 2022-02-10:
I'm not able to help much beyond this, as I don't have an odroid to test with. Maybe someone else can help...

westminsterflip, 2022-03-07:
The issue seems to be with the read-only filesystem.  On motioneyeos-odroidc1-20200606 I changed the bootargs to read-write and it connects properly.  Before changing the boot.ini it would disconnect from my ssh session and reboot the board every few minutes, that has also been resolved.  I don't know the specific issue but it stops logs from being written to /var/log/ and changing the mount option has worked for 2 boards.