motioneye-project/motioneyeos #2883: throttlewatch CHECK_INTERVAL

kristofejro, 2022-02-14:
hello,

In relation to [https://github.com/ccrisan/motioneyeos/issues/2667](https://github.com/ccrisan/motioneyeos/issues/2667) I wanted to change CHECK_INTERVAL for throttlewatch.
But this prcedure does not work for me.

System:
motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

Does anyone know how to change it? Thanks.

starbasessd, 2022-02-14:
I just tested on a PiZeroW, motionEye 0.42.1, motion version 4.3.1, motionEyeOS dev20201026, and  [2667](https://github.com/ccrisan/motioneyeos/issues/2667#issuecomment-776161263) instructions failed to set 600 seconds. Interesting. Should work. Researching.

kristofejro, 2022-02-14:
[image]

It looks like this for me.

starbasessd, 2022-02-14:
OK, new instructions:
make a backup of your cameras before proceeding 
(Settings, General, Configuration Backup) 
### "Here Be Dragons!" 
You've been warned.
rm -rf /data/ext/init.d
mount -o remount,rw /
nano /etc/init.d/S11throttlewatch
Change line Check_Interval from 60 to whatever you want. 
\[ctrl\] x , y \[enter\] to save
reboot
When it comes back up, it should be back to 'normal state' (read-only) but with the time set to however long in seconds you chose above. I picked 300 seconds, and it waited 5 minutes for the first logging, and 5 minutes after that, consistently.
I don't know what broke, but fixed now.

kristofejro, 2022-02-14:
> rm -rf /data/ext/init.d

shouldn't it be etc instead of ext ?

starbasessd, 2022-02-14:
Oops. Corrected.

starbasessd, 2022-02-14:
Also, since you are putting such a long interval in, you may want to reset your logs prior to checking/changing your settings:
logrotate -f /etc/logrotate.conf
comes in handy to clear/reset the logs.

closed 2022-02-15 by kristofejro