motioneye-project/motioneyeos #2885: Raspberry pi zero 2 W working as a dashcam or remote CCTV without internet

ArshadNooring, 2022-02-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi3.img, 0.42.1
 

### Board Model

I am using the following board/model: Raspberry PI zero 2 W.

### Camera

I am using the following type of camera: Raspberry pi camera 8 MP.

My camera model is: Raspberry pi camera 8 MP.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi is working perfectly but it does not work when without internet.

### Peripherals

 
### Log Files

I consider the following log files relevant to this issue:

dnsmasq.conf
hostapd.conf
os.conf
static_ip.conf

I would like to use a MotionEye camera without internet in Raspberry Pi Zero 2 W. I have used a raspberry pi camera and the following software [https://github.com/.../motioneyeos/releases/tag/20220119-dev](https://github.com/jawsper/motioneyeos/releases/tag/20220119-dev?fbclid=IwAR0vtwx0A1_6ZpJNbPefwPY0rCBiCo4jUUaQTkg5bVFat0q18DqhAfGXSy4)
. I am able to run it with wireless, but I cannot record anything if I go out of the internet. I have followed the instruction given in [https://kevinshumaker.wixsite.com/.../motioneyeos-stand...](https://kevinshumaker.wixsite.com/thethirdlevel/post/motioneyeos-stand-alone-setup?fbclid=IwAR2Cdlm_NLSZvvH7wU31I-TBFon5dlkTuB-a_KEzf52NKwtUSmLTNZG8WIw)
, but it is not working. If someone has solved the issue, please let me know. Thank you.

starbasessd, 2022-02-20:
@jawsper  can you test your image with the [Access Point](https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point) setup instructions, please? I believe that is the solution he's looking for.

ArshadNooring, 2022-03-24:
@jawsper and @starbasessd, I would appreciate it if you could solve the issue. Thanks for your kind support.