motioneye-project/motioneyeos #2886: Dropbox uploads fail after a few hours: "code doesn't exist or has expired"

mikestefanello, 2022-02-20:
### motionEyeOS Version

I am running motionEyeOS version: [20200606](https://github.com/ccrisan/motioneyeos/releases/tag/20200606)

### Board Model

Raspberry PI 2

### Camera

Amcrest 1080p USB Webcam (https://www.amazon.com/gp/product/B08FCQY62D/)

### Network Connection

My motionEyeOS unit is connected to the network via WiFi

### Log Files

```
Exception: code doesn't exist or has expired
2022-02-20 16:00:32: [motioneye]    ERROR: dropbox: refreshing credentials failed
2022-02-20 16:00:32: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/2022-02-20/16-00-00.jpg" with service dropbox: code doesn't exist or has expired
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 726, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 782, in _request
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 827, in _request_credentials
Exception: code doesn't exist or has expired
2022-02-20 16:15:31: [motioneye]    ERROR: dropbox: refreshing credentials failed
2022-02-20 16:15:31: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/2022-02-20/16-15-00.jpg" with service dropbox: code doesn't exist or has expired
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 726, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 782, in _request
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 827, in _request_credentials
Exception: code doesn't exist or has expired
2022-02-20 16:27:07: [motioneye]  WARNING: 403 GET /login/?_=1645392427224&_username=_&_signature=f4bbab00c2f83ab1e265b376664353b5dc028f15 (192.168.68.109) 7.73ms
```
 
### Problem

I setup photos to be taken every 15 minutes. I enabled uploading to Dropbox. After authorizing and entering the code, the uploads work fine. After a few hours, they stop, and I see the logs pasted above. I've been able to reproduce this twice in a row (the only times I've tried).

starbasessd, 2022-02-20:
Please try dev20201026

mikestefanello, 2022-02-20:
@starbasessd Is there an easy way to upgrade or do I have to flash the card?

starbasessd, 2022-02-20:
### Here Be Dragons
Backup your current installation:
Settings, General, Backup
Then Settings, Expert, Enable Prereleases 
Apply
Then Settings, General, Software Update, and follow the prompts.
If it corrupts (it happens occasionally) then re-image, and do the restore.

mikestefanello, 2022-02-20:
Thanks. I completely missed the "Enable Prereleases" option.

I'll run that now, re-configure Dropbox, and let it export until tomorrow to see if it holds up. I'll report back. Thanks again for the help.

mikestefanello, 2022-02-21:
Same exact problem. It worked for about 4 hours.

starbasessd, 2022-02-21:
Are you on a free plan, or paid?

mikestefanello, 2022-02-21:
Free

starbasessd, 2022-02-21:
Are you saving movie files, or snapshots, or both? What is the total file size saved when it stops working?
I've never run into a limitation (I have a 'free' account, but I only do 1 camera for testing, and never exceed 50GB/day. 
I've run into more issues with GDrive (with number of files rather than size of files) but still prefer it over dropbox.

mikestefanello, 2022-02-21:
Only snapshots. The files are about 1.6mb in size each. I'm doing 1 every 15 minutes. The free plan has 2gb and I have nothing else in it. There's less than 100mb in total usage around the time it stops working (~4 hours). If it was a space issue, would the error still be `ERROR: dropbox: refreshing credentials failed`? I'll try gdrive and see if that works.

thenib, 2022-03-04:
Found a post on dropbox forums about this same 4 hour token expiry issue. 

"Dropbox is migrating to issuing only short-lived access tokens (and optional refresh tokens) instead of long-lived access tokens."

[More info about it here ](https://dropbox.tech/developers/migrating-app-permissions-and-access-tokens)

starbasessd, 2022-03-04:
@thenib 
Thank you very much for pointing this out. I will change this to 'enhancement'/'feature request' and we can hope when a dev takes over, it can be fixed. It should be able to be modded easier with motionEye, than motionEyeOS, as the system is more exposed to modification.

motzke, 2022-03-30:
I added a refresh_token method for dropbox:
https://github.com/motioneye-project/motioneye/pull/2395

byWulf, 2022-06-11:
Greetings! Thanks for the fixing pull request for this issue. I'm facing it currently too.

Is there an easy way of updating motioneye to the dev version under motioneyeOS? Or what would you recommend me to get this fix up and running? 

When selecting "Enable PreReleases" and clicking on "Software Update: Check", I get the following error: "An error occurred. Refreshing is recommended."

Thanks for you help!

starbasessd, 2022-06-11:
https://github.com/motioneye-project/motioneyeos/issues/2886#issuecomment-1046332845

byWulf, 2022-06-11:
Thanks for you quick answer @starbasessd . I have seen your post and tried it out, but as I wrote, this didn't work for me because of the error message "An error occurred. Refreshing is recommended." when checking for an update.

starbasessd, 2022-06-11:
Part 1 was backup, then run the check for updates.
Part 2 was was re-image, then do the restore.

starbasessd, 2022-06-11:
The image is available [here](https://github.com/motioneye-project/motioneyeos/releases)
Be aware, though, the fix from https://github.com/motioneye-project/motioneyeos/issues/2886#issuecomment-1082692289 is not easily applied to motionEyeOS, only motionEye.

dba83, 2022-11-19:
Hi,
Same problem with me ( i have motioneyeOS dev20201026)  , i don't know if i a paid account will solve the issue ?