motioneye-project/motioneyeos #2887: Raspberry Pi Zero W 2 won't turn on with motion eye

TimmieF, 2022-02-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20200606).

### Board Model

I am using the following board/model: (Raspberry Pi Zero W 2.

### Camera

I am using the following type of camera: (Raspberry Pi Camera Rev 1.3).

### Network Connection

My motionEyeOS unit is connected to the network via: (thru WiFi using the wpa_supplicant file.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

32GB SanDisk Extreme

### Log Files

I consider the following log files relevant to this issue:

 I have no log files, when I boot the Pi up with just raspian OS it works fine, but with motioneye, It won't even turn on. The Pi is just 1 day old and using a other sd card doesn't work either. I'm not sure why it's not working with motion eye, I have a raspberry pi 3 B that works just fine with motioneye.

starbasessd, 2022-02-26:
You can't use a normal release on a PiZero2W, you have to use the image discussed [here](https://github.com/ccrisan/motioneyeos/issues/2875)

TimmieF, 2022-02-26:
Okay ill try that! Same setup as the regular Pi's right? And using wpa_supplicant for wifi

starbasessd, 2022-02-26:
Correct

mlacunza, 2022-04-30:
Confirmed, this version works, I tried with the official version and the PiZero2W don't even start.  Thanks!

I-iunnius, 2022-09-01:
> I can confirm the above-linked "not official" release is working for the Zero W2.

Thank you for your information.
Can you tell us the exact text of your wpa_supplicant?

starbasessd, 2022-09-01:
The format of the wpa_supplicant.conf file needs to be:
- - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="your_SSID"
    psk="your_password"
}
- - - - -
Country needs to be the country code for the country you are in
ssid and psk should be in double quotes: "
ssid and psk are case sensitive
be sure that if you use Windows notepad to create or edit the file, that the filename doesn't have a hidden .txt at the end.
Windows often creates the filename as wpa_supplicant.conf.txt

I-iunnius, 2022-09-01:
Thank you, starbasessd.
I have tried a lot of 'wpa_supplicant.conf', with different combinations.
Latest yours.
Result:
-
  Starting wpa_supplicant: failed (brcmfmac)
  Panic. Rebooting in 30s (caused by network)
-
So, I dont know what to do.
Image 20220119 written via Etcher, and also via Raspberry imager.exe
Pi Zero 2W
Sandisk ExtremePro 64 Gb XCI U3 A2 V30
Thank you again

starbasessd, 2022-09-01:
I don't recommend either of those imaging programs for motionEyeOS-types, they make changes in the background. I use Win32DiskImager, in Windows, and dd in linux.

starbasessd, 2022-09-01:
Another thing you can try, just to get it to boot, if you have a keyboard & monitor connected is to add a file called 
[os.conf](https://github.com/ccrisan/thingos/wiki/os.conf)  
and have the line: OS_NETWORKLESS=true
This will allow the PiZero2W (or any pi without network connectivity) to boot up.
You can then go through some diagnostics to see what the problem is.
BTW, can you confirm your PiZero2W version (printed in white on the back) and that the front looks like the pic [here](https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/)

I-iunnius, 2022-09-02:
Thank you starbasessd. You are very kind

> Another thing you can try, just to get it to boot, if you have a keyboard & monitor connected is to add a file called [os.conf](https://github.com/ccrisan/thingos/wiki/os.conf) and have the line: OS_NETWORKLESS=true This will allow the PiZero2W (or any pi without network connectivity) to boot up. You can then go through some diagnostics to see what the problem is. BTW, can you confirm your PiZero2W version (printed in white on the back) and that the front looks like the pic [here](https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/) Scammers out there sell PiZeros (no wireless even) as PiZeroWs and PiZero2Ws...

Thank you starbasessd. You are very kind.

wpa_supplicant.conf, edited with Notepad++ in mode "Unix" says:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev country=es update_config=1
network{scan_ssid=1 ssid="xxxxxxx" psk="xxxxxxxx" key_mgmt=WPA-PSK}

(As you see, both in one line, with no indent; and the country in lower case)

os.conf: done
Boot; but now:
- no mention to wpa_supplicant. (in the router app, no mention to any raspberry device connected)
- message:
- Welcome to meye-ba7d4e57!
  If you want to see a friendly user interface, print your web browser to the IP address of this board.
  meye-ba7d4e57! login:

Next, I'll try with Win32DiskImager, and another wpa_suppl..., 

- I have bought 2 zero's 2W, both at Kubi in France:
The one tested now says in the front:
   RP3A0 AU
       2104
    201207A

- Rear:
  Raspberry PI: Zero 2  FCC ID  2ABCB-RPIZ2
  Raspberry Pi 2021 IC: 20953-RPIZ2

The other (not tested yet) says the same, in both sides.

False?

Thank you.

I-iunnius, 2022-09-02:
Win32DiskImager:

It can only read *.img, but cannot see motioneyeos-raspberrypi3.img.gz.
If I select a *.*, it write something; and then, when I try to open it (to write wpa_supplicant), it ask me to format the unit.

I-iunnius, 2022-09-02:
More info:

Tested with bullseye, It has wifi.

starbasessd, 2022-09-02:
You can unzip the image using 7zip ( either the [full install](https://www.7-zip.org/a/7z2201-x64.exe) or the [portable apps version](https://portableapps.com/downloading/?a=7-ZipPortable&s=s&p=&d=pa&f=7-ZipPortable_22.01.paf.exe] then write the image file using Win32DiskImager.
I tested the wpa_supplicant.conf file format you used, and it did not work for me, using my info. I tried 
- - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev 
update_config=1
country=ES 
network={
ssid="xxxxxxx" 
psk="xxxxxxxx" 
}
- - - - -
using my appropriate info, and it works. country in caps, network={ ('=' was missing) and scan and WPA lines left out.

I-iunnius, 2022-09-02:
In Bullseye, wpa_supplicant, country appears in  uppercase

starbasessd, 2022-09-02:
Your posted info:
- - - - -
wpa_supplicant.conf, edited with Notepad++ in mode "Unix" says:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev country=es update_config=1
network{scan_ssid=1 ssid="xxxxxxx" psk="xxxxxxxx" key_mgmt=WPA-PSK}

(As you see, both in one line, with no indent; and the country in lower case)
- - - - -
you need to separate the lines as I just showed, put country in uppercase as I showed...

I-iunnius, 2022-09-03:
YES, it works. Thank you again.
The secret: Win32DiskImager (and the correct wpa). Result:
....
Starting wpa_supplicant: done (brmfmac)
.....
Starting motioneye: done
wlan0 IP ...
....
Welcome to meye-ba7d457!

If you want to see a friendly user interface, point your web browser to theIP address of this board.
meye-ba7d4e57 login:

And now I dont know what to write.
I have tried a lot of combinations I have read in other post.
But nothing.
If Login = admin + Pass = blanck then [root@meye-ba7d4e57 ']# . and stops there.

If I try via sshfs, I cannot going on.
 
I think I'm very close to the end. 

Instructions?

I-iunnius, 2022-09-03:
YES.

In google I simply write the IP, and started with admin + nothing, and motioneye is there.

Thank you starbasessd for your effort and kindly attention.

Best Regards.

starbasessd, 2022-09-03:
Glad to help.

I-iunnius, 2022-09-03:
Sorry, a question.

I have a sd 64Gb card. Is it necessary to expand filesystem to have the maximum capability to save pictures/videos?.
If yes, how?

starbasessd, 2022-09-03:
It should auto-expand the filesystem on first boot. 3 partitions, 
/boot about 30MB
/ about 213MB
/data the rest of the drive

I-iunnius, 2022-09-04:
Closed.
Thankyou starbasessd, very very much.
People as you makes things easiest.

Fenestrane21, 2024-06-16:
> It should auto-expand the filesystem on first boot. 3 partitions, /boot about 30MB / about 213MB /data the rest of the drive

Hi starbasessd, i need your help, seen you doing a lot of help to people having problems with motioneye,

I am also using a raspberry pi zero 2 w with a camera rev1.3, and jawspers version of motioneye.
When i use raspbian, it says supported=1, detected=1 but the interface on the web of motioneyeos just gives a black screen with a no camera logo.

The bxw whatever driver it is for the camera is also installed.

When i had first started it a few months ago, it was having the same problem but it got solved automatically. Everything was working great with motion detection, video capture etc. But i had kept it aside around 10 days while i was getting the power supply ready. When i again started it, it stopped working and has stopped working ever since been trying everything but just does not work.