motioneye-project/motioneyeos #2888: UI doesn't start anymore with motioneye: failed

dmangonakis, 2022-03-01:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Network Connection

My motionEyeOS unit is connected to the network via: communication via WiFi (Fritz Box / offline)

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/8161050/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/8161052/dmesg.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/8161054/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/8161055/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/8161062/messages.log)
[wpa_supplicant.log](https://github.com/ccrisan/motioneyeos/files/8161118/wpa_supplicant.log)

**Context:**

Everything was working fine until we had a power outage. Then the reboot problem occured very similarly to https://github.com/ccrisan/motioneyeos/issues/2877. I followed all the steps, fixed the rebooting problem but wasn't able to restart motion eye even via /etc/init.d/S85motioneye failed. 

Thank you guys so much for motioneyeos and your support in advance!

starbasessd, 2022-03-01:
The ultimate solution to #2877 was a clean image.
Your dmesg log file shows /boot being corrupted.
Your motionEye log shows /data/media/sda1 not being mounted
>2022-02-06 20:37:41: [motioneye]    ERROR: could not find mounted partition for device "/dev/sda1" and target dir "/data/media/sda1"
>2022-02-06 20:37:41: [motioneye]    ERROR: could not find mounted partition for device "/dev/sda1" and target dir "/data/media/sda1"
>2022-02-06 20:45:05: [motioneye]    ERROR: failed to create root directory "/Users/CU/Videos/Rasberrypi Videos": [Errno 30] Read-only file system: '/Users'
>Traceback (most recent call last):

So it looks like lots of file systems got corrupted during the power off/reboot that caused your original problems. I would suggest starting over. 
You don't say what board you are using. If it's a Pi3B+ or 4B, you could install to a USB drive instead of the SDCard, and make a change to the /boot/cmdline.txt file, and boot from it, with dev20201026.

dmangonakis, 2022-03-01:
Thank you so much for your quick feedback!

I work with multiple boards but I'm glad to hear that this one is definitely corrupted. FYI - the above logs are somehow old, the power off happened just last night.

1. Regarding your first approach, did you mean to format the SDCard and simply write a new OS Image on SDCard?
2. Regarding your second approach, did you mean to write new OS Image on USB drive and adjust the /boot/cmdline.txt file on the SDCard (and then boot)?

starbasessd, 2022-03-01:
Correct on both.
The change in the cmdline.txt file:
> change 
>root=/dev/mmcblk02 
>to 
>root=/dev/sda2

Do not add any spaces, make sure it is all still on the same line. This allows you to boot a USB Drive ONLY using dev20201026

There is an issue with logrotate where it doesn't seem to run on its own regularly. To work around this you can run the command:
>logrotate /etc/logrotate.conf

to force it.