motioneye-project/motioneyeos #2892: Raspberry Pi 4B - 2gb not loading

srgbriggs, 2022-03-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

YES

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

YES

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

20201026

### Board Model

I am using the following board/model: Raspberry Pi 4B

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: PI HQ Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None. Unplugged everything but camera and monitor

### Log Files

I consider the following log files relevant to this issue:

idk
 
I am having the same issue as https://github.com/motioneye-project/motioneyeos/issues/2419 with the new 1/25/2022 bootloader. It's getting stuck in the "This board requires newer software" for start4.efl loop.

I was able to load Rasbian on it and I just had Octoprint on it so I don't think its the card or the pi this time.

Any ideas? I can follow instructions but this is out of my wheelhouse now. Thanks y'all!

venport1, 2022-04-03:
I have the same issue:
Raspberry Pi 4B+ 8GB
Raspberry Pi 400

with two different cards:
uSD SanDisk Extreme PRO 64GB U3 A2 V30
uSD Samsung EVO Plus 128GB U3 A2 V30

when using motioneyeos-raspberrypi4-20200606.img.xz
[image]
[image]

starbasessd, 2022-04-03:
You HAVE to use the dev20201026 for Pi4, and you need a 2021 boot loader. The pre-2021 and 2022 boot loaders have issues. There is a new Dev team, but they don't have a priority for motionEyeOS updating.

venport1, 2022-04-03:
Thank you very much for your reply.

Any idea where I can find the 2021 boot loader you mentioned?

starbasessd, 2022-04-03:
The older bootloaders can be found [here](https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/old)
Depending on how far you want to go back, they are also found in the folder tree:
/usr/lib/firmware/raspberrypi/bootloader/ in the stable, beta, and critical folders

srgbriggs, 2022-04-03:
Awesome! Thanks so much! I'll try this tonight and report back.

venport1, 2022-04-03:
Thank you so very much.
I took an screenshot and was about to ask you if the place where I took it is the right place.

starbasessd, 2022-04-03:
No screenshot got attached to the github message.
My bootloader is:
[root@meye-57b21018 ~]# vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
and also allows me to boot to USB using dev20201026 with a modded cmdline.txt (change root=/dev/mmcblk0p2 to root=/dev/sda2)

venport1, 2022-04-04:
I just updated the post and the commands that have been used in
[image]
 A Single Raspberry Pi 4 4GB Won't Boot dev20201026 #2772 (https://github.com/motioneye-project/motioneyeos/issues/2772#)
regarding this issue gave me:

starbasessd, 2022-04-04:
The bootloader is too new. Try rolling back to 4/29/2021. 
You aren't under-volting or over-heating, but then again, that wasn't considered in your issues...

srgbriggs, 2022-04-04:
I cant seem to get the old bootloader to flash over. Is it b/c the 1/25 release was critical? https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/critical

Am I able to flash backwards to 4/29/21 again? If so, is am I missing something other than copying the .bin file to a card and putting it in the pi?

starbasessd, 2022-04-04:
In RPiOS the command to update to a specific version of the bootloader is:
sudo rpi-eeprom-update -f <path-to->/pieeprom-yyyy-mm-dd.bin
In the case of my Pi4s, the command and path would be:
sudo rpi-eeprom-update -f /usr/lib/firmware/raspberrypi/bootloader/critical/pieeprom-2021-04-29.bin
This will cause the PiOS force-reload the 04/29/2021 bootloader, and prompt for a reboot. Let it come all the way back up, check for the correct bootloader, then shut down and insert the motionEyeOS card

venport1, 2022-04-04:
Hi starbasessd,

Thank you very much for your assistance.
I was able to see all the binaries in  /usr/lib/firmware/raspberrypi/bootloader/critical/
"...Let it come all the way back up...": how long can this take? I followed the command you posted, it asked for a reboot, rebooted and I've been waiting for over 10min.

venport1, 2022-04-05:
Never mind!

Shut it down, started it back again and:
[image]
[image]

venport1, 2022-04-05:
Thank you so very much!!!
[image]

Totalausfall, 2022-04-09:
Hey guys, 

i don´t know what i´m doing wrong...

04/29/2021 bootloader is installed and i use the dev20201026 img.

did i miss something? :<

[image]

[image]

starbasessd, 2022-04-09:
Did you make any other changes / modifications in the /boot/ partition?
I just fired up my Pi4-4GB and Pi4-8GB and both are working with dev20201026 (I don't SDCard the 8GB, though, due to the case I use, I use an USB Boot Drive. Both have the 04-21 boot loader.

Totalausfall, 2022-04-09:
no, nothing

-download motioneyeos-raspberrypi4-dev20201026.img.xz
-unzip it
-flash the img with Pi Imager (without errors)
-put it in the raspi 
-no success :(

starbasessd, 2022-04-09:
The USB drive requires a change to cmdline.txt
I use Win32DiskImager, personally. 
I don't know. Hopefully someone with a 2GB can try and help. 
I can relate to your last comment, me and gcc, RegEx, and SQL go round and round a lot...

Totalausfall, 2022-04-09:
okay...

i bought the pi4 with 2gb today and only changed the bootloader.
just for fun i`ve put the sd card with MotionEyeOS in my Nextcloud-pi and it and it works immediately. there is the 09-20 bootloader installed.

i try the same with my pihole and if it works, i change the 2gb pi with the 4gb pihole. the pihole doesn't need that much ram...

Thank you for answering anyway :)

and sorry for my english. German suits me better :D

starbasessd, 2022-04-09:
Do you remember the date of the original bootloader on the 2GB? Was it a 2022 date? If so, it was mentioned in another issue where they updated to January's, it didn't work, and then tried to roll it back, and it still didn't work. I really hope the new devs get it together and start working on motionEyeOS to enable the Pi4s and PiZero2Ws...
Mein Deutsch ist nicht sehr gut, das Abitur war vor 50 Jahren...

Totalausfall, 2022-04-09:
The original bootloader of the 2gb pi was from 2019 but i can't remember exactly which version. Out of desperation I tried different ones but none worked.

Now MotionEyeOS is running on the PiHole-Pi. at first attempt. it has the 04-21 bootloader. Fortunately, PiHole runs immediately on the 2gb pi.

Very strange but ok...

Your German is definitely better than my English :D

Schönen Abend noch und vielen dank :)

starbasessd, 2022-04-09:
Mein alter Gymnasiallehrer für Deutsch und Google Translate :) Du willst deine Mutter nicht versehentlich einen Schuh nennen... :)

benjohnson2001, 2022-06-18:
@Totalausfall I had the same problem, I ended up downloading the new start4.elf and fixup4.dat from https://github.com/raspberrypi/firmware/tree/master/boot and then it finally worked

ellow7, 2022-07-01:
> @Totalausfall I had the same problem, I ended up downloading the new start4.elf and fixup4.dat from https://github.com/raspberrypi/firmware/tree/master/boot and then it finally worked
> 
> EDIT: Well it finally got past the start.elf error but then the camera didn't work. I needed to add start_x=1 to the config file and then copied over start4.elf, start4.elf, start4x.elf, start_x.elf, fixup.dat, fixup4.dat, fixup4x.dat, fixup_x.dat from the repo to the boot folder and then everything worked (using the most recent 2022 bootloader)

This worked for me. I did not need to edit anything to get my network camera working.

crashedrobot, 2022-08-17:
> @Totalausfall I had the same problem, I ended up downloading the new start4.elf and fixup4.dat from https://github.com/raspberrypi/firmware/tree/master/boot and then it finally worked
>
> EDIT: Well it [...]

Thanks! This worked for me as well - RaspberryPi4 2GB. I ended up trying some of the other recommendations before seeing this, so ended up having the 4/29/2021 bootloader instead of the newest one.  Just for clarity, these are the files copied to the boot folder on the SD card (Ben has start4.elf listed twice instead of including start.elf):

start.elf
start4.elf
start_x.elf
start4x.elf
fixup.dat
fixup4.dat
fixup_x.dat
fixup4x.dat

and added "start_x=1" to the config.txt file.

Nicolas1203, 2022-08-24:
Having to work without a monitor it was hard to see the errors on boot.

I tried the solutions above but none worked for me.

My setup :
- raspberry pi 4 2gb
- InnoMaker Pi-Camera-OV5647

I ended up making things work doing the following. I'm posting it here, sorry if this is not the right place.

1. Install Raspberry Pi Os Lite (Legacy) using Raspberry Pi imager.
2. Install motioneye following [this tutorial](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian)
3. Enjoy motioneye on http://[your_ip]:8765/

Notes : 
- You can in theory install any linux distro and use the corresponding tutorial from [here](https://github.com/motioneye-project/motioneye/wiki/Installation) but I found the distro based on Raspbian 11 Bullseye or later did not work with my camera, so I would advice keeping to Raspbian 10 Legacy or Earlier and using the corresponding tutorial.
- I had a hard time getting my camera working with v4l2. I believe that if after flashing the board, the command ```vcgencmd get_camera``` says ```supported=0``` you should try flashing the board with another distro. At least this is what worked for me.

Hope this helps.

FootsoreBird03, 2023-06-12:
If anyone on this post is still active, could someone link me to a tutorial or a youtube video on how to make the changes to get it working? Also, if I need to have the standard Raspi os on it, how do I overwrite the current os instal on the SD card? I ask since I only have one card. 

My original post with my problems: https://forums.raspberrypi.com/viewtopic.php?p=2113013#p2113013