motioneye-project/motioneyeos #2893: 3 Cams and then out of memory?

lmarceg, 2022-03-25:
Hi,
I have been using MEOS since August 2019 and it's always worked fine. I am using motionEyeOS dev20190820
I have a Raspberry 3+ with 1GB RAM and three cams (low-res for motion detection and high-res for video recording in case of motion)
Four days ago I bought a new camera and I added it to MEOS, but I saw missing alerts in case of motion.
It's interesting to see that I see the red square in the GUI when I move and I also get the videos and images stored in the sdcard; however, the pushover notifications doesn't work. If I remove one of the four cameras and go back to three, it all works fine again.

I saw that with three cameras my memory usage is about 590MB over 747MB
With 4 cameras I use around 740 MB over 747MB, and I am left with just 7 MB, and I think this is the issue, since I didn't change anything else.

Does anyone of you have more than 3 cameras on a Rpi3+?
I am using gpu_mem=256 in config.txt (don't remember why, maybe it's the default, maybe not), does this change anything?

What would be the options here? Use a Rpi4 with 2GB RAM?
Thanks!

starbasessd, 2022-03-25:
I can run 5 network cameras on Pi3B+, using mEOS dev20201026, used to do so regularly testing for user issues.

lmarceg, 2022-03-25:
I was thinking I have 747MB available because 256MB are taken up by gpu_mem. Should I lower the gpu_mem? What do you have oin your config.txt? I see that more or less, every camera takes 200MB, so 5 cameras would make sense

starbasessd, 2022-03-25:
I have mine set to the 128M default, as I don't run cameras at a higher resolution than 1280x720. GPU mem only needs more if you are running higher resolution cameras at higher resolution.  And no, each camera doesn't take 200M, if it did, a hard limit would be 5. I have had up to 7 on the Pi3B+ (extreme test) and IIRC others have had 10. It becomes a matter of how much simultaneous motion on how many cameras, whether you are doing much processing, do Pass Through recording, if you are using OMX formats, etc.

lmarceg, 2022-03-25:
Mmmm
Simultaneous motion basically happens on one camera at a time since I don't have overlapping parts of video.
The configuration already includes pass-thru and OMX

```
movie_codec mp4:h264_omx
movie_passthrough on

```

The fourth camera doesn't work very well during the day, so I just configured it with motion off (next step was to turn motion on just during the night)

I will lower memory to 128GB to see what happens, I don't know if lowering GPU memory increases the rest of the memory (747 + 256 = 1003 but I don't know if it is just a coincidence). But in any case, I wonder how I could reach 7 cameras this way...

starbasessd, 2022-03-25:
Yes, lowing one increases the other, memory-wise, as there is only 1GB shared between CPU and GPU. 
You can easily script pausing motion detection per camera, using simple cron jobs. Before doing anything else, I would update your mEOS to dev20201026 before doing anything else. Go into Settings, General, and do a backup of your current settings, and then re-image or try updating through Settings, Expert, Enable Prereleases, Click Apply then go up to Settings, General, Software Updates, Check. Be SURE to do the backup FIRST. After the update, if you re-image, do the Restore.

lmarceg, 2022-03-25:
OK, thanks.
In the meantime I have reduced memory from 256 to 128 and now I have 4 cameras and it's all working fine! I'll consider upgrading as well.

closed 2022-03-28 by lmarceg