motioneye-project/motioneyeos #2894: Camera stopped working Pi Zero

ShosMeister, 2022-03-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 
motionEye Version | 0.42.1
-- | --
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

### Board Model

I am using the following board/model: Pi Zero.

### Camera

I am using the following type of camera: MMAL.

My camera model is: RPI-CAM-V2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Log Files

I consider the following log files relevant to this issue:
[MEye-02_dmesg.log](https://github.com/motioneye-project/motioneyeos/files/8357945/MEye-02_dmesg.log)

ShosMeister, 2022-03-27:
This setup worked last year - use them for bird nest watching. Got them out this year and the hub (rPi-3) and my other Pi Zero (exactly the same setup) are working fine. This one does not. The camera does not show up:
[image]
[root@meye-645baf70 ~]# /usr/bin/vcgencmd get_camera
supported=1 detected=0
[root@meye-645baf70 ~]#

[root@meye-645baf70 ~]# mmal_vc_diag components
4 created, 4 destroyed (0 destroying), 2 create failures
ril.isp                         : destroyed: pid -1072865080 address 0x1f3bd014 pool mem alloc size 0
[root@meye-645baf70 ~]#

During boot, the red light on the camera module does flash for about 1/2 second and is then off.

I've checked the cable connections and they are the correct way - I believe evident by the camera module light flashing during boot.

I've reimaged the sd card - twice - and have not even restored my previous config. Just setup wpa file for WiFi and tried to add the camera from the web UI.

starbasessd, 2022-03-27:
Install RPiOS, enable the camera in raspi-config, and test the camera in an app.

ShosMeister, 2022-03-30:
I've put RPiOS on the Pi Zero and when it boots up, the led on the camera module comes on. I don't have it hooked up to a monitor so I'm just using the status lights.

If you know a simple APP that I can use to "test" other than just looking that the light is on, could you provide that? Also, if I need to actually get to the desktop, I'll need to do some additional work to get it hooked up to my TV or something and get a keyboard.

starbasessd, 2022-03-30:
If using Legacy (Buster) RPiOS, you can use Raspivid or raspistill to capture a pic or movie, then copy it to another computer to look at it. If you re using the Bullseye check out
https://www.arducam.com/docs/cameras-for-raspberry-pi/raspberry-pi-libcamera-guide/

qkum, 2022-04-02:
My Raspberry Pi Zero W gave me the same error when setting it up today.

The problem was that the cable was plugged in the wrong way.

I'm using version: 
```
motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606
```

ShosMeister, 2022-04-04:
Got side-tracked for a bit but managed to get buster loaded up and while I don't have a way to get to the desktop - no keyboard/mouse/USB on Pi Zero - I can SSH in. Updated it and then enable the camera via raspi-config. Tried raspistill and it failed but notice the dmesg log showing it trying to load the but warning something about the mmal module. Don't know what that is.

pi@raspberrypi:~ $ sudo raspistill -o /tmp/test.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

pi@raspberrypi:~ $ sudo dmesg | egrep --ignore-case "mmal|vc|bcm"
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:5B:AF:70 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=77b6c82a-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000182] bcm2835: system timer (irq = 27)
[    0.131408] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.224551] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[    2.300999] bcm2708_fb soc:fb: FB found 1 display(s)
[    2.313933] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 656x416
[    2.323239] bcm2835-rng 20104000.rng: hwrng registered
[    2.323856] vc-mem: phys_addr:0x00000000 mem_base=0x1ec00000 mem_size:0x20000000(512 MiB)
[    2.325573] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x20200000
[    3.310430] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[    3.314547] mmc-bcm2835 20300000.mmcnr: could not get clk, deferring probe
[    3.315581] sdhost-bcm2835 20202000.mmc: could not get clk, deferring probe
[    3.349390] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    3.352210] mmc-bcm2835 20300000.mmcnr: mmc_debug:0 mmc_debug2:0
[    3.352248] mmc-bcm2835 20300000.mmcnr: DMA channel allocated
[    3.431173] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[   14.946988] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[   14.983606] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   14.983653] [vc_sm_connected_init]: start
[   15.000605] [vc_sm_connected_init]: installed successfully
[   15.044962] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   15.047499] bcm2835_audio bcm2835_audio: there is not valid maps for state default
[   15.135304] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   15.143419] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   15.189050] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[   15.191546] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   15.296371] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   15.884078] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   15.884184] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   15.884890] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
[   15.885905] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
[   15.886739] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
[   15.887464] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
[   15.887530] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[   15.887573] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[   15.887614] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[   15.887648] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[   15.903233] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[   15.904025] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   15.904114] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   16.042783] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   16.042874] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   16.113812] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[   16.113912] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[   19.572391] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   19.921919] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   19.922186] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   19.931076] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
pi@raspberrypi:~ $

ShosMeister, 2022-04-04:
pi@raspberrypi:~ $ vcgencmd get_camera
supported=1 detected=0
pi@raspberrypi:~ $

ShosMeister, 2022-04-04:
Funny thing is that if I disable the camera in raspi-config then reboot, the red light on the camera module is on. vcgencmd returns 0 for both supported and detected.

Enabling the camera and rebooting, the red light is off and vcgencmd returns as above.

Bad camera?

starbasessd, 2022-04-05:
> mmal: Camera is not detected. Please check carefully the camera module is installed correctly
> pi@raspberrypi:~ $ vcgencmd get_camera
> supported=1 detected=0

Either the cable is bad or not facing the correct way at both ends, or the camera is bad.
If it came with 2 lengths of cable, try the other, but I'm betting a bad camera.

ShosMeister, 2022-04-06:
I'm certain the cable is on the right way because if I shutdown, and swap the cable on either end, after reboot, the red led on the camera module no longer is lit. Also, if I enable the camera in raspi-config, it no longer shows up as supported=1.

BUT! I did figure out what it was - yes it's fixed. The Pi Zero camera has a really small board connection on the side where the camera is. When it's mounted into the Pi Zero case (the one for the camera), you can't see it. I took it out to triple check the cables and noticed that it had come loose. It just snaps into the board.
[image]

Now I just need to put motioneye OS back on there and see if it's working. Will update later.

ShosMeister, 2022-04-06:
We are working. So anyone using a Pi Zero camera like this, don't forget about this connection.

Thanks!

closed 2022-04-06 by ShosMeister