motioneye-project/motioneyeos #2895: Is this still being worked on

Steeleshh, 2022-03-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (motionEyeOS dev20201026).

### Board Model

I am using the following board/model: (Raspberry pi 4).

###question

I was wondering if this project was still being worked on or is it not?

starbasessd, 2022-03-28:
Not currently, there is a new group of developers working on motionEye, but not motionEyeOS.

starbasessd, 2022-03-28:
See: https://github.com/ccrisan/motioneyeos#readme

Steeleshh, 2022-03-30:
Thank you, we currently use your program for our business so were just wondering

closed 2022-04-01 by Steeleshh