motioneye-project/motioneyeos #2897: Upload to Google Drive's shared drives

andreluisos, 2022-04-01:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: ccrisan/motioneye:master-amd64

### Board Model

I am using the following board/model: intel i5

### Camera

I am using the following type of camera: Network

My camera model is: custom

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

Is it possible to upload the movies directly to shared drives in Google Drive?

starbasessd, 2022-04-01:
Settings, File Storage, Upload Media Files, Select for Upload Media Service GDrive. Follow ToolTips to set up.

andreluisos, 2022-04-01:
But how could I set the path to the shared drive? I tried the folder id directly and /shared-drives/folders/FOLDERID. However, motioneye keeps considering that a path in "My Drive", not "Shared Drives"

andreluisos, 2022-04-01:
> Settings, File Storage, Upload Media Files, Select for Upload Media Service GDrive. Follow ToolTips to set up.

But how could I set the path to the shared drive? I tried the folder id directly and /shared-drives/folders/FOLDERID. However, motioneye keeps considering that a path in "My Drive", not "Shared Drives"

starbasessd, 2022-04-01:
If it is someone else's drive being shred with you, I don't think you can. 
I just created a shared folder in my GDrive, shared it to another account, turned on File Uploads to GDrive, used /shred (path I created) and obtained key and applied it, and it works. 
Other account is able to see and access the files. 
I cannot access someone else's shared to me folder that way, though. I have to be the owner.

starbasessd, 2022-04-01:
BTW, that would be a security issue, where you don't control who has access to the videos or pics being posted. If the cameras were in a 'questionable' location (bedroom, bathroom, kids area, etc) or even entry areas where people could see you coming and going, knowing you are not at home, etc., and being monitored by someone who hacked that drive outside of your control, would that be a good thing? That is why people are shifting back out of 'cloud' storage that they don't own or control.

andreluisos, 2022-04-01:
Maybe you're confusing Shared Drives with Shared with me. They are two different things.

starbasessd, 2022-04-01:
Shared Drives are like the Shared With Me in that you don't own them.
"Shared drive files belong to the team instead of an individual."
The resulting issue is still the same. 
Mapping is only allowed in their app.
https://support.google.com/a/answer/7212025?hl=en#zippy=%252Ccompare-shared-drives-with-my-drive

andreluisos, 2022-04-03:
Only the owner of the shared drive has access to it. And also the ones the owner allows.
But the question is: there isn't a way to upload media to a shared drive, correct? Could we make this a feature request?