motioneye-project/motioneyeos #2899: Working schedule without motion detection

Sheep225, 2022-04-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: (Raspberry Pi Zero).

### Camera

I am using the following type of camera: MMAL.

My camera model is: original Pi Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: none

 I use a Pi Zero with the original Pi camera module as a timelapse cam on a construction site to document construction progress. The camera is set to take a picture every 60 seconds.
Working schedule of the site is Mon-Fri from 7-5, so the camera does not need to take pictures over night or on weekends.
Unfortunately, the working schedule settings on motioneye apply only to motion detection but not as a general operating schedule.
With my setup, I am generating 100rds of meaningless pictures which I have to delete manually.

Is there a way to force the general operating time, similar to the motion detection working schedule? (e.g. per any system settings/configuration file)

starbasessd, 2022-04-05:
I wrote a set of scripts a while back to do this. It involves 2 config files and a cron job to automatically switch between the 2 sets. You can try searching for it here or in the motionEye github (I've done it for both) or this weekend I can find and post again.

Sheep225, 2022-04-05:
Would be great if you could guide me to the right posts... can wait til weekend!

starbasessd, 2022-04-05:
https://github.com/motioneye-project/motioneyeos/issues/2779#issuecomment-882773284
is one link. Change times as needed, etc...

Sheep225, 2022-04-05:
As far I understand the camera_X.conf file contains all the settings for the camera, also if the camera is set to manual capture or motion detection?
So I will create a "daytime" conf file with automatic capture and a "nighttime" conf file with manual capture (which means no images because no manual trigger). Is that correct?

Sheep225, 2022-04-28:
I have now tried to get this thing working, but I failed.
Both camera-1.conf.primary and camera-1.conf.secondary files are present, also the daytime and nighttime scipts and the crontab entrys, but nothing happens.
Do I need to "switch on" the cron somewhere?

starbasessd, 2022-04-28:
Yes, you need to edit crontab and add the lines to switch to the appropriate conf files at the appropriate times, as listed above, and in the [comment](https://github.com/motioneye-project/motioneyeos/issues/2779#issuecomment-882773284)
Look for the part that says 'edit crontab'

Sheep225, 2022-04-28:
I have added the lines to the crontab, but nothing happens.
The SD card contains more than one crontabs, they are named 'crontab-1579' / 'crontab-3781' 
When I edit the crontab with the 'crontab -e # select nano' command, always the latest version of the crontab will be opened, but saving it with <ctrl>x, y, <enter> will create a new crontab number...

starbasessd, 2022-04-28:
The command is:
> crontab -e

and the first time you run that command, it should ask you which editor to use, and I select "nano"
OK, let's take a step back:
You should have 2 scripts that you wrote, and 2 config files
did you set the execute for both scripts? 
>chmod +x /data/etc/daytime
>chmod +x /data/etc/nighttime

If you did, what happens when you run the scripts manually?
>/data/etc/daytime 

Should trigger the settings for daytime by switching the configuration script to the daytime config and restarting the motion service. Then you should run the 
>/data/etc/nighttime

and it should change the config file to the nighttime config, and restart the motion service.

If those work as expected, we can look at the crontab issue.

Sheep225, 2022-04-29:
I have now installed motioneyeos from scratch to have a clean os without traces of any unsucessful attempts.
Now I fail to save the daytime script: "Read-only file system"
How can I get write access to the SD card?

starbasessd, 2022-04-29:
If you use the path /data/etc for the scripts and the config files, that particular path is read-writable. Most of the rest of the system is read-only as you have found.

starbasessd, 2022-04-30:
Have you been able to sort out your issue, @Sheep225, or do you have any other questions?

Sheep225, 2022-05-01:
Sorry for the late answers, did not find the time to work on the Raspi...

When I run the scripts manually, they work (ssh shell says "stopping motioneye: done" / "starting motioneye: done"
and running the nighttime script does not create new pictures

so let's go to the crontab issue

starbasessd, 2022-05-01:
For testing purposes ONLY, lets set the crontab triggers to every other minute

1-59/2 * * * * /data/etc/daytime
*/2 * * * * /data/etc/nighttime

Let it run for 10 minutes or so and then we'll check the log:

cat /var/log/messages
to see if it ran.

Sheep225, 2022-05-01:
Output of cat /var/log/messages:

`May  1 19:14:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:15:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:16:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:17:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:18:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:19:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:20:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:21:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:22:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 19:23:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:24:10 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 45 C
May  1 19:25:11 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 45 C
`
seems it is not working....

starbasessd, 2022-05-01:
OK, interesting

For testing purposes ONLY, lets set the crontab triggers to every other minute

1-59/2 * * * * /data/etc/daytime && echo 'day' >> /var/log/messages
*/2 * * * * /data/etc/nighttime && 'night' >> /var/log/messages

Let it run for 10 minutes or so and then we'll check the log:

cat /var/log/messages
to see if it ran.

What this change will do is confirm that the line is being read by writing 'day' and 'night' to the messages file. 
It is strange that it doesn't appear that the crontab lines are running. 
I just threw up another PiZeroW with dev20201026 up, added lines via crontab -e:
1-59/2 * * * * echo 'odd' >> /var/log/messages
*/2 * * * * echo 'even' >> /var/log/messages
and I get:
May  1 13:37:00 meye-4cec78b8 cron.info crond[526]: USER root pid 8159 cmd echo 'odd' >> /var/log/messages
odd
May  1 13:38:00 meye-4cec78b8 cron.info crond[526]: USER root pid 8262 cmd echo 'even' >> /var/log/messages
even
May  1 13:39:00 meye-4cec78b8 cron.info crond[526]: USER root pid 8362 cmd echo 'odd' >> /var/log/messages
odd
in my messages log...

Sheep225, 2022-05-01:
Here's what I get:
'May  1 19:40:11 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 47 C
May  1 19:41:11 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 46 C
May  1 19:42:11 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 46 C
May  1 19:43:11 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 46 C
May  1 19:44:11 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 45 C
May  1 19:45:11 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 46 C
'

starbasessd, 2022-05-01:
Confirm which version you are using.
In the WebGUI, what versions do you have for motionEye, motion and OS?
[image]

Sheep225, 2022-05-01:
[image]

I re-edited the crontab again, and now I get the following:
'
May  1 19:48:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:49:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13001 cmd echo 'odd' >> /var/log/messages
odd
May  1 19:49:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:50:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13116 cmd echo 'even' >> /var/log/messages
even
'

starbasessd, 2022-05-01:
OK, so now set your daytime and nighttime settings to what you want (but leave the added && echo... parts) and let me know tomorrow if it stops and starts as you expect.
(change 1-5/2 * * * * and */2 * * * * to what you want...)

Previously:

00 7 * * 1-5 /data/etc/daytime && echo 'day' >> /var/log/messages
00 17 * * * /data/etc/nighttime && echo 'night' >> /var/log/messages

which would start the daytime settings at 7:00a Mon-Fri and night time settings at 5p every day (in case there is a random reboot in there somewhere)

starbasessd, 2022-05-01:
You should really upgrade to dev20201026, though. If you go to the WebGUI, Settings, Expert Settings, Enable PreReleases, then Settings, General, Software Update, Check, you can get the dev20201026 update automagically.

starbasessd, 2022-05-01:
If you removed the /data/etc/daytime and /data/etc/nighttime parts when you re-edited, then it still isn't running correctly...

Sheep225, 2022-05-01:
Ok, i will do the upgrade asap!

here is what i get after the last crontab edit (without the weekday settings):
'
ay  1 19:49:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13001 cmd echo 'odd' >> /var/log/messages
odd
May  1 19:49:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:50:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13116 cmd echo 'even' >> /var/log/messages
even
May  1 19:50:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:51:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13233 cmd echo 'odd' >> /var/log/messages
odd
May  1 19:51:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:52:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13345 cmd echo 'even' >> /var/log/messages
even
May  1 19:52:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 45 C
May  1 19:53:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13459 cmd echo 'odd' >> /var/log/messages
odd
May  1 19:53:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:54:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13574 cmd echo 'even' >> /var/log/messages
even
May  1 19:54:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:55:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13691 cmd echo 'odd' >> /var/log/messages
odd
May  1 19:55:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:56:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13804 cmd echo 'even' >> /var/log/messages
even
May  1 19:56:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:57:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 13920 cmd echo 'odd' >> /var/log/messages
odd
May  1 19:57:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:58:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 14032 cmd echo 'even' >> /var/log/messages
even
May  1 19:58:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 19:59:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 14171 cmd echo 'odd' >> /var/log/messages
odd
May  1 19:59:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 20:00:00 meye-eb2dff2a cron.info crond[1067]: USER root pid 14312 cmd /usr/sbin/dyndns-update
May  1 20:00:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 14313 cmd echo 'even' >> /var/log/messages
even
May  1 20:00:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 20:01:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 14429 cmd /data/etc/daytime
May  1 20:01:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
May  1 20:01:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May  1 20:01:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May  1 20:01:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2+gitUNKNOWN Started
May  1 20:01:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May  1 20:01:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May  1 20:01:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 20:02:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 44 C
May  1 20:03:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 14788 cmd /data/etc/nighttime
May  1 20:03:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
May  1 20:03:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May  1 20:03:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May  1 20:03:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2+gitUNKNOWN Started
May  1 20:03:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May  1 20:03:05 meye-eb2dff2a user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May  1 20:03:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 20:04:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 15032 cmd echo 'even' >> /var/log/messages
even
May  1 20:04:12 meye-eb2dff2a user.notice throttlewatch: currently: ok, temperature: 43 C
May  1 20:05:00 meye-eb2dff2a cron.info crond[1070]: USER root pid 15148 cmd echo 'odd' >> /var/log/messages
odd
'

I will try to get back on it tomorrow, but can't promise... come back asap!
Thank you for your patience!

starbasessd, 2022-05-01:
Weird, you have a mix of messages there. When you get a chance, post what's in crontab -e now...

Sheep225, 2022-05-02:
crontab -e is the follwowing:
'
0 20 * * * /data/etc/daytime # run daytime script at 5:59a
21 30 * * * /data/etc/nighttime # run nighttime script at 5:59p
30 04 * * * /data/etc/daytime
1-59/2 * * * * echo 'odd' >> /var/log/messages
*/2 * * * * echo 'even' >> /var/log/messages
'
I let it run over night and it started and stopped as scheduled. I think i got it now!
Will try to add the weekday settings and come back if there are any issues.

starbasessd, 2022-05-02:
0 20 * * * will run at 8p  (20:00) local time (or UTC if you haven't set your time zone)
21 30 * * * will never run, as there isn't a 30:21 hour in a day 
&nbsp;  &nbsp; (unless it triggers at 6:21a but I  woudn't bet on it)
30 04 * * * will trigger at 04:30a local time (or UTC if you haven't set your time zone)
1-59/2 * * * * triggers ever odd minute
*/2 * * * * triggers every even minute
&nbsp;
You should only have 2 lines in there when all is well and good, one with /data/etc/daytime, and one with /data/etc/nighttime. The others can be deleted.

Sheep225, 2022-05-02:
0 20 * * * should be 40 20 * * * 
(missed some digits when copying from terminal)

the current active crontab is now
'59 6 * * 1-5 /data/etc/daytime # Start Daytime settings at 6:59 Mon-Fri
59 17 * * * /data/etc/nighttime # Start Nighttime settings at 18:00 every day
'

(I have 2 PI Zero here, an active out in the garden and a second to play around and test stuff. So all my trials here were on the test Pi, and after I get it to work I will now do the settings on the active Pi)

starbasessd, 2022-05-02:
Those 2 lines should give you what you are looking for. Please keep me updated.

Sheep225, 2022-05-09:
So after testing the settings for a while, everything is working proprely. The camera stopped working on friday evening and started again today.
Thanks starbasessd for your support!

closed 2022-05-09 by Sheep225