motioneye-project/motioneyeos #2900: MotionEyeOS not working with wifi on Raspberry pi ZERO 2W

pdemeure, 2022-04-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2022-04-06:
Please post your wpa_supplicant.conf file
or confirm it is in this format:
- - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="password"
	key_mgmt=WPA-PSK
}
- - - - -
Country should be in the 2 character format
ssid and psk are case sensitive.
If using Windows to create the wpa_supplicant.conf file make sure there isn't a .txt at the end (happens when you create it with Notepad.  If you turn on 'file extentions' in the File Explorer window
you can remove it. If you don't turn it on, you'll never see it, and it will never work.
https://www.thewindowsclub.com/show-file-extensions-in-windows

pdemeure, 2022-04-06:
Hi,
Thanks for reply. I confirm that my wpa_supplicant.conf is exactly like you
write in your message. Created with notepad++

starbasessd, 2022-04-06:
Did you check to see if you have file extensions turned on?
Are you using the version for PiZero2W found [here](https://github.com/jawsper/motioneyeos/releases/download/20220119-dev/motioneyeos-raspberrypi3.img.gz)?
Without a monitor, it's hard to go at it blind...

pdemeure, 2022-04-06:
Hi,
I use the image from motioneye os site, the version for the raspberry zero.
The image in your link has pi3 in its name. Are you sure that it can be
used in a pi zero 2W ?
Best regards
Pierre

starbasessd, 2022-04-06:
Yes, because the PiZero2W has a Pi3 CPU on it and enough different other hardware that the Pi3 regular image doesn't work on it.  If you look through all the issues here about the PiZero2W here, @jawsper came up with a modded image that works on the PiZero2W and has been tested by many users.

pdemeure, 2022-04-07:
Hello,
Many thanks to all guys who helped me finding a solution.
The image of @jasper works great and it is based on a pi3 instead of simply
pi.
Note that it is mandatory to create with notepad++ the two files ssh and
wpa_supplicant.conf
Then my raspberry pi zero 2w started immediately and was reachable directly
with its ip address given by wifi.
No use of keyboard and screen.
HAve a nice day
Pierre

starbasessd, 2022-04-07:
Please close if you are working.

pdemeure, 2022-04-07:
Great support. Thanks !!!

closed 2022-04-07 by pdemeure

Fenestrane1, 2024-05-26:
hi starbasessd, saw you in a lotta conversations like this trying to find out what the problem is with my pi zero 2w, could you please help me out? im making a motionIOS camera and the following problem occurs everytime.

 the console just keeps saying 'configuring wired network: no device panic: rebooting in 10 seconds(caused by network)'. I dont want to connect via ethernet but through wifi itself, would be really happy if you can help me out.

A bit more specifications:
Im using the pi zero 2 w
It says a few things in the console and few after the message and few before above but restarts again and again after a few second

The first time i tried it, it restarted after 10 seconds, then after rebooting it started after 20, after taking out the SD card and checking everything and starting it, it restarted after 30 second, so like it adds 10 seconds after every time it restarts, just told if it could be useful or worth noting.