motioneye-project/motioneyeos #2901: Dahua camer DH-SD1A404XB-GNR

chersoph, 2022-04-11:
The camera DH-SD1A404XB-GNR has a stream rtsp://192.168.1.108:554/cam/realmonitor?channel=1&subtype=0 that I can read with VLC.
I have set it upp in MotionEye on RPi4 running Home assistant., it opens a window but the window remains grey with out presenting a a live stream.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: ( 0.18.0).

### Board Model

I am using the following board/model: (RPi4).

### Camera

I am using the following type of camera: (Ip-camera).

My camera model is: (Dahua DH-SD1A404XB-GNR).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

External Hard Disk

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2022-04-11:
The [iSpy database](https://www.ispyconnect.com/camera/dahua) recommends this URL:
rtsp://192.168.1.108:8554/cam/realmonitor?channel=1&subtype=00&authbasic=YWRtaW46YWRtaW4=
using your camera's user & password.

starbasessd, 2022-04-11:
Don't use the HA installation. Uninstall it, and install motionEye from [here](https://github.com/motioneye-project/motioneye/wiki/Installation) for your particular setup/OS or use the Docker version.

chersoph, 2022-04-11:
I'm running HASSIO.

starbasessd, 2022-04-11:
Same Difference.