motioneye-project/motioneyeos #2902: using both networks on rpi 3

festrito, 2022-04-26:
Hi, I have the latest motioneyeos and I want to ask how to set up raspberrypi3 to connect via wifi to the home network and ethernet to the network camera and assign it an ip address. thank you for your help

starbasessd, 2022-04-26:
it was not designed with that use case in mind. It is not set up with the routing tables needed. You can do so with motionEye on RPiOS.

festrito, 2022-04-26:
Thank you for your response. if I set the camera to a static ip address would it work? without dhcp server?

starbasessd, 2022-04-26:
Probably not, as there is no way to set a static IP address on the Ethernet Port if you are using the WiFi also. See: [here](https://github.com/ccrisan/thingos/wiki/static_ip.conf)
Specifically, [here](https://github.com/ccrisan/thingos/wiki/static_ip.conf#static_ip)

festrito, 2022-04-27:
Yes, thank you for the explanation.

starbasessd, 2022-04-27:
If you use RaspiOS Buster (Legacy) and motionEye, it could be done, though.
[mEye on RPiOS](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian)
And you can find instructions for setting static IPs on the Ethernet to do what you want to do on lots of sites.

festrito, 2022-04-27:
yes thanks. I still have a question, is omx / v4l2 encoding only for raspberrypi or is it possible to use it on another platform?