motioneye-project/motioneyeos #2903: apt missed on Raspi

bad-friday, 2022-04-30:


starbasessd, 2022-04-30:
MotionEyeOS is Debian based, but is a minimalist OS (can be run in a 2GB SDCard), set to use a mostly read-only file system to protect it from the issues associated with using SDCards. There are ways around the read-only file system, but not usually needed. @bad-friday what are you trying to add? I may be able to help you.

bad-friday, 2022-04-30:
Thank  you Michalng and starbasessd.
I'd like to install th MotionEyeOS on an USB-Stickbecause of the limitations of the SSD-card, but wasn't able to boot from the USB-Stick, four strawberrys on th top of the screen and that's it

MichaIng, 2022-04-30:
> MotionEyeOS is Debian based

Thanks for clarification, I didn't know that.

@bad-friday 
Do you still have an SD card attached? AFAIK by default the bootloader will search for an OS on the SD card.
And do you have other USB devices attached with may cause a voltage issue? The four USB ports on the Raspberry Pi share an overall limit of 1.2 A, so e.g. a purely USB powered 2.5" drive, including SSD, can cause voltage issues, especially on boot when a power draw peak occurs.

starbasessd, 2022-04-30:
To boot motionEyeOS from a USB stick, you need to use dev20201026, and change the cmdline.txt file:
replace
> root=/dev/mmcblk0p2 

to
> root=/dev/sda2

prior to first boot.

This works for me on everything from PiZeroW to Pi4

starbasessd, 2022-04-30:
@bad-friday BTW, not 'strawberries', they are 'raspberries' :)

bad-friday, 2022-04-30:
@MichaIng  
I run the Pi with an original powersupply and use only sticks, no other kind of USB-devices.

@starbasessd  
Thank's for the hint with the cmdline.text.
Of course your're right. It's a Raspberry Pi, so it shows some of it.

btw. how do you add emoticons?

starbasessd, 2022-04-30:
The little smiley face at the top right of the message box...

MichaIng, 2022-04-30:
Ah, good hint with the root filesystem entry in `cmdline.txt`. Instead of headcoding it to a specific device path, we should make it a PARTUUID based entry, like `root=PARTUUID=907af7d0-02`. This way it is assured to check for the correct root filesystem, regardless on which drive it is located.

A task for the build scripts, but for now good to know and sufficiently easy to edit `cmdline.txt` manually.

starbasessd, 2022-04-30:
I do believe it was on the 'wish list' for the next version of motionEyeOS, it was an issue with the interaction of buildroot, thingOS, and the general system. Calen had added the changes needed to allow the change of root path in dev20201026, and would probably had that as part of the next overall build, and then started testing the PARTUUID change in the next dev version. It should be an option built into buildroot by now. I believe by enabling the PARTUUID change, motionEyeOS could be run in EMC memory on the CM4s, too...

MichaIng, 2022-04-30:
Sounds good, so we should be able to revive this idea.

> I believe by enabling the PARTUUID change, motionEyeOS could be run in EMC memory on the CM4s, too...

Jep, exactly, and same for other devices, too, when doing a similar change to the U-Boot configs. But one by one 🙂.

bad-friday, 2022-05-01:
The change in cmdline.txt did it..
Why are the file systems read-only? Can this be changed?

MichaIng, 2022-05-01:
```sh
findmnt -no PARTUUID /
```
Read our above commends about why it's read-only (increasing SD card life time, not intended to be used as fully fledged OS but more like a plug and play container with a fixed set of software). Usually one can change it in `/etc/fstab` (removing the `ro` option from the root mount entry), but I'm not sure whether this is all on motionEyeOS.

starbasessd, 2022-05-01:
In the [FAQ's](https://github.com/motioneye-project/motioneyeos/wiki/FAQ):
[Why do I get Read-only file system when trying to edit a file?](https://github.com/motioneye-project/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file)

starbasessd, 2022-05-01:
> ```shell
> findmnt -no PARTUUID /
> ```

findmnt doesn't exist in motionEyeOS
motionEyeOS uses busybox for most of its 'standard' linux command structure, and blkid and findmnt aren't included in the version of busybox used in motionEyeOS.

Why? I don't know. maybe something to be 'fixed' (build in the full busybox next version?) It was a design choice made by the creator.

MichaIng, 2022-05-01:
Is `lsblk` included? Those three are all tool I know to read the PARTUUID 😄.

starbasessd, 2022-05-01:
Nope
Currently defined functions:
 [, [[, arch, arping, awk, basename, bc, blkdiscard, bzcat, cat, chgrp, chmod, chown, chroot, cksum, clear, cmp,
 cp, crond, crontab, cut, date, dd, delgroup, deluser, devmem, df, diff, dirname, dmesg, dnsdomainname,
 dos2unix, du, echo, egrep, eject, env, expr, factor, fallocate, false, fatattr, fgrep, find, fold, free,
 fsfreeze, getty, grep, halt, head, hexdump, hexedit, hostid, hostname, hwclock, i2cdetect, i2cdump, i2cget,
 i2cset, i2ctransfer, id, ifdown, ifplugd, ifup, init, insmod, install, ip, ipaddr, iplink, ipneigh, iproute,
 iprule, iptunnel, kill, killall, killall5, less, link, linux32, linux64, linuxrc, ln, logger, login, logname,
 losetup, ls, lsmod, lsscsi, lsusb, lzcat, lzopcat, makedevs, md5sum, mesg, mkdir, mkdosfs, mke2fs, mkfifo,
 mknod, mkpasswd, mktemp, modprobe, mount, mv, nameif, netstat, nice, nl, nohup, nologin, nproc, nsenter,
 nslookup, nuke, od, partprobe, passwd, paste, patch, pidof, ping, poweroff, printenv, printf, ps, pwd,
 readlink, realpath, reboot, renice, reset, resume, rev, rfkill, rm, rmdir, rmmod, route, run-init, run-parts,
 sed, seq, setfattr, setpriv, sh, sha1sum, sha256sum, sha3sum, sha512sum, shred, sleep, sort, ssl_client, stat,
 stty, su, svc, svok, swapoff, swapon, sync, sysctl, syslogd, tail, tc, tee, test, time, timeout, top, touch,
 tr, traceroute, true, truncate, ts, tty, ubirename, uevent, umount, uname, uniq, unix2dos, unlink, unlzop,
 unshare, uptime, usleep, uudecode, uuencode, vi, vlock, w, watch, watchdog, wc, which, who, whoami, xargs, xxd,
 xzcat, yes, zcat

found using cmd
> busybox --help

in motionEyeOS

MichaIng, 2022-05-01:
Another way is to attach the SD card/drive to an external Linux system and examine the PARTUUID there. Not sure whether there is some info in `/sys/class/block`, but AFAIK it's not.

starbasessd, 2022-05-01:
in motionEyeOS:
ls -la /sys/class/block/mmcblk0p1
drwxr-xr-x    5 root     root             0 Dec 31  1969 ./
drwxr-xr-x   11 root     root             0 Dec 31  1969 ../
-r--r--r--    1 root     root          4096 May  1 09:30 alignment_offset
-r--r--r--    1 root     root          4096 May  1 09:30 dev
-r--r--r--    1 root     root          4096 May  1 09:30 discard_alignment
drwxr-xr-x    2 root     root             0 May  1 09:30 holders/
-r--r--r--    1 root     root          4096 May  1 09:30 inflight
-r--r--r--    1 root     root          4096 Dec 31  1969 partition
drwxr-xr-x    2 root     root             0 May  1 09:30 power/
-r--r--r--    1 root     root          4096 May  1 09:30 ro
-r--r--r--    1 root     root          4096 Dec 31  1969 size
-r--r--r--    1 root     root          4096 Dec 31  1969 start
-r--r--r--    1 root     root          4096 May  1 09:30 stat
lrwxrwxrwx    1 root     root             0 May  1 09:30 subsystem -> ../../../../../../../../../../class/block/
drwxr-xr-x    2 root     root             0 May  1 09:30 trace/
-rw-r--r--    1 root     root          4096 Dec 31  1969 uevent

partition just shows 1

MichaIng, 2022-05-01:
I also checked on Debian and couldn't find a way to read out UUID or PARTUUID like this. However, not such a big issue: With current images, users can safely use the device path, and for future motionEyeOS images, we can set the PARTUUID based root argument at compile time.

starbasessd, 2022-05-01:
Debian 10 (Buster):
sudo blkid
gives me a list of all the partitions & partuuids:
 sudo blkid
/dev/nvme0n1p1: UUID="b1a8da34-b7b0-4710-a970-13b4d7fe7aaa" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="d47fbb55-01"
/dev/nvme0n1p5: UUID="0ec50f82-9397-46af-9831-76c6a4b5f873" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="d47fbb55-05"
/dev/nvme0n1p6: UUID="1cbe409f-b2b2-468a-a1de-9bf245d3ade6" TYPE="swap" PARTUUID="d47fbb55-06"
/dev/nvme0n1p7: UUID="987b13c3-989f-4bfb-81e1-b953c7d9e076" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="d47fbb55-07"
/dev/nvme0n1p8: UUID="76b4fb90-ebd4-4f6f-a4c4-c630615234f0" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="d47fbb55-08"
/dev/sdc1: BLOCK_SIZE="512" UUID="552224E734328397" TYPE="ntfs" PARTUUID="cf8d9a84-f3ab-4ed8-ba2a-e429a60eeb09"
/dev/sdb1: LABEL="Movies2" BLOCK_SIZE="512" UUID="1F4CF3717E36B4FF" TYPE="ntfs" PARTLABEL="Movies2" PARTUUID="f468fbbb-9ab2-41aa-ba94-b5f43018238f"
/dev/sda1: UUID="1dda8eec-2db8-40b8-b0bd-9c92d0908a69" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="ff58fb9c-01"
/dev/sdd1: LABEL="MovieData" BLOCK_SIZE="512" UUID="6851AAAA0399F882" TYPE="ntfs" PARTLABEL="MovieData" PARTUUID="61bb546a-3f09-41c6-820b-2bab8928b5ed"

It's just motionEyeOS with the limitation (due to the limited busybox install, probably)

MichaIng, 2022-05-01:
On Debian/Raspbian, all three commands are provided by the `util-linux` package, which is an "essential" package in the APT repository, hence it is pre-installed nearly forcefully on every APT-based distro, trying to remove it prompts with a verification message and `apt upgrade` as well as any `apt install ...` will automatically reinstall it. So jep, definitely dedicated to motionEyeOS, which is however fine according what it aims to be.

starbasessd, 2022-05-01:
Debian Buster:
sudo lsusb -o NAME,PARTUUID
shows
NAME        PARTUUID
sda
└─sda1      ff58fb9c-01
sdb
└─sdb1      f468fbbb-9ab2-41aa-ba94-b5f43018238f
sdc
└─sdc1      cf8d9a84-f3ab-4ed8-ba2a-e429a60eeb09
sdd
└─sdd1      61bb546a-3f09-41c6-820b-2bab8928b5ed
sr0
nvme0n1
├─nvme0n1p1 d47fbb55-01
├─nvme0n1p2 d47fbb55-02
├─nvme0n1p5 d47fbb55-05
├─nvme0n1p6 d47fbb55-06
├─nvme0n1p7 d47fbb55-07
└─nvme0n1p8 d47fbb55-08

MichaIng, 2022-05-02:
Jep, basically every partition has a PARTUUID, which also does not change when reformatting the filesystem (in which case the UUID changes). Other than UUID, PARTUUID can be usually used in boot configs without initramfs, i.e. on RPi which by default does not use/require an initramfs with the FAT boot partition and builtin `ext4` module.

closed 2022-05-05 by bad-friday