motioneye-project/motioneyeos #2904: imx519 is not working

jhall0326, 2022-05-02:
MotionEyeOS is phenominal.  Great Job.  Unfortunately, MEye is just unable to detect and use my camera.  As I understand it, no one has been successful in getting MEye to use the imx519.  I have found hundreds of discussions regarding futile attempts.

Can you please get MEye to support this camera and I would think you also would want to support Arducam's next big release. https://www.arducam.com/64mp-ultra-high-res-camera-raspberry-pi/

This is their 64 MP camera with Autofocus.

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (motioneyeos-raspberrypi4-20200606.img).

### Board Model

I am using the following board/model: (Raspberry PI 4B+).

### Camera

I am using the following type of camera: (MMAL).

My camera model is: (imx519).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2022-05-02:
I would look forward to the next official release of mEOS release, as the imx519 wasn't supported in the kernel release that 20200606 used (as the ultra high resolution cameras were just being released at the time). Have you tried the dev20201026 release of motionEyeOS?

jhall0326, 2022-05-06:
Actually that is what I am running.  I have the RPi 4B and the 20200606 release wouldn't run on this board.

starbasessd, 2022-05-06:
We who try to help are dependent on your answers. 
At this point, you will have to wait for the new developers to trigger a new build. They are currently working on the motionEye on other OSs software.