motioneye-project/motioneyeos #2908: motioneyeos does not recognise the camera

bad-friday, 2022-05-10:
I am running motionEyeOS version 20201026
On a Raspberry Pi 4B 4GB
Camera is a Raspberry Pi V2

Now i would like to go to motioneye.eo
Unfortunately the camera is not recognised.
Camera Device: /dev/video10
Camera Type: V4L2 Camera

On motioneyeos it Works as it should.
Camera Device:  vc.nil.camera
Camera Type: MMAL Camera

starbasessd, 2022-05-10:
Try with Raspbian10(Buster/Legacy)

bad-friday, 2022-05-10:
But that would really be a step backwards.
And the camera I have in mind (Arducam Camera) would not work with it.
In any case, I have not yet received an answer from them in this regard.

starbasessd, 2022-05-10:
[RPiFoundation](https://www.raspberrypi.com/news/bullseye-camera-system/)
[v4l2 rtsp server breaks](https://github.com/mpromonet/v4l2rtspserver/issues/257)
[libcamera](https://github.com/raspberrypi/libcamera-apps/issues/168)

bad-friday, 2022-05-13:
I now have a system with Raspbian Buster
Unfortunately, the installation of MotionEye.eo does not work as expected. The following happens after sudo apt-get install python3-tornado python3-jinja2 python3-pillow python3-pycurl python3-babel python3-numpy python3-boto3
has run through satisfactorily.
Then:
/home/swe/ sudo pip install motioneye.eo
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye.eo
  Could not find a version that satisfies the requirement motioneye.eo (from versions: )
No matching distribution found for motioneye.eo
/home/swe/

starbasessd, 2022-05-13:
I am not sure where you are getting 'motioneye.eo', I do not see any mention of it on the motioneye github. It sounds like you may have found a forked version.
I would look at this thread: [2373](https://github.com/motioneye-project/motioneye/issues/2373) which discusses how to get the 0.43.0 motionEye installed. They are discussing how to get it available in Pypi, too.
If you post in the motionEye issues github, a new developer may be able to help you further.

bad-friday, 2022-05-13:
I've found it here: https://jmichault.github.io/motioneye.eo-dok/en/

starbasessd, 2022-05-13:
It is definitely a fork, and I would try to contact the author.

starbasessd, 2022-05-13:
This is the specific github repository for motioneye.eo with their issues: [motioneye.eo](https://github.com/jmichault/motioneye.eo/issues)

bad-friday, 2022-05-13:
> It is definitely a fork, and I would try to contact the author.

What's a «fork» ?

bad-friday, 2022-05-13:
> (https://github.com/jmichault/motioneye.eo/issues)

starbasessd, 2022-05-13:
A [fork](https://en.wikipedia.org/wiki/Fork_(software_development)) is where someone takes a package  (like motionEye) or OS and makes their own proprietary changes for a specific use case or fix. 
RaspberryPiOS is a fork of Debian, as is Ubuntu, Mint and many others.
When some one or some group forks a package, they assume responsibility for any changes they make.

bad-friday, 2022-05-13:
> This is the specific github repository for motioneye.eo with their issues: [motioneye.eo](https://github.com/jmichault/motioneye.eo/issues)

I am not at all versed in the use of github!

starbasessd, 2022-05-13:
If you click on this link: [motioneye.eo](https://github.com/jmichault/motioneye.eo/issues) you can add an issue there that the author of motioneye.eo JMichault can respond to more easily. 
Or, you can follow the install instructions for [motionEye on RaspberryPIOS Buster](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) on a clean install of RaspberryPiOS Buster (Legacy) 
Or you can use motionEyeOS which you state works...

starbasessd, 2022-05-13:
BTW, thinking a bit, did you enable the camera in 
>sudo raspi-config
>Interfaces
>Camera

In your RaspberryPiOS Buster?
What does
>vcgencmd get-camera

report when you are running Buster?

bad-friday, 2022-05-13:
Thanks a lot for your kind help.

bad-friday, 2022-05-13:
vcgencmd get-camera
error=1 error_msg="Command not registered"
Use 'vcgencmd commands' to get a list of commands

No idea what you'd like to know and there's no man-page

starbasessd, 2022-05-13:
Sorry, my bad. It's
>vcgencmd get_camera

typo on my part...

bad-friday, 2022-05-13:
vcgencmd get_camera
supported=1 detected=1

starbasessd, 2022-05-13:
The RaspberryPiOS sees the camera. motionEyeOS sees the camera and works. At this point, I would bet if you used the [motionEye install instructions](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) on a clean install, it would work.
Help for motioneye.eo would have to come from them.

bad-friday, 2022-05-13:
> The RaspberryPiOS sees the camera. motionEyeOS sees the camera and works. At this point, I would bet if you used the [motionEye install [...]
I'm not sure about the state of my Buster-Raspberry.
I've installed it according to this: https://jmichault.github.io/motioneye.eo-dok/en/instalado_en_debian/
But I'm not sure if the backup I've restore rightnow was drawn right after the install.
I'll see.
Anyway thanks a lot.

starbasessd, 2022-05-13:
I assume you are doing the motioneye.eo install for a specific language translation. If it were me, I'd work with the developers here to get your specific language added in the official package.
@michalng @cclauss and others are doing the new dev work on motioneye. I know they would appreciate any input you can give on the translation, over on the [motioneye](https://github.com/motioneye-project/motioneye/issues) pages.