motioneye-project/motioneyeos #2909: Rpi4 IMX219 Camera Detected and auto setup as MMAL but "Unable to open Video Device"

jhall0326, 2022-05-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20201026.

### Board Model

I am using the following board/model: Raspberry Pi 4 B.

### Camera

I am using the following type of camera: MMAL.

My camera model is: IMX219.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Rpi4 Motioneyeos dev20201026 imx219. Camera works fine on a Bullseye build. vcgencmd get_camera reports supported=1 detected=1. MotioneEyeOS detects it and sets it up as MMAL Camera with Camera Device=vc.ril.camera but within the MotionEyeOS GUI it is a gray square saying "Unable to open video device.

Motion.log:
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)

jhall0326, 2022-05-16:
I purchased 2 of these cameras and they both have the same issue.  They work fine using libcamera-still but Unable to open video device in MEYEOS.

jhall0326, 2022-05-17:
After a period of time, the error in motion.log changes to the below:

mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
[1:ml1:Camera1] [ERR] [VID] create_camera_component: Failed to create MMAL camera component vc.ril.camera
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)

jhall0326, 2022-05-17:
Log Files Attached:

[motion.log](https://github.com/motioneye-project/motioneyeos/files/8707900/motion.log)
[dmesg.log](https://github.com/motioneye-project/motioneyeos/files/8707902/dmesg.log)
[boot.log](https://github.com/motioneye-project/motioneyeos/files/8707904/boot.log)
[messages.log](https://github.com/motioneye-project/motioneyeos/files/8707905/messages.log)
[motioneye.log](https://github.com/motioneye-project/motioneyeos/files/8707906/motioneye.log)

starbasessd, 2022-05-19:
I notice that in dmesg log it is overloaded with the PixArt Gaming Mouse constantly being re-discovered. I would try to take that off the Pi, first.
Are you manually setting any settings? Like wireless, or camera?
What have you set frame rate, and resolution to?
If you have monitor and keyboard attached (don't need / can't use a mouse anyway) can you run 
vcgencmd get_throttled
and
vcgencmd get_camera
and report results, please?

starbasessd, 2022-05-19:
Let's force clear the logs, so we're not looking at a lot of old data:
>logrotate  /etc/logrotate.conf

reboot
then post the logs.
I am wondering if the IMX cameras are newer than the kernel in mEOS dev20201026 (like some of the new 16MP and higher resolution that weren't yet supported in the dev20201026 kernel.
If that is the case (and looking more likely since the 5MP camera DOES work) you may have to wait for the new developers to post a new version (with updated kernel). Another way to test this thought is to install an older version of RPiOS (Pre-Bullseye) like maybe:
[archive releases](https://downloads.raspberrypi.org/raspios_lite_armhf/images/)
[24/08/2020](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/)
If this image is 'broken' for the IMX but works on the 5MP, then the odds that it's not in the kernel just went way up...

jhall0326, 2022-05-19:
That certainly sounds like what I'm dealing with.  Unsupported camera for the driver version.

[motioneye (3).log](https://github.com/motioneye-project/motioneyeos/files/8732566/motioneye.3.log)
[motion (5).log](https://github.com/motioneye-project/motioneyeos/files/8732567/motion.5.log)
[dmesg (5).log](https://github.com/motioneye-project/motioneyeos/files/8732568/dmesg.5.log)
[boot (3).log](https://github.com/motioneye-project/motioneyeos/files/8732569/boot.3.log)
[messages (3).log](https://github.com/motioneye-project/motioneyeos/files/8732570/messages.3.log)

jhall0326, 2022-05-19:
Also, when I delete and recreate the camera, I am now seeing under Camera Type=local V4L2 Camera, when I click the dropdown for camera, there is an entry at the bottom that used to not be there called "mmal service 16.1.  When I choose this option, I get the same symptom.

starbasessd, 2022-05-19:
Your logs didn't reset (or you are not grabbing the latest/current files)
Want to give the kernel check test a go? Load the RPiOS from 24 Aug 2020? Do NOT add any additional drivers, as it is difficult to impossible to do so in mEOS.

starbasessd, 2022-05-19:
Bullseye _is_ the 'latest and greatest', and motionEyeOS has been somewhat neglected for almost 2 years (new devs are working hard to correct the python2 vs python3 issues currently and will move into mEOS here soon)

starbasessd, 2022-05-19:
4.163245] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1280x720
I would recommend 1280x720 (if the IMX camera is attached during this log's boot)

jhall0326, 2022-05-19:
After setting to 1280X720 then reboot.  New logs.

[motion (9).log](https://github.com/motioneye-project/motioneyeos/files/8733781/motion.9.log)
[dmesg (8).log](https://github.com/motioneye-project/motioneyeos/files/8733782/dmesg.8.log)
[boot (5).log](https://github.com/motioneye-project/motioneyeos/files/8733783/boot.5.log)
[messages (5).log](https://github.com/motioneye-project/motioneyeos/files/8733784/messages.5.log)
[motioneye (5).log](https://github.com/motioneye-project/motioneyeos/files/8733785/motioneye.5.log)

starbasessd, 2022-05-19:
I think we're chasing our own tail here. What happened with RPiOS Buster 24 Aug 2020?

jhall0326, 2022-05-19:
It took me a while to find it and download it.  Getting ready to create the img now.

starbasessd, 2022-05-19:
I put links to it above...

jhall0326, 2022-05-19:
Okay!  It's booted and camera enabled.  Now what?

starbasessd, 2022-05-19:
Try raspivid or even install motionEye for Raspberry Pi from here: 
https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian

jhall0326, 2022-05-19:
pi@raspberrypi:~ $ raspivid
-bash: raspivid: command not found

starbasessd, 2022-05-19:
I don't think you grabbed the correct image. it's this one:
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/

jhall0326, 2022-05-19:
sudo pip install motioneye
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Downloading https://files.pythonhosted.org/packages/c5/ff/139b4b02ed089b2b50eee65d1059ae0d29064ab5f4ad429e45c702396697/motioneye-0.42.1.tar.gz (758kB)
    100% |████████████████████████████████| 768kB 579kB/s
Requirement already satisfied: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye) (2.10)
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye) (5.4.1)
Collecting pycurl (from motioneye)
  Downloading https://files.pythonhosted.org/packages/ef/05/4b773f74f830a90a326b06f9b24e65506302ab049e825a3c0b60b1a6e26a/pycurl-7.43.0.5.tar.gz (216kB)
    100% |████████████████████████████████| 225kB 1.5MB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-j2562o/pycurl/setup.py", line 961, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-install-j2562o/pycurl/setup.py", line 623, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-j2562o/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-j2562o/pycurl/setup.py", line 238, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-j2562o/pycurl/
pi@raspberrypi:~ $

starbasessd, 2022-05-19:
Building a Pi3B+ machine, stand by...

starbasessd, 2022-05-19:
Also under thunderstorm, will be back shortly.

starbasessd, 2022-05-19:
BTW, quick test. I installed 08/2020 image from link just above, without doing anything else, ran command:
>raspivid

pi@raspberrypi:~ $ raspivid

"raspivid" Camera App (commit f97b1af1b3e6 Tainted)

Display camera output to display, and optionally saves an H264 capture at requested bitrate

usage: raspivid [options]

Image parameter commands

-b, --bitrate   : Set bitrate. Use bits per second (e.g. 10MBits/s would be -b 10000000)
-t, --timeout   : Time (in ms) to capture for. If not specified, set to 5s. Zero to disable
-d, --demo      : Run a demo mode (cycle through range of camera options, no capture)
-fps, --framerate       : Specify the frames per second to record
-e, --penc      : Display preview image *after* encoding (shows compression artifacts)
-g, --intra     : Specify the intra refresh period (key frame rate/GoP size). Zero to produce an initial I-frame and then just P-frames.
-pf, --profile  : Specify H264 profile to use for encoding
-td, --timed    : Cycle between capture and pause. -cycle on,off where on is record time and off is pause time in ms
-s, --signal    : Cycle between capture and pause on Signal
-k, --keypress  : Cycle between capture and pause on ENTER
-i, --initial   : Initial state. Use 'record' or 'pause'. Default 'record'
-qp, --qp       : Quantisation parameter. Use approximately 10-40. Default 0 (off)
-ih, --inline   : Insert inline headers (SPS, PPS) to stream
-sg, --segment  : Segment output file in to multiple files at specified interval <ms>
-wr, --wrap     : In segment mode, wrap any numbered filename back to 1 when reach number
-sn, --start    : In segment mode, start with specified segment number
-sp, --split    : In wait mode, create new output file for each start event
-c, --circular  : Run encoded data through circular buffer until triggered then save
-x, --vectors   : Output filename <filename> for inline motion vectors
-if, --irefresh : Set intra refresh type
-fl, --flush    : Flush buffers in order to decrease latency
-pts, --save-pts        : Save Timestamps to file for mkvmerge
-cd, --codec    : Specify the codec to use - H264 (default) or MJPEG
-lev, --level   : Specify H264 level to use for encoding
-r, --raw       : Output filename <filename> for raw video
-rf, --raw-format       : Specify output format for raw video. Default is yuv
-l, --listen    : Listen on a TCP socket
-stm, --spstimings      : Add in h.264 sps timings
-sl, --slices   : Horizontal slices per frame. Default 1 (off)

H264 Profile options :
baseline,main,high

H264 Level options :
4,4.1,4.2

H264 Intra refresh options :
cyclic,adaptive,both,cyclicrows

Raw output format options :
yuv,rgb,gray

Raspivid allows output to a remote IPv4 host e.g. -o tcp://192.168.1.2:1234or -o udp://192.168.1.2:1234
To listen on a TCP port (IPv4) and wait for an incoming connection use the -l option
e.g. raspivid -l -o tcp://0.0.0.0:3333 -> bind to all network interfaces,
raspivid -l -o tcp://192.168.1.1:3333 -> bind to a certain local IPv4 port

Common Settings commands

-?, --help      : This help information
-w, --width     : Set image width <size>
-h, --height    : Set image height <size>
-o, --output    : Output filename <filename> (to write to stdout, use '-o -'). If not specified, no file is saved
-v, --verbose   : Output verbose information during run
-cs, --camselect        : Select camera <number>. Default 0
-md, --mode     : Force sensor mode. 0=auto. See docs for other modes available
-gps, --gpsdexif        : Apply real-time GPS information to output (e.g. EXIF in JPG, annotation in video (requires libgps.so.23)
[... 36 lines cut ...]
-ae, --annotateex       : Set extra annotation parameters (text size, text colour(hex YUV), bg colour(hex YUV), justify, x, y)
-ag, --analoggain       : Set the analog gain (floating point)
-dg, --digitalgain      : Set the digital gain (floating point)
-set, --settings        : Retrieve camera settings and write to stdout
-fw, --focus    : Draw a window with the focus FoM value on the image.

Notes

Exposure mode options :
off,auto,night,nightpreview,backlight,spotlight,sports,snow,beach,verylong,fixedfps,antishake,fireworks

Flicker avoid mode options :
off,auto,50hz,60hz

AWB mode options :
off,auto,sun,cloud,shade,tungsten,fluorescent,incandescent,flash,horizon,greyworld

Image Effect mode options :
none,negative,solarise,sketch,denoise,emboss,oilpaint,hatch,gpen,pastel,watercolour,film,blur,saturation,colourswap,washedout,posterise,colourpoint,colourbalance,cartoon

Metering Mode options :
average,spot,backlit,matrix

Dynamic Range Compression (DRC) options :
off,low,med,high

starbasessd, 2022-05-19:
Sorry, haad thunderboomers, then had to feed grandkids. Finishing up:

>sudo apt update && sudo apt upgrade -y

Following instructions from Wiki motionEye, Installs, Install on RaspberryPiOS Buster:
https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian
Stage one:
>sudo su
>apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y

completed successfully
Stage two:
>wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb
>dpkg -i pi_buster_motion_4.3.2-1_armhf.deb

completed successfully
Stage three
>apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev -y

completed successfully
test which versions of python and pip are default:
>python --version

Python 2.7.16
>pip --version

pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)
Stage four
>pip install motioneye

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Downloading https://files.pythonhosted.org/packages/c5/ff/139b4b02ed089b2b50eee65d1059ae0d29064ab5f4ad429e45c702396697/motioneye-0.42.1.tar.gz (758kB)
    100% |████████████████████████████████| 768kB 305kB/s
Collecting jinja2 (from motioneye)
  Downloading https://files.pythonhosted.org/packages/7e/c2/1eece8c95ddbc9b1aeb64f5783a9e07a286de42191b7204d67b7496ddf35/Jinja2-2.11.3-py2.py3-none-any.whl (125kB)
    100% |████████████████████████████████| 133kB 1.5MB/s
Collecting pillow (from motioneye)
  Downloading https://files.pythonhosted.org/packages/b3/d0/a20d8440b71adfbf133452d4f6e0fe80de2df7c2578c9b498fb812083383/Pillow-6.2.2.tar.gz (37.8MB)
    100% |████████████████████████████████| 37.8MB 6.5kB/s
Collecting pycurl (from motioneye)
  Downloading https://files.pythonhosted.org/packages/ef/05/4b773f74f830a90a326b06f9b24e65506302ab049e825a3c0b60b1a6e26a/pycurl-7.43.0.5.tar.gz (216kB)
    100% |████████████████████████████████| 225kB 778kB/s
Collecting tornado<6,>=3.1 (from motioneye)
  Downloading https://files.pythonhosted.org/packages/e6/78/6e7b5af12c12bdf38ca9bfe863fcaf53dc10430a312d0324e76c1e5ca426/tornado-5.1.1.tar.gz (516kB)
    100% |████████████████████████████████| 522kB 441kB/s
Collecting MarkupSafe>=0.23 (from jinja2->motioneye)
  Downloading https://files.pythonhosted.org/packages/b9/2e/64db92e53b86efccfaea71321f597fa2e1b2bd3853d8ce658568f7a13094/MarkupSafe-1.1.1.tar.gz
Collecting backports_abc>=0.4 (from tornado<6,>=3.1->motioneye)
  Downloading https://files.pythonhosted.org/packages/7d/56/6f3ac1b816d0cd8994e83d0c4e55bc64567532f7dc543378bd87f81cebc7/backports_abc-0.5-py2.py3-none-any.whl
Collecting futures (from tornado<6,>=3.1->motioneye)
  Downloading https://files.pythonhosted.org/packages/d8/a6/f46ae3f1da0cd4361c344888f59ec2f5785e69c872e175a748ef6071cdb5/futures-3.3.0-py2-none-any.whl
Collecting singledispatch (from tornado<6,>=3.1->motioneye)
  Downloading https://files.pythonhosted.org/packages/3f/09/2a17c6505a8373229cb3de6161e3af74ff90afcf4516719aa6cf32fad151/singledispatch-3.7.0-py2.py3-none-any.whl
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from singledispatch->tornado<6,>=3.1->motioneye) (1.12.0)
Building wheels for collected packages: motioneye, pillow, pycurl, tornado, MarkupSafe
  Running setup.py bdist_wheel for motioneye ... done
  Stored in directory: /root/.cache/pip/wheels/b1/b5/20/266f8871b2b1f598c96a17e5545a16f2c2198956890f17fb5e
  Running setup.py bdist_wheel for pillow ... done
  Stored in directory: /root/.cache/pip/wheels/f6/0a/7c/5e6567101a10388b915c4ebf73edb849f73908ad154e9eb9bc
  Running setup.py bdist_wheel for pycurl ... done
  Stored in directory: /root/.cache/pip/wheels/98/a3/19/9439c79b88a742f508af0c4e410ae3f685db2e13d3374e798e
  Running setup.py bdist_wheel for tornado ... done
  Stored in directory: /root/.cache/pip/wheels/6d/e1/ce/f4ee2fa420cc6b940123c64992b81047816d0a9fad6b879325
  Running setup.py bdist_wheel for MarkupSafe ... done
  Stored in directory: /root/.cache/pip/wheels/f2/aa/04/0edf07a1b8a5f5f1aed7580fffb69ce8972edc16a505916a77
Successfully built motioneye pillow pycurl tornado MarkupSafe
Installing collected packages: MarkupSafe, jinja2, pillow, pycurl, backports-abc, futures, singledispatch, tornado, motioneye
Successfully installed MarkupSafe-1.1.1 backports-abc-0.5 futures-3.3.0 jinja2-2.11.3 motioneye-0.42.1 pillow-6.2.2 pycurl-7.43.0.5 singledispatch-3.7.0 tornado-5.1.1

Stage five
>mkdir -p /etc/motioneye
>cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

Stage six
>mkdir -p /var/lib/motioneye

Stage seven, fini
>cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
>systemctl daemon-reload
>systemctl enable motioneye
>systemctl start motioneye

- - - - -
Reboot, camera (CSI RPiCam V2) discovered, no issue, working.

jhall0326, 2022-05-20:
I'm going to reimage SD Card and start over.