motioneye-project/motioneyeos #2910: Logitech c920 keeps trying to refocus on a loop, need fixed focus

MagmaGuy, 2022-05-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Board Model

I am using the following board/model:  Raspberry PI 3B

### Camera

I am using the following type of camera: V4L2

My camera model is: logitech c920

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

Hello,

This is a pretty simple issue, my logitech c920 keeps getting stuck on trying to focus on a 1 second loop which basically makes it completely unusable. I would love to know if there's a way to turn autofocus off so I can force focus, I really don't need it to be automatic.

I've seen people complain that this model didn't focus in 2019, and didn't see any other issues related to focus so I am hoping someone can tell me how to force it off.

starbasessd, 2022-05-17:
I was unable to confirm this, as I don't have that model of camera.
go to console or SSH
>/usr/bin/v4l2-ctl --list-ctrls
if focus-auto is listed
>crontab -e

add line:
>@reboot /usr/bin/v4l2-ctl -c focus_auto=0
 
If it is not listed, then it's possible the version of v4l2 included with motionEyeOS doesn't have that feature available. I found the command [here](https://community.octoprint.org/t/step-by-step-to-disable-auto-focus-for-logitech-c920/4226/3)

MagmaGuy, 2022-05-17:
Thank you so much for the reply! This commands works when I run it manually, but it seems like the cron job doesn't work. I'm not sure exactly where it could've gone wrong, but it must be something I'm doing wrong.

starbasessd, 2022-05-17:
The command you use manually should be the same command you use following the @reboot. Another option is to tell it to wait a bit before running the command by using:
>@reboot sleep 30 && /usr/bin/v4l2-ctl -c focus_auto=0

be sure to add the full path in the crontab command line.