motioneye-project/motioneyeos #2916: Motorized Focus Arducam (not autofocus)

FireflyJar, 2022-06-04:
Hoping someone that knows how to do this.   what code to use to move the focus a bit at a time.

! I am not looking for Autofocus, just to setup the camera  and adjust the focus once and leave on.   and yes if it reboots, the focus will be reset to default, it only stays as long it is powered.

so MotionEye has action-buttons to run scripts and Arducam has an example python code to move the focus  , I don't know coding enough how to go about  making a python code  that the motioneye action button can trigger.  Is action buttons still working?

this is the example code ArduCam gave for using a keyboard to focus it
```
import os
import time
import sys
import threading
import pygame,sys
from pygame.locals import *
from time import ctime, sleep
pygame.init()
screen=pygame.display.set_mode((320,240),0,32)
pygame.key.set_repeat(100)
def runFocus(func):
 temp_val = 512
 while True:
   for event in pygame.event.get():
	if event.type ==KEYDOWN:
           print temp_val
	   if event.key == K_UP:
              print 'UP'
	      if temp_val < 1000:
		     temp_val += 10
	      else:
		     temp_val = temp_val
              value = (temp_val<<4) & 0x3ff0
              dat1 = (value>>8)&0x3f
	      dat2 = value & 0xf0
              
              os.system("i2cset -y 0 0x0c %d %d" % (dat1,dat2))
           elif event.key==K_DOWN:
	      print 'DOWN'
	      if temp_val <12 :
                     temp_val = temp_val
              else:
                     temp_val -= 10
              value = (temp_val<<4) & 0x3ff0
              dat1 = (value>>8)&0x3f
              dat2 = value & 0xf0
              os.system("i2cset -y 0 0x0c %d %d" % (dat1,dat2))

def runCamera():
    cmd = "sudo raspistill -t 0"
    os.system(cmd)
if __name__ == "__main__":
 t1 = threading.Thread(target=runFocus,args=("t1",))
 t1.setDaemon(True)
 t1.start()	
 runCamera()
```

heikkivirtanen, 2022-07-24:
I have a similar type of problem. My target is to control a servo with an action button integrated with the motioneyeos. This is my first raspberry pi project by the way. After reading about the action buttons I have tried the following.

- I have motioneyeos 20200606 installed and I can get both the camera and the UI working
- Then I first checked that the `motioneye.conf` is in `/etc` and it says that `conf_path` is in `/data/etc/`. There is no `/etc` subforder in `/data` folder so I created one.
- Subsequently, I created `right_1` file
```
#!/bin/bash
/usr/bin/python /data/etc/right.py
```
and my right.py looks like this
```
#!/usr/bin/python
import RPi.GPIO as GPIO
from time import sleep

gpioPIN = 12
freq = 20

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(gpioPIN,GPIO.OUT)
pwm=GPIO.PWM(gpioPIN,freq)
pwm.start(0)
sleep(1)

dutyC = 4.2
pwm.ChangeDutyCycle(dutyC)
sleep(1)

pwm.stop()
GPIO.cleanup()
```
- I tested that right.py works in raspberry pi os (the servo really moves).
- I also did a `chmod +x /data/etc/right_1` and `chmod +x /data/etc/right.py` to make both files executable. Even though as far as I understand this should be carried out only for  _1 files.
- I moved both files to `/data/etc/` and rebooted the system.
- I don't get the action button visible in camera. Even though I click on the camera view, the action button doesn't appear.
- I also tested: [https://github.com/motioneye-project/motioneye/wiki/Monitoring-Commands](url) but it did not work in my opinion either.
- I have raspberry pi zero w

Is there a hint or help what I am doing wrong?

starbasessd, 2022-07-24:
First thing, I would backup my camera config (Settings, General, Backup)
I would then re-image with dev20201026,as there are a number of issues with 20200606.
I would then restore the camera config, and put your script in the /data/etc folder.
Then try it.
Please include the motioneye log and messages log (Settings, Expert Settings)

starbasessd, 2022-07-24:
Although the script works in RRPiOS, does it work if you manually call it in motioneyeOS?

starbasessd, 2022-07-24:
I did load a copy of your 'right.py' script into /data/etc in a dev20201026 on a PiZeroW and it didn't give any errors when run. I wasn't able to check it on the GPIO pins, though.

heikkivirtanen, 2022-07-24:
Thanks a lot for the help. The problem was actually in my second bullet point. I will edit the answer there so maybe it helps somebody else.