motioneye-project/motioneyeos #2917: AI Thinker ESP 32 + Mini Default Camera + MotionEyeOS - Intermittent connections

marcusobrien, 2022-06-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

I saw these posts that mention ESP32,  but they are both closed with no resolution :
https://github.com/motioneye-project/motioneyeos/issues/2513
https://github.com/motioneye-project/motioneyeos/issues/2507
https://github.com/motioneye-project/motioneyeos/issues/2493
https://github.com/motioneye-project/motioneyeos/issues/1952

### motionEyeOS Version

I am running motionEyeOS version: 
motionEye Version : 0.42.1
Motion Version : 4.3.1
OS Version : motionEyeOS dev20201026

### Board Model
Raspberry Pi3 B running MotionEyeOS

### Camera
I am using the following type of cameras 

Network Camera
Simple MJPEG Camera

My camera model are

1. A locally connected (to Pi 3 with Camera interface) - Night vision camera module for Raspberry Pi - 
https://www.pishop.ca/product/night-vision-camera-module-for-raspberry-pi/ 

2. A network camera - the AI Thinker ESP32 with Camera 
[image]

running firmware built from this repo
https://github.com/circuitrocks/ESP32-RTSP

As directed in the ESP 32 boot up serial output, I use the following URL for network setup in MotionEyeOS 

rtsp://192.168.0.137:8554/mjpeg/1

### Network Connection

My motionEyeOS unit is connected to the network via: Pi3 WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Power Supply for Pi 3 is official Pi Adapter 5 volt up to 2 Amp
Power Supply for is 5.1 volt up to 2 Amp

### Log Files

I consider the following log files relevant to this issue:

On the ESP32 the Serial/UART log shows the ESP 32 (and Camera) are connect sometimes, with the request for the stream hitting the right end point

17:05:37.934 -> WiFi connected with IP 192.168.0.199
17:05:37.934 -> Stream Link: rtsp://192.168.0.199:8554/mjpeg/1
17:05:37.934 -> 
17:05:37.934 -> RTSP task up and running
17:05:45.444 -> RTSP client started connection
17:05:45.444 -> Creating TSP streamer
17:05:45.444 -> Created streamer width=800, height=600
17:05:45.444 -> Creating RTSP session
17:05:46.527 -> RTSP received OPTIONS
17:05:47.572 -> RTSP received DESCRIBE
17:05:48.718 -> RTSP received SETUP
17:05:50.128 -> RTSP received PLAY

### Explanation
The Local Camera connected directly to the Pi 3 host running MotioneEyeOS works all the time. 
Occasionally the ESP 32 works and MEOS displays the picture. But most of the time MEOS displays 

"Unable to Open Video Device"

I can see the ESP32 local UART output is showing the stream request coming from MotionEyeOS

RTSP received OPTIONS

But most of the time this OPTIONS sending from MEOS is just repeated, its like the client (browser) is resetting the connection - or just re-trying rather than waiting. Any ideas on this ?

starbasessd, 2022-06-07:
2513 was using an esoteric environment, and the user wasn't willing to answer the questions asked. I was unable to re-create their environment as described. 
2507 was from a user with no experience with arduino to make sure the correct information was compiled for the RTPS functions. Another expert user was able to compilke and test as 'proof of concept' but it never went any further.
2493 went all over the board, ended up as a beginners tutorial. Ran into issues with encrypted RTSP output (unsupported in motionEyeOS or motionEye
1952 wasn't able to find a consistent answer, before my time.
The real issue is that some of the esp32 sketches don't follow the same standards as most of the RTSP cameras out there (or even the jpeg/mjpeg). I run several different branded cameras that all work (personal preference is FosCam).  
The best resource for working URLs I've found is: [here](https://www.ispyconnect.com/camera/esp32)
There is just too much variability between the versions various devs modify to work, and the ones that do work usually are following good standards for their output.

rradar, 2022-07-31:
I would give [esphome](esphome.io/) a try for the esp32cam as it has a pr mentioning support for motioneye:

https://github.com/esphome/esphome/pull/2777

> It appears that MotionEye is picky about HTTP protocol
and Connection: close. For non-chunked encoding it requires
to use HTTP/1.0 with a properly aligned boundaries of next
images.

marcusobrien, 2022-08-21:
Thanks for the feedback - I actually first switched my Pi 3 running the MotionEye server to using a LAN cable - just to up the available bandwidth (all other cameras use Wifi). I have the following working 

Local infra red camera working on the Pi 3 server https://www.amazon.ca/Infrared-Vision-Raspberry-Megapixel-Fisheye/dp/B07WMY1VQ6/ref=sr_1_1_sspa?keywords=Raspberry+Pi+3+Infrared&qid=1661048706&sr=8-1-spons&psc=1

MaixII Dock https://www.seeedstudio.com/Sipeed-MAIX-Dock-p-4815.html

Pi Zero W with OV2640

Pi Zero 2 W with OV2640

2 ESP 32 AI Thinker Cams

The ESP32 cams seem to be the most unreliable - I did get them working by dropping the streamed frames to 2 FPS, and no post processing i.e. scaling etc.

The best cameras are the Pi Zeros - if I could buy more of these I would, I can post process to the max with the Zero 2 W - just cant buy any at the moment !