motioneye-project/motioneyeos #2918: Manually set date and time, no internet connection

Lidbaum, 2022-06-07:
Is it possible to manually set date and time on motioneyeos? Cannot find it in gui, but can it be done from console? I have no internet 
connection and each time I power off the RPi it restarts at 1970-01-01. 

I assume an RTC module should work if I once connect to internet/NTP. Or setup a local NTP.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 4

### Camera

I am using the following type of camera: MMAL.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet but no Internet connection.

starbasessd, 2022-06-07:
BTW you would have to manually set the date & time every time it reboots, without a RTC module or a server on the Pi's subnet that can give it the time. I do it 2 different ways without a RTC module. (Be sure to set the correct time zone in Settings, General)
1st method is a Pi that has a GPS module that I use for a Chrony server to give time to any other computer or device on that network. In Settings, Expert Settings, Date Method, I select NTP Server and put in the IP address of the GPS Chrony Server.
2nd method is a web server that the Pi has access to (but no internet) that gets it's time from another source. In Settings, Expert Settings, Date Method, I select HTTP and put in the IP address of that web server.
Both work well for the retrieval of the correct time.

Lidbaum, 2022-06-08:
Excellent, date -s YYYY.MM.DD-HH:MM:SS was exactly what I needed!

closed 2022-06-08 by Lidbaum