motioneye-project/motioneyeos #2921: Add instructions to the wiki for turning off the camera LED

karlmanait, 2022-06-16:
The instructions defined [here](https://github.com/motioneye-project/motioneyeos/wiki/Turn-Off-LEDs) only pertains to the on-board LEDs of the raspberry pi.

I suggest also adding an instruction in turning off the camera module LED. Which is done by simply adding `disable_camera_led=1 `
in `/boot/config.txt`.

For context I'm using a Raspberry Pi Zero with this [camera module](https://www.aliexpress.com/item/32901059454.html) and it worked for me.

starbasessd, 2022-06-16:
Considering the shear number of CSI cameras available, and not all of them following the PiCam standard, it would be hard to include instructions for all of them. It is the same reason there isn't a list here for all the IP Camera URLs, all the variations of RTSP URLs, and users are referred to the [iSpy Connect database](https://www.ispyconnect.com/cameras)

karlmanait, 2022-06-16:
It doesn't have to be "work-for-all". I was thinking of something like "For standard Raspberry Pi cameras ...". But if you disagree, you can close this issue 😃