motioneye-project/motioneyeos #2922: Random MAC Address on Ethernet-Adapter leads to new IP address on every reboot

Level74, 2022-06-19:
### Preliminary Docs
I confirm that I have read the CONTRIBUTING guide before opening this issue.
I confirm that I have read the FAQ before opening this issue.

### motionEyeOS Version
I am running motionEyeOS version: 20200606

### Board Model
I am using the following board/model: Raspberry Pi Zero (V1.3)

### Camera
I am using the following type of camera: MMAL
My camera model is: Raspberry Pi Camera V2.1

### Network Connection
My motionEyeOS unit is connected to the network via: Ethernet (see board below)

### Peripherals
I am using the following peripherals that I consider relevant to this issue:
Pi Zero Enc28j60 Network Adapter Module
(also see issue at end of https://wiki.52pi.com/index.php/EP-0088)

### Log Files
I consider the following log files relevant to this issue:
[motioneye.log](https://github.com/motioneye-project/motioneyeos/files/8935440/motioneye.log)
[boot.log](https://github.com/motioneye-project/motioneyeos/files/8935441/boot.log)

**ISSUE**
Hello World ;-)

In the past I was using a "simple" WLAN Router Setup in my home network, where I temporarely added two Raspberry PI Zero's with the above listed GeeekPi network adapter (both connected via ethernet cable). At each reboot, the old router always assigned the same IP address to both RasPi's (probably based on their hostnames).

In the meantime I set up a complete new home network based on a Unifi Edge Router X (with VLAN, DHCP, etc.). Everything works fine but RasPi's (just trying with one yet): At every reboot, it gets assigned a new IP address. In the lease table of the router I noticed, that the RasPi always has a different MAC address, but the hostname remains the same (see image):
[image]

As I could find out, the used network adapter has no own MAC address. Instead, the MAC address is generated by the OS (not sure on that point, but it seems that the OS "scrambles" a MAC address using the S/N of the Pi, etc. at each startup).

As you can imagine, it's not the idea to have a new IP address each time I attach/boot the Pi. So I was looking for a solution since two days now, but didn't find out any that worked. I already tried the following:

- Assign a static IP on my Edge-Router: Doesn't work, because I can't map a static address based on the hostname, only. I must enter a MAC address as well (also tried with the same static IP on MEOS > I can't access it anymore via network, because it has fixed - let's say 10.XXX.XXX.56, but gets 10.XXX.XXX.61)

- Tried to figure out tweaks like this one, that describes how to stop the Pi to do MAC address randomization:
https://www.codegrepper.com/code-examples/whatever/mac+address+of+raspberry+pi+keeps+changing
Didn't work, because (sorry I'm quite new to MEOS, I assume, there is no "network manager" on MEOS)

- Also tried this one: https://wiki.52pi.com/index.php/EP-0088
I put that line in the config.txt with no result (not being sure, if it's the right place to do).

Any ideas on how to "fix" this? I know, it might not be a real "issue" of MOES, but any help to fix this would be great.

Many thanks!

starbasessd, 2022-06-19:
Unfortunately, mEOS / ThingOS / Buildroot options don't take your device into consideration. I use Belkin USB-Ethernet devices without issue. That said, you may want to consider using the Static_IP.conf method given [here](https://github.com/ccrisan/thingos/wiki/static_ip.conf) to set your ip for the Pis using that network module. I also prefer using reservations on a router/dhcp server but sometimes, the best laid plans of mice and humans...

Level74, 2022-06-20:
> Unfortunately, mEOS / ThingOS / Buildroot options don't take your device into consideration. I use Belkin USB-Ethernet devices without issue. That said, you may want to consider using the [...]

Thank for the reply and the tips. I tried with the Static_IP.conf, but it didn't work. The PI tries to connect to the router, but doesn't succeed, because the router expects the former MAC address to assign that IP. That leads to the (already known) problem, that the PI can't be accessed at all remotely.
Well, I will try with another ethernet adapter (or will set up the whole ME on the old router). The coll thin about the Geeekpi HAT is its compact format, but the downside is, that the mounted NIC misses MAC capapilities :-(

starbasessd, 2022-06-20:
Remove the reservation for the IP address from the router. Then try. You don't want to be creating your own conflict.

starbasessd, 2022-06-20:
Oh, and the file name needs to be all lower case. (static_ip.conf), My oops, the autocorrect sucks...

Level74, 2022-06-21:
Hurray! I got it! Many thanks again for your help.

As mentioned in my issue description, I first set the fixed ip via the MEOS web ui (several times, starating again and again). Apparently, the gateway address is somehow set wrong and sets the same address as the fixed ip? I couldn't see it, because after the reboot, I couldn't connect any more to the PI.

So this time, I set the fixed ip again via the web ui (same problem occoured of course again). After rebooting, I connected the PI to a monitor and started it up. The boot log wrote:
...
Configuring wired network: 10.XXX.XXX.62/24       >> OK, it's the fixed ip
Setting static gateway to 10.XXX.XXX.62: done      >> that's wrong, shoud be ending with .1
Setting static DNS server to 10.XXX.XXX.1: done    >> that's fine
Setting current date using SNTP: failed                  >> that was, what came into my eye ;-)
...

Having read your contribution and visited the site [linked](https://github.com/ccrisan/thingos/wiki/static_ip.conf) by you, I logged in via terminal directly on the PI, opend the the /data/etc/static_ip.conf and changed the gateway address to 10.XXX.XXX.1. Saved, closed and rebooted and voilà! (the date was set correctly).

However, the router doesn't show the device under the "leases" section, but that's OK. I have now set the fixed ip of the PI to an address outside of the DHCP range of that VLAN to prevent ip-conflicts. It still works and I can access the PI.

So many thanks again. Without that link, I never could find the location of the static_ip.conf file.

Best regards!

starbasessd, 2022-06-21:
You closed then reopened? 
Since the ip is no longer received from DHCP, it's not going to be shown in leases, it's manually set.
Do you have another issue?

Level74, 2022-06-21:
Definitely closed.