motioneye-project/motioneyeos #2923: 2022 Raspberry Pi 4 8gb - eternal bootloop

tfes-git, 2022-06-26:
Raspberry Pi 4 8gb (early 2022 version)

Have been trying to install Raspbian (works well) but can't get motioneye running on it, having followed the instructions on git...

Have tried installing MotioneyeOS, both dev20201026 and 20200606 adding the updated start4.elf and fixup4.dat to /boot before first boot.

Have tried both using wpa_supplicant for wifi and just ethernet.

Has anyone recently got RPi4 8gb working with MotionEye? whether 64bit or 32....

any useful thoughts?

starbasessd, 2022-06-26:
What bootloader version do you have? The command:
> vcgencmd bootloader_version
shows version installed, and anything after Apr 29 2021 has been shown to have issues.

starbasessd, 2022-06-26:
motionEyeOS is only 32 bit. You can install motionEye on 64 bit OSs if you don't want/need CSI cameras.

tfes-git, 2022-06-26:
> What bootloader version do you have? The command:
> 
> > vcgencmd bootloader_version
> 
> 
> 
> shows version installed, and anything after Apr 29 2021 has been shown to have issues.
> 
> 

Bootloader: c2f8c388 Apr 29 2021

Does that suggest I give up on this board altogether?

tfes-git, 2022-06-26:
> motionEyeOS is only 32 bit. You can install motionEye on 64 bit OSs if you don't want/need CSI cameras.
> 
> 

I thought it was 32 bit, but in desperation tried on raspbian 32 and 64 on several versions, tried flashing through etcher and the official RPi software...

starbasessd, 2022-06-26:
No, that is a GOOD version of the boot loader (on both my Pi4-4GB and Pi4-8GB).
As to motionEye on another OS, I have successfully installed in RaspiOS Buster and Bullseye, Debian Buster and Bullseye (Bullseye being the 64 bit versions) and I run it in PCs both iron and VM in 32 bit and 64 bit OSs.
Imaging from Windows, I use Win32DiskImager. From Linux I use the dd command. I never use the 'Raspi-Imager software anymore since the changes you can make (presetting options like Wifi and such) breaks specialty OSs like motionEyeOS.

tfes-git, 2022-06-26:
> No, that is a GOOD version of the boot loader (on both my Pi4-4GB and Pi4-8GB).
>
> As to motionEye on another OS, I have successfully installed in RaspiOS Buster and Bullseye, Debian Buster and [...]

I'll try the Win32DiskImager tonight then, and see how far I get- I've normally found Balena-Etcher pretty good when the official gets awkward.

The current message I'm stuck on is Start4.elf is not compatible. This board requires newer software. 

If you were to pick a single OS to try, which would you go for?  

Presumably no issue to ignore wpa supplicant and just stick with Ethernet too?

starbasessd, 2022-06-26:
If you need CSI camera, then RPiOS Legacy.
If you don't need a CSI Camera, then RPiOS Bullsyey 64 bit.
Ethernet is best.

The error confuses me, though. The RPi4 dev20201026 image doesn't contain a start4.elf file at all. I will look at it this afternoon, as I am tying up my RPi4-8GB with a test of WoR Win10 right now.

tfes-git, 2022-06-26:
No need for csi at the moment- should have a few ESP32cams and a few dahua ip cams and nothing particularly exotic there.

I'll try the bullseye 64 again, I think installing motioneye by the git instructions only throws an error/comment that python version is old, but I never seem to get the actual motioneye service running...  

Anywhere particular to find useful errors/logs?

starbasessd, 2022-06-26:
Depends on which instructions you are following.
> [RPiOS Bullseye Original](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye) warns of Python2 errors which can be ignored.

motionEye then runs a WebGUI at http://ip_address_of_pi:8765
You can look for it running at the CLI level by running 
> netstat -tulpn | grep 8765

which should report:
> tcp        0      0 0.0.0.0:8765            0.0.0.0:*               LISTEN

starbasessd, 2022-06-26:
Between issues, I imaged a USB drive with [RPi4-dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz), changed the section of /boot/cmdline.txt from root=/dev/mmcblk0p2 to root=/dev/sda2, saved and inserted into the Pi4-8GB. It booted and set up normally. I ssh'd in, ran command 
> vcgencmd bootloader_version

and it shows:
[root@meye-57b21018 ~]# vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
[root@meye-57b21018 ~]# free
              total        used        free      shared  buff/cache   available
Mem:        8011700       65900     7878524        1204       67276     7881316
Swap:             0           0           0
[root@meye-57b21018 ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    190.7M      7.3M  96% /
devtmpfs                  3.8G         0      3.8G   0% /dev
tmpfs                     3.8G     16.0K      3.8G   0% /dev/shm
tmpfs                     3.8G      1.2M      3.8G   0% /tmp
/dev/sda3                27.9G     46.4M     26.4G   0% /data
/dev/sda1                29.9M     12.6M     17.3M  42% /boot
/dev/sda3                27.9G     46.4M     26.4G   0% /home/ftp/sdcard
/dev/sda3                27.9G     46.4M     26.4G   0% /home/ftp/storage
overlay                  27.9G     46.4M     26.4G   0% /usr
overlay                  27.9G     46.4M     26.4G   0% /var/log
overlay                  27.9G     46.4M     26.4G   0% /var/lib
/dev/sda3                27.9G     46.4M     26.4G   0% /data/media/sda3
/dev/sda3                27.9G     46.4M     26.4G   0% /home/ftp/storage/sda3

It can and does work.

tfes-git, 2022-06-26:
apologies for delay after your rapid assistance earlier - son needed attention!

I have no idea what I have been doing for the past 24 hours... I have just installed the Bullseye 64 bit with desktop, ran through the Git instructions for install on Bullseye, and first time trying [ip]:8765 from another machine and the admin page is there... 

I think over tired and mixing instructions/OS got me in a complete knot, and googling unrelated issues I properly messed up!

Many thanks, seems I'm up and running!

gitberry, 2022-08-09:
I too am having trouble getting it running on a RPi4B with 8GB - I followed instructions from @starbasessd in this (https://github.com/motioneye-project/motioneyeos/issues/1969) to ensure hardware was all up to date.  Was still giving me this: 

[image]

So then I followed the advice of getting the dev release - and now it's not giving anything (ie blank).  I then tried @starbasessd advice on the cmdline.txt of June 26, 2022 post - and same blank.  

 @tfes-git do you have a link to how you installed Bullseye 64?  Many Thanks if you do!

starbasessd, 2022-08-09:
There isn't a start4.elf or fixup4.dat in dev20201026, so it sound like you are not using [this image dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz).
Depending on your skills, you can copy the start.elf, initrd.gz, fixup.dat, and bcm2711.dtb files from Legacy Buster 32 bit to the /boot partition after a fresh image. If you had previously updated the bootloader to most current, then pushed it back to the 4/2021 version, it will not allow it to work.
As to Bullseye 64, you would also need to install Python2.7 and pip2.7, and they are having issues. Ditto with Bullseye 32...

gitberry, 2022-08-09:
> There isn't a start4.elf or fixup4.dat in dev20201026, so it sound like you are not using [this image [...]

thanks @starbasessd - what you say makes sense, since the screenshot was from the latest, not the dev (which wouldn’t display anything). I won’t be able to attend to this for 24 hours but was wondering if you had recommended links to instructions on how you roll back hardware to specific versions - that’ll be the first thing I’ll do.. I’m pretty new at RPi images and whatnot…was using official tool - Will try the other ones that were recommended…

starbasessd, 2022-08-09:
For motionEyeOS, you can't and have either 20200606 or dev20201026 work out of the box. Something about the boot loaders after 4/2021 broke the RPi4 images. . .

To load a specific boot loader, here are the instructions: (for example, 2021-04-29)
Use a new, clean install of RPiOS on a SDCard or USB drive and run the command:
> sudo vcgencmd bootloader_version

Mine reports:
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1645126804
capabilities 0x0000001f

If it reports anything before 2021-04-29, you can run:
> sudo rpi-eeprom-update -f /lib/firmware/raspberrypi/bootloader/critical/pieeprom-2021-04-29.bin

If you do a 
> ls -la /lib/firmware/raspberrypi/bootloader/

you'll find all the 'supported' versions of the bootloader that RPiFoundation programmers support. You'll notice the 'default' folder points to 'critical, and 'latest' points to 'stable'. I don't recommend using the 'beta' versions unless they fix a known issue, or have a new feature you want to test.

If your RPi had ever been updated to something after2021-04-29, you can roll it back, but it won't work until the Dev Team creates a new image. 
Dev is the only one I've ever gotten to work on any Pi4. 

BTW, I only boot USB drives on the Pi4s with RPiOS Buster, Bullseye 32 or 64, or motionEyeOS (takes a minor change to cmdline.txt in motionEyeOS) and have better speed and reliability.

Solarbot, 2023-07-04:
Hi starbasessd, please forgive me as this might be a very basic question :-)  I've just bought a RPi4 and RPiV3 camera with a view to running MotionEyeOS and I didn't realise that this might not be the path of least resitance!  So starting from where I am..... I can see you have a similar setup running which is encouraging.  I have rolled back the bootloader to Apr 29 2021 17:11:25 but now a little stuck on the steps I need to take to complete the patch/hack/fix?  I have dev20201026 ready to roll on one SD card and Raspberry PI OS (32 bit) installed and running on another SD card to make the required changes..... I just don't know what those are - any chance of some pointers to get MotionEyeOS dev20201026 running on this new RPi4?

starbasessd, 2023-07-04:
@solarbot As noted in several threads, rolling back the bootloader doesn't usually fix the boot loader issue, as some other parts of the eeprom are also modded and prevent the system from working with motionEyeOS.
Using a PiCam3 is not supported, as you need to change / update a bunch of other files including DTB and the kernel manually because PiCamV3 didn't exist in 2020. 
Since you have a Pi4-4GB and a PiCamV3, I suggest getting the RPiOS Buster Legacy Lite and installing the motionEye from the [instructions here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) as this works with your setup. You need to enable the CSI interface in raspi-config, though, don't forget.

Solarbot, 2023-07-05:
@starbasessd  Many thanks, great to know its possible :-)  I've now downloaded and installed **2023-05-03-raspios-buster-armhf-lite.img** and then followed the instructions you have kindly written and linked to.  Again this will be a noobie type question so applogies: when I enter the command:  `wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb` I get the error in the image attached and I don't know what I'm doing wrong?  
 
[image]

Solarbot, 2023-07-05:
I'm trying again as I think my distribution is the wrong version - will report back.... it will probably work with the right version :-)

starbasessd, 2023-07-05:
Under Before Proceeding:
>   Line 3: All commands require root; use sudo before each command or become root using sudo -i.

None of the following were installed in Step 1 (probably forgot to use the above instruction)
>   apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y

starbasessd, 2023-07-05:
FYI, that is the correct distro to use "2023-05-03-raspios-buster-armhf-lite.img "

Solarbot, 2023-07-05:
@starbasessd Thanks - I used sudo -i to enter root so hopefully that bit was done right.  I've now reverted to "2023-05-03-raspios-buster-armhf-lite.img" and run through the process again.  This first command ` apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y` mostly completes correctly but there seem to be some parts missing from the remote server?

`E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libw/libwebp/libwebp6_0.6.1-2+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libw/libwebp/libwebpmux3_0.6.1-2+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libx/libx11/libx11-xcb1_1.6.7-1+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/mariadb-common_10.3.38-0+deb10u1_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/libmariadb3_10.3.38-0+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]`

Solarbot, 2023-07-05:
I'm in the UK

Solarbot, 2023-07-05:
thanks for checking :-)

starbasessd, 2023-07-05:
32 GB sdcard
image: 2023-05-03-raspios-buster-armhf-lite
win32diskimager (not etcher or pi imager)
install in pi, power on, wait for reboot
>sudo apt update && sudo apt upgrade -y

38 updates
>sudo -i
apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y

I was able to get 154 pkgs and they installed correctly
Looks like your system picked a bad random mirror. Try running the install command again (repeat) until it gets them all...
> wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb
 dpkg -i pi_buster_motion_4.3.2-1_armhf.deb

grabbed successfully
>dpkg -i pi_buster_motion_4.3.2-1_armhf.deb

Installed successfully

Solarbot, 2023-07-05:
Very cool - MotionEye is now up and running :-)  Not sure what made the difference but it did.  Now I just have to get the camera working as it is showing "unable to open video device"?  I have it set to LOCAL V4L2 and MMAL SERVICE 16.1 currently but have tired the other options for the LOCAL V4L2 as well with the same result?  Very close :-)

Solarbot, 2023-07-05:
The camera seems to be detected:
vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0

starbasessd, 2023-07-05:
Did you enable the camera in 
>sudo raspi-config
Interface options
Camera

???
What result do you get when you run the command:
> sudo vcgencmd get_camera

???

Solarbot, 2023-07-05:
The camera is enabled and I just tried rebooting but no change - the camera seems to be registered but with no interfaces?

```
sudo vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0
```

starbasessd, 2023-07-05:
Interesting. The camera is showing, and would be found automatically.
did python-pil complete successfully, or did it quit at 99% and go to a prompt without installing anything else?
Just to be sure, run the following as root:
>apt-get install python-pil -y 
>pip install motioneye
>systemctl restart motioneye

then re-connect to the web interface. If not there, reboot and check again.

Solarbot, 2023-07-05:
Thanks - i think the python install went to plan but I followed your instructions above to be sure but with no joy including a full power down reboot but no change.  I'm not sure if this helps but if I type `libcamera-hello` into the terminal I get 'command not found'?

starbasessd, 2023-07-05:
That would be expected, as you should have installed Buster. That command is in Bullseye, and breaks the Python 2.7 wrapper.
The command you should try is:
>raspistill -o Desktop/image.jpg

starbasessd, 2023-07-05:
or correction, I cut and pasted, didn't edit for a good path:
>raspistill -o ./image.jpg

starbasessd, 2023-07-06:
Interesting. Did you run 
>sudo apt update && sudo apt upgrade 

??? I've never seen those kinds of errors with the cameras I have.
I just did a quick google search  using [raspberry pi os buster imx708](https://forums.pimoroni.com/t/is-it-possible-to-get-a-pi-camera-3-working-with-raspberry-pi-4-with-buster-os/21212)
Looks like the camera isn't supported in raspicam (raspistill etc).

Solarbot, 2023-07-06:
Thanks again.  I just tried `sudo apt update && sudo apt upgrade` but alas no change.  I think I'm going to have to get a V2 camera to resolve this?  I will still be happy having it run on the RPi4 and grateful for the help you have given me - THANK YOU :-)