motioneye-project/motioneyeos #2925: Raspberry Pi 4 çGB Model B

merlincom, 2022-07-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2022-07-08:
Which bootloader is in your Pi?
> vcgencmd bootloader_version
known good: Apr 29 2021 17:11:25
try dev20201026: https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz

merlincom, 2022-07-08:
Not working :-(
boot: mode SD 1 order f41
Firmware: xxxxxxx   Jul 17 2020 11:00:29
start.elf: is it not compatible

starbasessd, 2022-07-08:
Need to update your bootloader to Apr 29 2021 17:11:25 
Updating to the latest will mess up booting motionEyeOS until they update the kernel and system.
I just imaged an SDCard with dev20201026 and was successful.
Be careful, you want to look for instructions to install a specific bootloader, not do the latest update.

merlincom, 2022-07-09:
Sorry - I don't see the Option "Beta Test Bootloader" in my Raspberry Pi Imager V.1.7.2. This is the same on Windows 10 64 bit an Raspberry Pi 4 with 64bit OS. I do not find a other solution and as you said: Updating to the latest will mess up booting motionEyeOS

merlincom, 2022-07-09:
motioneyeos-raspberrypi4-dev20201026.img.xz don't work even with bootloader Version 2021-04-29
Same error in my 
`Raspberri Pi Model B  - 4GB`
`bootloader: c2f8c388 Apr 29 2001`
`update-ts: 1657353958`
...
`fw: start.elf fixup.dat`
...
`start.elf: is not compatible`

starbasessd, 2022-07-09:
The instructions you used to update were correct.
My 2 Pi4s:
- - - - -
pi@raspi1:~ $ vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1645126804
capabilities 0x0000001f
- - - - -
pi@raspi5:~ $ vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1621697261
capabilities 0x0000001f
- - - - -
Both work with dev20201026, both boot from SDCard and USB for RaspiOS and motionEyeOS (with modification for USB)