motioneye-project/motioneyeos #2926: No view image After reboot

salvatoremaietta89, 2022-07-13:
Hi,
please help me. After a blackout mu server goes down and After reboot motioneye show the cameras as attached image
[image]

Id i delete and re-add cameras the issue persist.

Thanks a lot
Salvatore

starbasessd, 2022-07-13:
Which versions of motion, motionEye, and OS? (settings, general)
Did you reboot the cameras?

salvatoremaietta89, 2022-07-14:
Hi,
first of all thanks for u're support.
OS version is 2022.7.4 and motioneye version is 0.18.0
Yes i have rebooted camera too but issue persist

Thanks
Salvatore