motioneye-project/motioneyeos #2927: start.elf is not compatible (RPi 4B 4GB)

blacc01, 2022-07-14:
Hi,

I read several posts about this issue, but for me it does not work, i got the folowing errors:
[image]
[image]

I downgraded bootloader to the following, which should be ok as i know:
~~~
pi@indi-allsky:~ $ vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1657788958
capabilities 0x0000001f
~~~

What i had done.
1. Downgraded bootloder to Apr 29 2021 version
2. Downloaded `motioneyeos-raspberrypi4-dev20201026.img.xz` from [https://github.com/motioneye-project/motioneyeos/releases/tag/nightly-dev](url).
3. Created with the downloaded xx.img.xz file an SD card with the Raspberry Pi Imager:
[image]
4. Inserted to the Raspberry Pi 4B 4GB
5. And get the error screens above...

Please help what i'm doing wrong, or what should i do? Or is there another way to get the MotionEye software work on my Pi?

Thanks!

blacc01, 2022-07-14:
At point 3., i also tried to first unzip the xz file, and mounted the img file to the sd card using Balena Etcher software. But the result is the same, start.elf is not compatible. Please help!

starbasessd, 2022-07-14:
Unfortunately, if you had a newer bootloader than 29/4/2021, then downgraded, it will not work with motionEyeOS. You will have to wait until the new Devs update motionEyeOS and the underlying buildroot/ThingOS. This has been confirmed several times, including myself.
There are 2 options until they update: Build your own image to update the motionEyeOS system, or switch to RPiOS Legacy Lite, and install motionEye from [here](https://github.com/motioneye-project/motioneye/wiki/Installation)

starbasessd, 2022-07-14:
They are working on motionEye first. They need to find a dev who understands buildroot/ThingOS and brings it into this environment.

blacc01, 2022-07-14:
@starbasessd thanks
In the meantime is created a new lite RPiOS SD card, and installed everything as described here: [https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye](url)

No errors during install, service is running:
~~~
root@motioneye:~# systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-07-14 15:29:41 CEST; 5s ago
   Main PID: 9741 (meyectl)
      Tasks: 5 (limit: 4915)
        CPU: 1.268s
     CGroup: /system.slice/motioneye.service
             ├─9741 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             └─9751 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Jul 14 15:29:41 motioneye systemd[1]: Started motionEye Server.
Jul 14 15:29:41 motioneye meyectl[9741]:     INFO: hello! this is motionEye server 0.42.1
Jul 14 15:29:42 motioneye meyectl[9741]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jul 14 15:29:42 motioneye meyectl[9741]:     INFO: cleanup started
Jul 14 15:29:42 motioneye meyectl[9741]:     INFO: wsswitch started
Jul 14 15:29:42 motioneye meyectl[9741]:     INFO: tasks started
Jul 14 15:29:42 motioneye meyectl[9741]:     INFO: mjpg client garbage collector started
Jul 14 15:29:42 motioneye meyectl[9741]:     INFO: server started
root@motioneye:~#
~~~

But now how to reach it? When i add the IP of the PI i got only 404, no page, nothing at all.

blacc01, 2022-07-14:
I have it, on port 8765...