motioneye-project/motioneyeos #2929: Authentication using just the 7 first characters via SSH / CLI

MPBatista, 2022-07-25:
### motionEyeOS Version

I am running motionEyeOS version: motioneyeOS 20200606.

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: MMAL.

My camera model is: vc.ril.camera

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * None

### Log Files

I consider the following log files relevant to this issue:

 * None

The issue I'd like to describe is that:
-  when I connect to MotioneyeOS via SSH, CLI, try as password :
    
admin@192.168.0.52's password: L!ttl&T
Permission denied, please try again.
admin@192.168.0.52's password: L!ttl&T35701
[root@cam-back ~]# exit
logout
Connection to 192.168.0.52 closed.

But the real password is L!ttl&Te5701, it look like the auth logic just treat 7 first characters.

Could you confirm this?

starbasessd, 2022-07-25:
Try without any password (blank)

MPBatista, 2022-07-25:
With blank password or just ENTER  it not working:
ssh admin@192.168.0.52
Welcome to cam-back!
admin@192.168.0.52's password: <blank>
Permission denied, please try again.
admin@192.168.0.52's password: <ENTER>
Permission denied, please try again.
admin@192.168.0.52's password:

starbasessd, 2022-07-25:
Can you try a password without the '&' character, please?

MPBatista, 2022-07-26:
I changed the password via CLI for L!ttleTe5701:
ssh admin@192.168.0.52
Welcome to cam-back!
admin@192.168.0.52's password: L!ttle
Permission denied, please try again.
admin@192.168.0.52's password: L!ttleT
Permission denied, please try again.
admin@192.168.0.52's password: L!ttleTe
[root@cam-back ~]#