motioneye-project/motioneyeos #2930: Webhook to local adress - edit hosts file?

Schdefoon, 2022-07-30:
Hi,

I try to start a webhook notification to a local url. The DNS doesn't know where it can find my adress so i get

`ERROR: failed to call webhook: <urlopen error [Errno -2] Name or service not known>`

I couldn't manage to insert a static IP with my local DNS without losing time (1970-01-01). And without time the motion detection doesn't seem to work.
So I would like to edit the hosts file. But there only is a file /etc/hosts@
After editing this file it is losing my entries after reboot.

Any ideas?

Best regards

starbasessd, 2022-07-30:
/etc/hosts points to /tmp/hosts:

> -rw-r--r--    1 root     root           394 Oct 25  2020 group
> lrwxrwxrwx    1 root     root            10 Oct 25  2020 hosts -> /tmp/hosts
> drwxr-xr-x    2 root     root          1024 Oct 25  2020 init.d/

the file is destroyed / recreated every reboot.
The loss of DNS resolution is usually caused by a bad entry for DNS server. Instead of using a name in the webhook, have you tried an IP address? 
What URL are you using for the webhook?
Can you post the motion and motioneye logs? (Settings, Expert)

starbasessd, 2022-07-30:
Another command to try is 

> nslookup (servername)

for example:

> nslookup www.google.com

this will report what DNS server you are using, authoritative answer, and non-authoritative answers:
- - - - -
> nslookup camera-server
> Server:         192.168.x.y
> Address:        192.168.x.y:53
> 
> Name:   camera-server
> Address: 192.168.x.z
> 
> Non-authoritative answer:

- - - - -
If you don't get something like this, something isn't set up correctly.

starbasessd, 2022-07-30:
As to the time/date issue, if you have an internal web server, you can use it as your time source. I have another motionEye server on another machine. I set the Settings, Expert, Date Method to HTTP, and the Date HTTP Host to <ip_address>:8765, and it retrieved the correct time and date from it. I tried a number of others I have, including WebMIN (<ip_address>:10000) and regular web services, and they all worked.

Schdefoon, 2022-07-31:
Thank you very much for your help!
I have to admit that my understandings of network settings are not very good. So I think you're right that there might be something wrong in my DNS settings. I tried again to insert values in the network settings to use "static IP" and had the result
`bad address google.com`
when inserting the command
`ping google.com`

Here are my example settings when I use a client in my network (I always use static addresses)
- IP-address: 192.168.1.201
- Network mask: 255.255.255.0
- Gateway: 192.168.1.1 (my FritzBox)
- DNS: 192.168.1.1 (usually)

Now I have a server that can only reached inside my network. It can not be reached from internet. There are several services running on it so you can't reach the individual services by the IP address. You have to use domain names like pixelfed.local or funkwhale.local. The server has the IP 192.168.1.151
When I use a client where I can't manipulate the hosts file without problems (modern Samsung devices) then I insert the server IP as DNS. So when I switch DNS from 192.168.1.1 to 192.168.1.151 the device finds pixelfed,local

One service on my server is pihole. I found that pihole is changing entries on /etc/hosts on my server. So I edited this file and inserted entries that pixelfed and funkwhale can be found on 192.168.1.151.
So far I believed this is the reason that clients in my network can find the server even if the hosts file is not edited.

But on my MotionEye RaspberryPi I get "bad addresses" with DNS 192.168.1.1 or 192.168.1.151

Only when I change settings to DHCP I get a response for ping google.com
When I use nslookup google.com with this setting, I get
- Server: 192.168.1.1
- Address: 192.168.1.1:53

starbasessd, 2022-07-31:
PiHole is capable of being a DHCP server and DNSserver.
I would turn off DHCP on the Fritzbox, and turn it on on PiHole, that way PiHole knows the local names of the local machines, as well as the internet machines, and that the first DNS server PiHole looks at is the 192.168.1.1. 
For example, since my router (your fritzbox) resolves external IP addresses correctly, you could set Settings, DNS Tab, Custom 1 to 192.168.1.1, and edit the /etc/hosts file on the PiHole to include your internal machines like motionEye. 
You could turn on DHCP on PiHole to issue IP addresses, and static IP addresses like motionEye. 
You want DHCP to issue the PiHole as your primary DNS server (or the fritzbox)
Gets complicated when you have multiple servers that handle DHCP and DNS...

starbasessd, 2022-07-31:
With nslookup command, you can run
> nslookup <servername> [pihole_ip_address]

and
> nslookup <servername> [192.168.1.1]

(might have to include the .local extension on servername)
This will tell you which/if/either DNS server gives the correct answer.
You can then test with www.google.com in both tests.
You can use "https://fuzzthepiguy.tech/adtest/" to see if your PiHole is working right with a browser.

It took me years to learn how to set these things up (DHCP, DNS, and other services) to all work correctly and play nicely together.

Schdefoon, 2022-07-31:
Thank you very much. I will have a closer look at my system to make this right.
But to keep it simple for me to understand: The settings of my FritzBox are older than my local server is. So at the moment I have the choice to connect my devices to my FritzBox or I can take a little detour over my local server. The FritzBox doesn't know that there might be a Pihole in between.

So if I connect my MotionEye with the FritBox (DNS 192.168.1.1) and select static IP then I would expect to get an internet connection in general. But even this is failing. I only seem to have an internet connection with MotionEye when selecting DHCP in the settings.

starbasessd, 2022-07-31:
To solve the time setting issue, use your PiHole, for the motionEyeOS:
Settings, Expert,
Date Method: HTTP, 
and the 
Date HTTP Host: \<ip_address_of_PiHole\> 
or
Date HTTP Host: 192.168.1.1  ## your fritzbox, assuming it has a Web interface ##
When this is working, we can try fixing your DNS issues

Schdefoon, 2022-07-31:
Thank you - jumped on the first step 😄 
IP Configuration: Manual (Static IP)
IP Address: 192.168.1.201
Network Mask: 255.255.255.0
Default Gateway: 192.168.1.1
DNS Server: 192.168.1.1

Date Method: HTTP
Date HTTP Host: 192.168.1.1

I have the correct time.
But I still have no internet connection:
ping google.com -> bad address...

starbasessd, 2022-07-31:
What is your fritzbox using for DNS?

Schdefoon, 2022-07-31:
I have no special DNS settings in my box. In the settings is selected:
Use DNSv4 assigned by internet provider (recommended)

starbasessd, 2022-07-31:
OK.
assuming the pihole machine is 192.168.1.151, and you have funkwhale and pixelfed in its /etc/hosts file,
on the motionEyeOS, run the command:
> nslookup pixelfed.local 192.168.1.151

and 
> nslookup funkwhale.local 192.168.1.151

and 
> nslookup www.google.com

Do they report correct IP addresses?

Schdefoon, 2022-07-31:
Oh, ok...
They do.
I think the last "Non-authoritative answer" is IPv6? I have no IPv6 for my server so the last address for funkwhale and pixelfed is ::1. But the first ones are correct 192.168.1.151
Even Googles addresses are correct (142.250.186.132)

ping 142.250.186.132 delivers answers but ping www.google.com not

starbasessd, 2022-07-31:
Does motionEyeOS need to see the internet? It shouldn't.
In the DHCP settings on fritzbox, the DNS it hands out should point to pihole (192.168.1.151)
PiHole Settings, DNS, none of the upstrream services should be checked, and your Custom 1 should point to 192.168.1.1
All the machines should then be able to ping the other local machines, and the outside world.

Schdefoon, 2022-07-31:
Thank you very much for you patient help!
It's somehow crazy. I'm about to give up - at least for this weekend....
Unfortunately it did not work.
I can see that some devices that have the IP of the FritzBox as DNS now can be found in the query log of pihole - this should be a logic consequence of the settings.
But this time I get
`ERROR: failed to call webhook: <urlopen error [Errno -3] Temporary failure in name resolution>`

And nslookup funkwhale.local 192.168.1.1 delivers server can't find funkwhale.local

starbasessd, 2022-07-31:
Does 
> nslookup funkwhale.local 192.168.1.151 

work? If it does, then your DHCP server needs to hand out the PiHole as the DNS server, not 192.168.1.1

There is an alternate solution:
turn off DHCP service on the Fritzbox, and turn it on in PiHole. You can use the same settings in the DHCP server in PiHole that you used in the Fritzbox. You can add the same 'static' IP addresses, too. The PiHole would hand out IP addresses, and have the DNS Server set correctly.

Schdefoon, 2022-07-31:
Thank you very very much!
I will try next week.
Maybe I will start with the quick solution: Edit /tmp/hosts and not rebooting
Works as expected 🤣

Schdefoon, 2022-08-05:
There are some disadvantages for me to generally use my pihole app on my Raspberry Pi for all devices in my network. It's better for me to have the choice which device is using pihole and which one not.

So I decided to enable debugging in the expert settings of MotionEye and edited the file /etc/init.d/S04hostname
I changed the line
`echo "127.0.0.1 localhost ${hostname}" > /etc/hosts`
into
`echo -e "127.0.0.1 localhost ${hostname}\n192.168.2.151 pixelfed.local" > /etc/hosts`

Now my file /tmp/hosts has an additional line with the IP and the domain name even after a reboot

Thak you very much for your help. I really appreciate this.

Best regards

closed 2022-08-05 by Schdefoon

starbasessd, 2022-08-05:
I use a function to have certain devices (a Nintendo Switch and iPad used for streaming services) to bypass my PiHole. It's in PiHole, under Group Managemant. I created 2 groups, Default, and Bypass, and for the Bypass group It doesn't use any ad lists. This way, all devices still point to PiHole, but a couple don't get filtered.