motioneye-project/motioneyeos #2933: Record for longer after motion stops motion detection

sts-ryan-holton, 2022-08-18:
Hi,

I'm trying out the motion detection feature. I'm trying to optimise how quickly recordings start along with how much is captured in my videos.

I came across an article explaining what each motion setting does, and I think I need to change either the motion gap or captured frames after/before.

Basically, right now, some recordings that I've got are less than a second long, it sees motion in the frame and records and then stops, whereas others, for instance, if a person is walking past it'll record for longer, such as a minute.

What I want, I don't know what setting to change of the value to put it at, but when motion stops, I want it to continue recording for X seconds, let's say 60 seconds as an example, then, if any new motion is detected straight after the 60 seconds, start a new motion recording. What setting would I need to set here in my settings, I'll attach screenshots.

[image]

[image]

[image]

starbasessd, 2022-08-18:
You would want the Video Device Frame Rate times the number of seconds you want to record for the Captured After. The Captured After means how many frames to record after the motion trigger. Motion Gap in your case should probably be at 1 (1 second between end of recording and start of next event)
So, in your case: 30FPS x 60 seconds is going to be 1800, but there seems to be about a 1000 frames limit, so you'll get just over 30 seconds).
Your current numbers would give you 1/3 of a second recording and wait 2 seconds before the next one would begin...

sts-ryan-holton, 2022-08-18:
@starbasessd Thanks for this, I'll give it a go later 👍

sts-ryan-holton, 2022-08-18:
Working much better now! thanks @starbasessd

closed 2022-08-18 by sts-ryan-holton