motioneye-project/motioneyeos #2934: Improve night time video via brightness & contrast on USB camera

sts-ryan-holton, 2022-08-18:
I'm using a USB camera plugged into my Raspberry Pi 4, running motioneyeOS on the same board that the camera is plugged in to. I know if I were using the Raspberry Pi camera through the ribbon cable I'd be able to change the ISO settings, but I can't seem to make the brightness / contrast settings make any effect on my video?

I'm trying to achieve a slightly brighter image for night video than what I currently have as I want to see a bit more in the image, I know I won't be able to get it super bright, that's not my aim here, but if I can bring some more detail to night video recorded in any shape that would be fantastic, I know resolution has a big factor here, but I can't really go larger as the recorded video will be very "laggy".

What can I change here?
[image]

starbasessd, 2022-08-18:
You can try in the "Extra Motion Options box:
https://motion-project.github.io/motion_config.html#video_params
and use these controls to adjust brightness and contrast...

sts-ryan-holton, 2022-08-19:
@starbasessd the options don't seem to make any difference in my testing, I tried adding four different variations, neither resulted in any change of the video image through the dashboard or live stream:

```
video_params brightness=125
video_params ID09963776=125
video_params brightness=30
video_params ID09963776=30
```

starbasessd, 2022-08-19:
I don't know. I do know what your next question was going to be, though, 'how to switch between day and night settings...' That might be easier to resolve.
Maybe a new Dev, or @jasaw might be able to help...