motioneye-project/motioneyeos #2936: Action Buttons - An error occurred. Refreshing is recommended.

Fillip38, 2022-09-02:
### motionEyeOS Version

motionEye Version | 0.42
Motion Version | 4.2.2
OS Version | Ubuntu 18.10

### Board Model

Debian (Docker)

Hello,
i have running motionEyeOS on a Debain System and Docker. I wanted to add action buttons for PAN / Tilt to a camera, that worked, but I get the error "An error occurred. Refreshing is recommended." when I activate it in the browser. The individual files (up_4, down_4, ...) can be executed in the terminal without errors...

And where can i find the logs in a Docker System of motionEye...? I don´t have an "log" folder under /var

starbasessd, 2022-09-02:
Which docker install instructions for motionEye did you use?
Which Docker? docker-ce or docker.io?
Where are you putting the up_4 file?
What command are you running to start motionEye? 
What directories are you attaching from the host to the container?

Fillip38, 2022-09-02:
I use the install instruction from https://github.com/motioneye-project/motioneye/wiki/Install-In-Docker at docker.io
the files (up_4,..) are in the same folder like the camera-4.conf and motioneye.conf files, i can see the buttons in the Cam image on motioneye... 

I've been using motioneye for a long time, but now I have a camera with PAN / Tilt and would like to integrate it into motioneye

The directories are volumes...

And i have found the log file (motion.log) but it contains no errors as far as that is concerned

starbasessd, 2022-09-02:
I am trying to reproduce the error.
What command are you running to start motionEye?
What directories are you attaching from the host to the container? (will/should be in your docker run command)

Fillip38, 2022-09-02:
I´m using Portainer to start motioneye, I don't know exactly which command motioneye starts with.

[image]

starbasessd, 2022-09-02:
If you look, there are many issues reported when using portainer in the issues for motionEye github (BTW, this is the motionEyeOS github you want https://github.com/motioneye-project/motioneye/issues/).  I would suggest using a direct start command.

Fillip38, 2022-09-02:
Oh okey... 

As I said, motionEye runs without any problems for me, until now this...

What do you mean by "direct start command". ?

Fillip38, 2022-09-02:
I have post this to the "correct" GitHub

https://github.com/motioneye-project/motioneye/issues/2587

starbasessd, 2022-09-02:
The command is found in the Docker Install instructions [here](https://github.com/motioneye-project/motioneye/wiki/Install-In-Docker):
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-amd64

(If running on an SBC/arm processor you would change amd64 for armhf)
The issue you are running into may be related to something that portainer is involved with.

Fillip38, 2022-09-03:
Yesterday evening I created and started a second Motioneye docker via the console, but I get the same error when I press the action button in the web front end... 

I also wonder why there is no error in the log

starbasessd, 2022-09-03:
If you create 2 containers using the same ports, you will have all kinds of conflicts if they are both running at the same time.

Fillip38, 2022-09-03:
No, I changed the port (8764:8765), double ports don't work either

starbasessd, 2022-09-03:
I created a docker (docker.io, [standard](https://github.com/motioneye-project/motioneye/wiki/Install-In-Docker) install) on Debian host, and my action buttons work, no issues. Can you post one of your scripts, please, that works if called from console, but doesn't work as button, and post the ownership and permissions on the file (root:root rwxr-xr-x)?

Fillip38, 2022-09-03:
The owner an permissions are 
`-rwxrwxrwx 1 root root     89 Sep  2 20:03 down_4 (I did a chmod 777 before "desperation")`

The content of the file is a simple curl command to get a URL...
`curl -u admin:password http://192.168.199.120/cgi-bin/hi3510/ytdown.cgi > /dev/null`
This works in the terminal correctly

starbasessd, 2022-09-03:
Your first line in the script should be:
- - - - -
#!/bin/bash
curl -u admin:password http://192.168.199.120/cgi-bin/hi3510/ytdown.cgi > /dev/null
- - - - -
with your command on the second line.

Fillip38, 2022-09-03:
I had that at the beginning, but then I get the following error when trying to run this in the console

`bash: ./down_4: /bin/bash^M: bad interpreter: No such file or directory`

Fillip38, 2022-09-03:
Ahhhh its work now!!!

Yesterday I created the files with notepad++ and loaded them onto the Docker container, the file must have been created incorrectly somehow. Now I created the file using the terminal and it works! Thank you for your hard work and help

closed 2022-09-03 by Fillip38

starbasessd, 2022-09-03:
I saw in the error message an extraneous ^M but I see you found it and fixed it. Glad to help.