motioneye-project/motioneyeos #2938: See nothing at night

mao1mao2, 2022-09-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

mao1mao2, 2022-09-06:
Hi expert,

I just started trying MotionEyeOS with RPi HD Camera; it is superb!
But I encounter one issue: day time, everything is perfect; night time, almost nothing can be seen. If I point cam to a road lamp the light is so dull. I found no way to set ISO or sensitivity. Finally searched out night compensation parameter but it is deprecated.
What shall I do? Any suggestion?
Thank you !!

starbasessd, 2022-09-06:
You don't say which version of motionEyeOS, or which Pi.
The standard RPi HQ camera doesn't have an IR light source, and depending on which version of the camera you have, may have an IR filter, which prevents it from working well at night.
Most cameras for RPis are optimised for either day (the HQ Camera is an example) or night (they have IR Lights on either side of the camera, or a ring of IR lights around the camera).
I would either try a different camera, or supply the area you want to observe with appropriate lighting for night recording.

mao1mao2, 2022-09-06:
Sorry I missed that: 20200606.imgand RPi W zeroI thought software ver and pi version do not matter.

mao1mao2, 2022-09-06:
I do not expect it to exceed the performance of a usual commercially available security cam on market, I just want it to perform equally or slightly worse but this HD cam in the first day cannot even see clearly a obvious road lamp.... This is way too under my expectation.

mao1mao2, 2022-09-06:
Besides, I checked the spec of the cam sensor in the HD cam and it is good, which makes me wonder what had happened..Thank you in advance!

starbasessd, 2022-09-06:
Infrared cut-off filters, sometimes called IR filters or heat-absorbing filters, are designed to reflect or block near-[infrared](https://en.wikipedia.org/wiki/Infrared) wavelengths while passing [visible](https://en.wikipedia.org/wiki/Visible_spectrum) light.
The camera is great for daylight, sucks at IR/Night, especially without supplemental light.
According to Raspberry Pi's product page, here’s what to expect out of this new unit:
·      Sony IMX477R stacked, back-illuminated sensor, 12.3 megapixels, 7.9 mm sensor diagonal, 1.55 μm × 1.55 μm pixel size
·      Output: RAW12/10/8, COMP8
·      Back focus: Adjustable (12.5 mm–22.4 mm)
·      Lens standards: C-mount, CS-mount (C-CS adapter included)
**_·      IR cut filter: Integrated  <<<===---_**
·      Ribbon cable length: 200 mm
·      Tripod mount: 1/4”-20

mao1mao2, 2022-09-06:
Thank you so much! I will definitely checkout the IR filter.
However, in the MotionEye why not there is no iso or shutter settings? Or I missed it somewhere?

starbasessd, 2022-09-06:
It is available, but only for certain camera models.
You could try dev20201026, but I do not get it for my PiCam V2.
You can also use the "Extra Motion" in Settings, Video Settings, Extra Motion Options, but those are manual, not automatic..
The IR Filter blocks seeing in the dark.

starbasessd, 2022-09-06:
You can also check [here](https://motion-project.github.io/motion_config.html#video_params) for other video parameters to put in the Motion Options box

mao1mao2, 2022-09-06:
Really appreciate your reply, starbasessd!
For the Extra Option, how to execute it after inputting string command into the dialog box?

starbasessd, 2022-09-06:
Click Apply at the upper right of the column.
You may have to click on another box for it to register you are making a change, like Camera Name.
Be sure to turn off Auto Brightness if you are setting brightness commands.

mao1mao2, 2022-09-06:
Really appreciate it starbasessd!

mao1mao2, 2022-09-07:
Just removed IR filter and tried out.
Now, everything reddish..... 
So I give up... MotionEye cause some issue as iso cannot be configured in ver 06062020; HD cam cause major part of issue as it has so poor performance in low illumination... Gush.... my Amazon Blink works perfectly, 100X better than Pi Cam stuff ...

starbasessd, 2022-09-07:
The HD cam is NOT designed for low light conditions. It is designed for higher resolutions and C/CS lens connection.
The PiCam v1 and V2 also is not great for low light. If you want a camera for working in low light, you need something like:
https://www.amazon.com/Infrared-Vision-Camera-Raspberry-Webcam/dp/B07T22X3PQ or
https://www.amazon.com/raspberry-pi-infrared-camera/s?k=raspberry+pi+infrared+camera
All of these are optimized for low light, and most come with IR lights.

mao1mao2, 2022-09-07:
Thank you for giving me the link starbasessd.
My purpose is not building a low light camera, but a day and night all purpose one.
My Blink camera works great - at night, with no IR light, it can display street light, moon very well and color is bright; day time, no reddish at all, all natural color.  Mmm... ever before I did not think this is difficult, and I guess IMX477 is even better than its image sensor so I want to build one better than Blink, with even adjustable focus and aperture.
Now I guess it is not easy, Blink must have its own illumination technology or algorithm etc to get its camera so well.

starbasessd, 2022-09-07:
No, it is not easy. Most of the cameras I listed can be used as day/night cameras. You can also find a number of USB or wifi/ethernet IP cameras for under $50US on sites like Amazon or AliExpress... Before buying one, though, I would check with the iSpy database at https://www.ispyconnect.com/cameras to make sure the camera has a valid http or rtsp port to get the video from...

mao1mao2, 2022-09-08:
Thank you again for the useful info, starbasessd!You are fabulous!

starbasessd, 2022-09-08:
You don't click on the ?, you just point at it.
Docs are at the Wikis:
[motionEye](https://github.com/motioneye-project/motioneye/wiki)
[motionEyeOS](https://github.com/motioneye-project/motioneyeos/wiki)
[motion-project](https://motion-project.github.io/motion_config.html) # the primary project that motionEye acts as a wrapper for, but is owned by a different group.

mao1mao2, 2022-09-08:
Got it. It works! Thank you starbasessd!