motioneye-project/motioneyeos #2942: No password when streaming externally

Milbredt, 2022-09-25:
I am having problems with the external access user/password.

I am using a raspberry pi 3 (B+) with the raspberry cameramodule V2.

The camera is automatically added and streams when i start the raspberry. 
I have addad an admin and user password. 

The problem I have is that when I put the ip into chrome and along with the port 8081, and thereby use the port forwarding that I have set up, I see the video stream, but nowhere do I need to enter either user name nor password. 

Does the camera itself need a password?
I tried removing and adding the camera again, and from what I got from some youtube video, I should use Local V4L2 Camera as Camera type, and  and mmal service 16.1 as Camera. Is this correct or is there another option here I should use instead?
The option I am using now does not let me set either usernam nor password.

In the motioneye interface I am using IP adress 192.168.1.101. 
In my router I have set a port forward with both port and forward port to 8081 along with the forward IP 192.168.1.101.

In chrome I write the external ip adress followed by :8081

What do I do? What am I missing?

I am using motionEye Version 0.42.1 and OS Version motioEye 20200606

starbasessd, 2022-09-25:
Don't need the password for the stream, just the administration page / conglomerate page.

Milbredt, 2022-09-25:
But what do I do if I want it? I don't want just anybody to be able to get access to the video stream simply by puting my ip and port in...

starbasessd, 2022-09-25:
If using motionEye, port forward 8765. If motionEyeOS port forward 80.

starbasessd, 2022-09-25:
Another option is to put up an nginx server and port forward through it. It has the advantage of also giving you an HTTPS capability. [Here](https://github.com/motioneye-project/motioneye/wiki/Running-Behind-Nginx)

Milbredt, 2022-09-25:
I am using motionEyeOS. I tried putting port 80 and that works. I get the login page. Thank you.

Just a question though. Now I just put in the IP from externally (no port) and end up att the login. I don't know much about ports and such, but what if I for example wanted access to something else from externally, like another motionEye (poor example, I know...) I'm guessing port 80 is some kind of standard that covers all or something. 
Would I then need to use two different, more "specific" ports?

starbasessd, 2022-09-25:
Yes. 
So, if you had 2 internal servers that both us port 80, and you want to be able to see both from outside, you could do a port forward from outside port 18080 to the first one's port 80, and outside port 19080 to the other's port 80.

Milbredt, 2022-09-25:
So if I wanted access them I would put in like 123.456.1.255:18080 and 123.456.1.255:19080, if I understanding correct?

starbasessd, 2022-09-25:
Yes.

Milbredt, 2022-09-25:
Ok, Thanks :)