motioneye-project/motioneyeos #2944: Best motion detection settings

ACJUK, 2022-10-04:
More of a question than an issue.

I've been running this for a week or 2.  What I find though is that I get loads of very short clips even when there appears to be continued activity.

I have:
[image]

Ideally, what I want is for it to start recording and then only stop recording if there is no motion for 10 seconds.  I'm using it to record hedgehogs in the back garden.  So a hedgehog will wander into frame.  It'll stop and have a scratch for a couple of seconds and recording will stop.  Then it'll stop scratching, move a bit and then stop for another scratch.  It'll be in frame doing this for maybe 30 seconds and I'll end up with a load of 3 to 5 second videos when I'd like one 30 second video.

Anyone got any ideas on the ideal settings for this?

starbasessd, 2022-10-04:
I would move Frame Change Threshold to 0.1%, Decrease motion gap to 2 seconds, lower Capture before to 2 frames, and decrease capture after to the same or slightly longer. 
Before and after do not affect the actual detection, they set limits after a detection is made. (It goes back and adds the previous number of frames in Capture Before to the movie, then continues after the detection ends the number of frames in Capture After. They can actually interfere with the length of a single shot, especially if you get new detections while it's processing the last one.
In your use case, you want to increase the sensitivity to the detection. There isn't a number to set here for minimum recording length, unfortunately.
Motion itself has many length options, but none for minimum recording length: [See Here](https://motion-project.github.io/motion_config.html#OptDetail_Movies)

ACJUK, 2022-10-05:
> I would move Frame Change Threshold to 0.1%...

Nice one.  Thank you for that.  I'll see how it goes tonight