motioneye-project/motioneyeos #2945: Motion eye and google drive again redux

bulgin, 2022-10-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
0.43.0
I am running motionEyeOS version: (insert your version here, e.g. 20180314).
4.3.2
### Board Model

I am using the following board/model: Raspbian 11.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
MMAL Camera
My camera model is: (insert your camera model, e.g. Logitech C290).
vc.ril.camera

Error when attempting to upload files to google drive
Error when attempting to upload files to google drive

To help protect your account, Google will soon block apps that don’t comply with Google’s security policies.
Make sure you are using the latest version of this app. You can let the app developer (gogole inserts developers email here) know that this app should stop using unsupported OAuth flows. Learn more
If you’re the app developer, see the migration guide.
### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
wifi and sometimes cable ethernet
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

bulgin, 2022-10-07:
When attempting to upload google docs this message comes forth.  Never happened before.  Only started today after changing OAUTH  credentials.

To help protect your account, Google will soon block apps that don’t comply with Google’s security policies.
Make sure you are using the latest version of this app. You can let the app developer (google inserts developers email here) know that this app should stop using unsupported OAuth flows. Learn more
If you’re the app developer, see the migration guide.

starbasessd, 2022-10-07:
0.43.0
I am running motionEyeOS version: (insert your version here, e.g. 20180314).
4.3.2
Indicates you are using motionEye on another OS, not motionEyeOS.
Did you delete the previous key you generated in Google Key Gen? Google doesn't allow 2 keys with the same name to be active.

bulgin, 2022-10-07:
My apologies I am using motioneye on raspberry pi Raspbian 11 - am I in the wrong github support?

I deleted all keys then reinitiated 1 key.  That key does not work and I get the errors as posted above.

starbasessd, 2022-10-07:
Yes, wrong github, and it may be the 0.43 version (I don't run it on any of my machines) is broken.
When you put in the new key in motioneye, did you copy/paste or did you put it in manually? It should be 16 characters, with no spaces (even though it shows the spaces in the generator).

bulgin, 2022-10-07:
Thanks starbasessd.  In motioneye I see the following.  Don't understand the difference between "motionEye Version" and "Motion Version".  Why is it categorized as such totally confusing.

motionEye Version | 0.43.0
Motion Version | 4.3.2
OS Version | Raspbian 11

bulgin, 2022-10-07:
The latest version posted here: 
https://github.com/motioneye-project/motioneye/releases/tag/0.42.1

So what is 0.43.0?  That's a higher number.  Now I'm really confused. . .

bulgin, 2022-10-07:
If someone could point me to that best working version that also has install instructins I would be forever grateful.  Thank you.

starbasessd, 2022-10-07:
motionEye is the wrapper for motion. It gives you the WebGUI and easy access to the settings of motion.
motion is the program of another group that does the 'heavy lifting' (actual detection, conversions, etc)
0.42.1 is the last stable motionEye, written by Calin Crisan, but is dependent on Python2.7.
0.43.0 is the new beta motiponEye written by new developers, removing the dependency of Python2.7, using only Python3.x.
Like I said, I do not use the 0.43 version of motionEye on any of my machines. I used to keep the documentation up to date for Calin, but since the new devs took over, I haven't, and they need to update a lot of docs to their new version. I still help occasionally here in the issues side, especially here in the motionEyeOS side, as they seem to be neglecting it.

starbasessd, 2022-10-07:
I am checking to see if my last set of instructions is now broken, or if they still work... May be in the morning before I have an answer for you (US EDT)...

starbasessd, 2022-10-07:
BTW if you are using a CSI camera (ribbon connected) you need to use Buster (10) not Bullseye (11) as RPiFoundation made a bunch of changes to Python Support for the CSI cameras, and it broke a lot of camera programs...

bulgin, 2022-10-07:
Oh wow starbasessd that is a lot of useful information.  I see I have to reinstall my pi cameras software but at least now I know which software to use.  So I should use the 0.42.1 on Python2.7 and Buster 10.  

It's really difficult to find coherent instructions, for example this set:

https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian
says "Raspbian Buster comes with motion version 4.1; it is however recommended that you install version 4.3.2, as indicated above"
and this set:
https://github.com/motioneye-project/motioneye/tree/dev#installation
says to use python 3.7.

I'm just looking for instructions for a generic install using the versions you suggested.  Can't seem to find them anywhere.

starbasessd, 2022-10-07:
If you notice:

Install On Raspbian
starbasessd edited this page on Jan 1 · 44 revisions 

I am checking them now to see if PyPi is broken or pulls 0.43 for the install, but last time I checked, (Jan 1 2022) it was working fine.

starbasessd, 2022-10-07:
As to the comment you made, the Wiki instructions are saying if you just did apt install motion, you would get 4.1.x, but if you use the instructions just above that info, you get the best supported version of motion, 4.3.2.

bulgin, 2022-10-07:
So to use the more stable version I would choose 4.1.  That seems to have less issues from what I can see.

starbasessd, 2022-10-07:
No, 4.3.2 is the most stable version of motion.
I just finished with the install on [RaspberryPiOS Legacy (Buster 10)](https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2022-09-26/2022-09-22-raspios-buster-armhf-lite.img.xz) Lite following the [instructions here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian).
It is up and running as expected with a CSI camera (PiCam2) as expected, and able to connect to IP cameras as well. (No USB Camera available, but I expect it to work, too.
This gives me 
[image]

starbasessd, 2022-10-07:
I also expect it to work on Legacy Desktop, but I don't want to try it tonight...

bulgin, 2022-10-07:
Okay thanks starbasessd.  I'll try that.  The whole purpose of this excersie is to get files sent to goolge drive.  Drive seems to not like the authentication code in motioneye - whereas it did before -  but maybe you can try and see if it works?

starbasessd, 2022-10-07:
I clicked on Upload Media Files
I selected Google Drive
I selected / for the path
I selected Obtain Key
I Enabled and copied the key (I thought it was shorter, my bad)
I came back to MotionEye and pasted the key
I clicked on Apply at the top
I clicked on Test Service at the bottom
Success!
I forced a couple of snapshots (not much moving this late) and having not changed anything else, put them in the folder 2022-10-06 with the filenames of (timestamp).jpg as expected.

starbasessd, 2022-10-07:
Offline until 08:00 EDT (UTC -4) G'night

bulgin, 2022-10-07:
Ok.  I'll try again tomorrow.  Thank you so much for your clear and understandable directions.  I think that is a rarety these days.  I'll report back tomorrow.  I just tried to authenticate again after removing all app passwords from Google but get the same message.  Attached.  I'm guessing it's a version thing.

[image]

starbasessd, 2022-10-07:
If you clicked on the blue link, "error details' and copied the URL it goes to it would be of assistance...

bulgin, 2022-10-07:
Error 400: invalid_request
The out-of-band (OOB) flow has been blocked in order to keep users secure. Follow the Out-of-Band (OOB) flow Migration Guide linked in the developer docs below to migrate your app to an alternative method.
Request details: access_type=offline response_type=code redirect_uri=urn:ietf:wg:oauth:2.0:oob client_id=removed-for-security-reasons.apps.googleusercontent.com scope=https://www.googleapis.com/auth/drive.file
Related developer documentation

bulgin, 2022-10-07:
I was able initially after hours of not using the service, to correctly obtain a new code, but, it didn't work meaning it didn't upload files to google.  So the above error code was then produced.

starbasessd, 2022-10-07:
And this was with the 0.43 version of motionEye, correct?

bulgin, 2022-10-07:
Yes, I haven't had time to reinstall so it's the below version.  Re-installing later today.

Ths is what was running when the error was produced:

motionEye Version | 0.43.0
Motion Version | 4.3.2
OS Version | Raspbian 11

bulgin, 2022-10-08:
Tried to do the install you linked to above on a fresh install of Debian Buster Lite with the instructions you linked to here:
https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian
but a lot of errors - dependencies not found.

starbasessd, 2022-10-08:
If you installed on Debian using the Raspbian instructions, I would not expect it to work...

bulgin, 2022-10-08:
I installed Rasberry Pi OS Lite (legacy) via the Raspberry Pi imager v1.7.2 on a raspberry pi. In that installer it states:

"A port of Debian Buster with security updates and no desktop environment"  Are you suggesting that Motion won't run on this combination of hardware/OS?

starbasessd, 2022-10-08:
You are not clear with your statements. There is a pure Debian for RPi, and I thought with your statement "Tried to do the install you linked to above on a fresh install of Debian Buster Lite " that you installed Debian and not RaspberryPiOS.  
I will try again tonight to Install RPiOS Buster Lite on a Pi4 and  see if I get the same issue. Strange.

starbasessd, 2022-10-09:
I have installed RaspberryPiOS Legacy (Buster) and followed the instructions as posted, and had no issues.

starbasessd, 2022-10-09:
I use [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/) not Raspberry Imager.

bulgin, 2022-10-10:
thanks for your patience.  I was able to install using the instructions provided so that my install version was 4.1 without errors.  I did that by just not installing  dpkg -i pi_buster_motion_4.3.2-1_armhf.deb, since, as you said version 4.1 is installed by default.

Problem is, with that version motion was not detected.  So I reinstalled the 4.3.2-1 version and motio was detected.  As well, I noticed that the password requirements for the non-admin user in 4.1 does not allow special characters but the admin password does allow special characters.  

I also tried installing using docker but it eats up too much resources and always hangs the Pi.

bulgin, 2022-10-10:
These are the errors when I follow the instructions at:
https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian

apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y

Err:129 http://raspbian.raspberrypi.org/raspbian buster/main armhf libsndfile1 armhf 1.0.28-6+deb10u1                                                                                       
  404  Not Found [IP: 93.93.128.193 80]

Err:147 http://raspbian.raspberrypi.org/raspbian buster/main armhf mariadb-common all 1:10.3.36-0+deb10u1                                                                                   
  404  Not Found [IP: 93.93.128.193 80]
Err:148 http://raspbian.raspberrypi.org/raspbian buster/main armhf libmariadb3 armhf 1:10.3.36-0+deb10u1                                                                                    
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libs/libsndfile/libsndfile1_1.0.28-6+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/mariadb-common_10.3.36-0+deb10u1_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/libmariadb3_10.3.36-0+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80

starbasessd, 2022-10-10:
pi@rp0-4:~ $ sudo apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
libmariadb3 is already the newest version (1:10.3.36-0+deb10u2).
libmariadb3 set to manually installed.
ffmpeg is already the newest version (7:4.1.9-0+deb10u1+rpt1).
The following NEW packages will be installed:
  libmicrohttpd12 libpq5
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 242 kB of archives.
After this operation, 893 kB of additional disk space will be used.
Get:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libmicrohttpd12 armhf 0.9.62-1 [86.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf libpq5 armhf 11.17-0+deb10u1 [156 kB]
Fetched 242 kB in 4s (57.7 kB/s)
Selecting previously unselected package libmicrohttpd12.
(Reading database ... 155430 files and directories currently installed.)
Preparing to unpack .../libmicrohttpd12_0.9.62-1_armhf.deb ...
Unpacking libmicrohttpd12 (0.9.62-1) ...
Selecting previously unselected package libpq5:armhf.
Preparing to unpack .../libpq5_11.17-0+deb10u1_armhf.deb ...
Unpacking libpq5:armhf (11.17-0+deb10u1) ...
Setting up libpq5:armhf (11.17-0+deb10u1) ...
Setting up libmicrohttpd12 (0.9.62-1) ...
Processing triggers for libc-bin (2.28-10+rpt2+rpi1+deb10u1) ...

Looks like the repository you hit "[IP: 93.93.128.193 80]" is broken. There are ways to get different repositories, but that is beyond the scope of my assistance here.

starbasessd, 2022-10-10:
"I also tried installing using docker but it eats up too much resources and always hangs the Pi."
I recommend 
> sudo apt install docker.io docker-compose

instead of using docker-ce, which is problematic for me in both RPiOS and Debian.
I like the dockers because they are compatible with ARM and AMD64 and doesn't matter for other dependencies.

bulgin, 2022-10-10:
Thanks for that.  I was able to install 4.1 and it seemed to work with google drive but unfortunately I couldn't thoroughly test it because in 4.1 my standard raspberry pi camera isn't recognized.  In 4.3.2.1 it is recognized, but in that version I can't get google to authorize third party apps.  So I'm stuck in the never never land.  I tried using snaps to install but they also don't recognize a standard raspberry pi camera.  I think I'll give up on this project.  Unfortunately it had a lot of good features but the discrepancies between the different versions/hardware is a hill too steep for me to climb. Apprecaite all your help.

ahlawahlaw, 2022-10-29:
I have the same issue since yesterday when I have a power outage that fry my PI.  I rebuild two using Motioneye20200606  which 0.42 and also 0.43 which using bulleye.  They both work except cannot obtain authenticate key, same error with bilgin.  I think Google have enforced the OAuth sometime this month.