motioneye-project/motioneyeos #2946: Button on GPIO to turn ON/OFF wifi (for Access Point conf)

BirdiPiduino, 2022-10-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: Raspberry Pi Zero W
But for some tests, I use a Raspberry Pi 3 for the ethernet and usb ports (keyboard and network)

### Network Connection

My motionEyeOS unit is **not** connected to the network via: it use is own Access Point
IP adress : 192.168.42.1

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Button on GPIO
Led indicator on GPIO

RTC DS1307 (because of isolated situation in Access Point)

### Improvement needed
In a configuration using the Access Point, make it possible to turn ON or OFF the WiFi. This function will be very useful for use in an isolated situation (no power -> use batteries, no network). It will offer more autonomy because it decrease the need of energy.

It is then possible to use an external button on the GPIO to activate the WiFi and a led on the GPIO to indicate that the WiFi is active.
WiFi should turn on at startup. Turn OFF the WiFi can be automatically with a timer.
Re-turn ON the wifi or re-init the timer can be done by pressing the button.

I wrote the script (attached) for manage the button and the led, but it miss the commands for the WiFi.
What would be the command lines to enable/disable WiFi ?

I t will better if it avoid to reboot when the wifi turn on or turn off.

Example of the utility of this issue :
With motionEyeOs, I built a standalone camera-trap to study wilds birds at nest.
It use batteries, USB storage, Wifi Access Point and Real Time Clock for date/time because it's in an isolated situation.
There is a success. And now, i try to decrease the consumption of energy.
That is wath, stop the wifi when I don't use it,  could be a way. I actually need the Wifi just 5 min every 4 days to check every things. 

To test the script attached, copy it in /data/etc/
[simulation_WiFi_button.zip](https://github.com/motioneye-project/motioneyeos/files/9731504/simulation_WiFi_button.zip)

BirdiPiduino, 2022-10-11:
I find the commands to enable and disable WiFi without rebooting (modprobe brcmfmac).

Find attached a solution into the script of the button.
[WiFi_with_button.zip](https://github.com/motioneye-project/motioneyeos/files/9758188/WiFi_with_button.zip)

starbasessd, 2022-10-11:
Does it actually shit off the WiFi chip, or does it simply disable it? Big difference in what you originally requested...

BirdiPiduino, 2022-10-11:
I'm not sure. But during my lasts tests on RPI3, I 've noted a difference of 20 mA (average, sometime null, somtime 40 mA).
Your question is very interresting. I don't know where i can find the response. But I will test all methods and continue to measure the consumption of current method by method. I will find if any methods are similars or if some of them are very more efficients.

BirdiPiduino, 2022-10-11:
It miss also some msg for the log file (button pressed detected, GPIO active, WiFi disable and enable)

starbasessd, 2022-10-11:
Removing the driver (modprobe brcmfmac) as far as I know, doesn't 'shut the chip off', just disables / removes the driver for the chip. You could ask in the RPi forums on [RPiFoundations](https://forums.raspberrypi.com/) site...

starbasessd, 2022-10-11:
If you need a script to run at bootup to monitor the GPIO, [here's the instructions](https://github.com/motioneye-project/motioneyeos/wiki/Tweaks#adding-custom-startup-commands)

BirdiPiduino, 2022-10-12:
I did some tests and find a best command (to economize energy) : sudo ifconfig wlan0 down (or up).
I have compared the consumption on RPi 3 with Bullseye between 3 methods (desktop/brcmfmac/ifconfig).

[image]

BirdiPiduino, 2022-10-12:
Do you know if a webpage show how to write a log file in motioneyeos. It could be util to know when button is pressed, if wifi have restarted, when it was stopped... ?

starbasessd, 2022-10-12:
the path you want is /var/log/{name_of_file}
it's the same as any other log file.
You just need to have the utility or script do something like 
> echo " $(/bin/date) button was pressed" >> /var/log/button.log

This will then put " Wed Oct 12 14:33:42 EDT 2022 button was pressed" in a file called /var/log/button.log

BirdiPiduino, 2022-10-13:
Thank you. It's donne with your help.

starbasessd, 2022-10-13:
Glad to help.

starbasessd, 2022-10-14:
You should be able to connect to http://ip_address with a web browser. 
I am not sure why you couldn't.
Once in the WebGUI, you can turn on SSH if you forgot to turn it on with a file called ssh or ssh.txt in the /boot partition when you created the SDCard.
> ip a 

gives you all active IP info
> netstat -tulpn

gives you all active TCP/IP ports and statuses. You are looking for port 80, this line specifically:
> tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      645/python

Unfortunately in mEOS, the command 
> system motioneye status

returns Usage: /sbin/service {start|stop|restart}, status was forgotten...
and systemctl is not installed...

BirdiPiduino, 2022-10-15:
Excuse me if I was not clear.

When I use the command `ip link set dev ap0 down`, the wifi switch down, as i want (economy of energy).
Whe I use the command `ip link set dev ap0 up`, the wifi switch up correctly.

If I compare the answers of the commands lines `ip a` and 'netstat -tulpn', before to switch off the wifi and after to switch on, they are exactly the same.

So, normaly, all are done. When I connect my computer to the AP-wifi, it's ok, that's right.
But it's impossible to connect with url : 192.168.42.1 on the web browser and impossible to connect in SSH.
So I reboot the system, and that's fine, I can connect the computer to the AP-wifi and open MEO in the web browser or access to it in SSH.

BirdiPiduino, 2022-10-15:
I'm sure MOE does'nt reboot after the switch OFF/ON of the wifi because, I watch the screen of MOE with a screen directly connected on HMDI.

starbasessd, 2022-10-15:
Then you can add the command 
> service motioneye restart

after the command that restarts your wifi.. You might need a 5-10 second delay for the wifi to come all the way back up, though which would be:

> sleep 10 &&  service motioneye restart

BirdiPiduino, 2022-10-15:
After the command `service motioneye restart`. It answer `Stopping motioneye : done` and `Starting motioneye : done`.
But there is'nt possible to open MOE in web-browser or SSH terminal.

starbasessd, 2022-10-15:
If you run 
> netstat -tulpn | grep ":80" 

what does it report?

My example:
[root@meye-4cec78b8 ~]# netstat -tulpn | grep ":80"
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      651/python
tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN      4367/motion

BirdiPiduino, 2022-10-15:
tcp 0 0 0.0.0.0:80   0.0.0.0:* LISTEN 1079/python
tcp 0 0 0.0.0.0:8081 0.0.0.0:* LISTEN   1092/motion

starbasessd, 2022-10-15:
Should be there. Did you check the ip address?
>ip a

BirdiPiduino, 2022-10-15:
Yes, I've checked and also with the commande line `ip a`

starbasessd, 2022-10-15:
Can you ping it?

BirdiPiduino, 2022-10-15:
I can not.
Destination Host Unreachable

BirdiPiduino, 2022-10-15:
I've checked the log files

What seems to be done  👍 
in dmes.log, I can read :     `IPv6 ADRCONF (NETDEV_CHANGE) link becomes ready`.
in hostapd.conf, I can read :     `ap0 interface enabled`

not done  👎 
in message.log, I can read :     Oct 15 14:57:54 daemon.info hostapd : ap0 : STA 74:c6:3b:...d5 IEEE 802.1 : dissociated

BirdiPiduino, 2022-10-15:
But on MOE from MOE, i can ping itself well.

I'm not sure it's an interesting indication...

BirdiPiduino, 2022-10-15:
And from MOE, i can ping my computer...

BirdiPiduino, 2022-10-15:
MEO... not MOE    sorry

starbasessd, 2022-10-15:
Can you disconnect whatever phone or tablet you are using, and re-connect?
Not the "automatic connect"...

BirdiPiduino, 2022-10-15:
Yes I've tested with my phone (Android), my Ipad, my laptop (ubuntu) and my computer (windows).
Disconnect/reconnect/forget this network, ...

starbasessd, 2022-10-15:
Ooo... Did your computer pull a 192.168.42.x ip address, or is it back on the home network???

BirdiPiduino, 2022-10-15:
Yes, when my computers or smartphone, ... loosed the Wifi of MEO, they automatically reconnected to my  home network.
So i wait few minutes, i reconnect to the wifi of MEO. It's seems to be done but it is'nt to communicate on web browser and SSH or SFTP...

starbasessd, 2022-10-15:
So when you told any of your devices to forget the network of mEOS, after the power wifi off then on, and you try to connect to it again, it shows in available networks?
Do the devices pull a 192.168.42.x IP address?

BirdiPiduino, 2022-10-15:
Yes, for example my phone have 192.168.42.143.

starbasessd, 2022-10-15:
Interesting. <thinking for a bit>

starbasessd, 2022-10-15:
But if you do  a hard reboot, when it comes back up it allows connections, correct?

BirdiPiduino, 2022-10-15:
Yes all be done.

BirdiPiduino, 2022-10-15:
When i ping into the two directions, I find the IP adress of my phone into my phone parameters.
What is the command on MEO to know the adress of devices connected ?

starbasessd, 2022-10-15:
There will be a *.leases file (IIRC) that will have the leases granted. 
I will have to image a new card here in a bit to confirm
Another way is to do a 'global' grep for the ip address in your mEOS :
> cd /
> grep -r "192.168.42." /

should find it. It'll be a bit before I can do the image, working on another project (grandkids)

BirdiPiduino, 2022-10-15:
I've never tested 

1. switch OFF the wifi : `ip link set dev ap 0 down`
2. Forget the network
3. switch ON the wifi : `ip link set dev ap 0 down`
4. `service motioneye restart`
5. reconnect on webbrowser

that's fine !

but this probleme of forgetting the network it's a source of mistakes.

BirdiPiduino, 2022-10-15:
I'll try to tested on different devices and verify than I could repeat it.

I understand for grandkids. For me, it's the opposite, broken leg give me time... but I need more skills.

starbasessd, 2022-10-15:
You have to remember that your scenario wasn't considered originally, nor when the AP functionality was written in (shutting off and turning on the wifi for power savings....)

BirdiPiduino, 2022-10-15:
Yes, I understand.
Lasts weeks, i tried to change and use motioneye on raspbian. But the newest versions of raspbian, I have'nt installed motioneye with success.
I will probably cancelled the power savings... and in long term abord motioneyeos for motion+nodered+raspbian lite if it's not to much for the CPU and power savings.

starbasessd, 2022-10-15:
Be aware, motionEye on Raspbian has lots of issues, as the New Devs are trying to 'upgrade' mE from Python2.7 to 3.x. Among the issues: CSI (ribbon Cable) PiCam is broken with Bullseye, (may be broken with Buster, too). Go over to that Wiki and look through recent issues...

BirdiPiduino, 2022-10-15:
That was the problems I had. Impossible to use any versions of raspbian and motioneye, no image of my camera CSI.

BirdiPiduino, 2022-10-15:
A lot of errors during the installation of motion with python...

starbasessd, 2022-10-15:
If you were using Bullseye, pip3 doesn't work for motionEye, and pip2 is flaky. Buster Legacy mostly works, but I think it's dependent on which updates are applied. mE 0.43 is supposed to be 'better', but ...

BirdiPiduino, 2022-10-16:
This is the script i use to test the effects of switch ON/OFF the wifi.

`#!/bin/bash

record_log() {
        IPANSWER=$(ip a | grep "ap0")
        TCPANSWER=$(netstat -tulpn | grep ":80")
        echo $IPANSWER >> /var/log/switch_wifi.log
        echo $TCPANSWER >> /var/log/switch_wifi.log
        echo "" >> /var/log/switch_wifi.log
}

echo "test SWITCH ON/OFF Wifi AP0"
echo "" >> /var/log/switch_wifi.log
echo "-------------------------- switch_wifi ------------------------" >> /var/log/switch_wifi.log
echo "" >> /var/log/switch_wifi.log

echo "$(/bin/date) : nothing, states before" >> /var/log/switch_wifi.log
record_log

echo "$(/bin/date) : Stopping Wifi"
echo "" >> /var/log/switch_wifi.log
echo "$(/bin/date) : ip link set dev ap0 down" >> /var/log/switch_wifi.log
ip link set dev ap0 down
record_log

echo "Sleeping for 60s, forget the network on your device"
sleep 60

echo "$(/bin/date) : Restarting Wifi"
echo "$(/bin/date) : ip link set dev ap0 up" >> /var/log/switch_wifi.log
ip link set dev ap0 up
record_log

echo "$(/bin/date) : Restarting the motioneye services"
echo "$(/bin/date) : service motioneye restart" >> /var/log/switch_wifi.log
service motioneye restart`

BirdiPiduino, 2022-10-16:
I will continue to search about services stopped and maybe dns problems...

Thank you very much for the help than you given me.

BirdiPiduino, 2022-10-19:
Awaiting the solution for a switch ON/OFF with a button and without rebooting, find attached a solution to manage the poweroff and the wifi's switch off with a button and a led.

[Button, Poweroff and WiFi.zip](https://github.com/motioneye-project/motioneyeos/files/9822615/Button.Poweroff.and.WiFi.zip)

starbasessd, 2022-10-19:
If you are waiting for me to come up with something, you are going to be waiting quite a while, sorry.
You've gone a long way to create your own solution, and I appreciate your efforts. Thank you.