motioneye-project/motioneyeos #2948: Matter support

leoheck, 2022-10-08:
Google is launching this Matter. I would like if the MotionEye project has any plan to support it?
We could integrate cameras with Google Home app if the support is provided.

Some relevant links:
https://www.theverge.com/2022/10/4/23386287/google-home-app-design-update-matter-features
https://developers.home.google.com/matter

starbasessd, 2022-10-08:
My concern is the fact that it is cloud based, as far as I can tell. I picked this project because it isn't. Yes, you can use Google to store pics and videos, but it is not a 'requirement'.  I don't necessarily want to risk some cameras videos to get out to where hackers can get them. I also don't use NEST and other cameras that require the cloud. HomeAssistant is my alternative to Alexa or Google Home. All my devices are local only. If Matter is truly an open standard, I'm sure the new devs would consider it.

starbasessd, 2022-10-08:
BTW, it would be best to post this over on the motionEye github, too. 
Also, in my (not so humble) opinion, I think it'll go the way of Amazon Show. It was promised to be 'open', but became very exclusive, and is on their 'to be terminated' list...

leoheck, 2022-10-08:
Oh, I thought this was the motioneye GitHub, it is motioneyeos. It makes more sense in the other repo, for sure.