motioneye-project/motioneyeos #2950: Rainbow Screen

soapy75, 2022-10-09:
I have used multiple different guides with the same outcome.
I am stuck on rainbow screen on boot after reformatting multiple times using slightly different methods from different articles.

I am using: 
raspberry pi 2w

latest version of the 'raspberry pi ' image

steps:
downloaded and unzipped image
used both raspberry pi imager to format and also balena etcher
added new text file inside sd card
copied wpa text and changed nessasary items
save as -> wpa_supplicant.conf and saved as all files
plugged into pi

rainbow screen

starbasessd, 2022-10-09:
There is no official raspberry pi image for the RpiZero2W. There is this image and thread:
https://github.com/motioneye-project/motioneyeos/issues/2875
or you can install RPiOS Legacy (Buster) and follow the instructions [here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian).

soapy75, 2022-10-09:
I have attempted that but I was unable to connect, were there supposed to be any further steps?

starbasessd, 2022-10-09:
If you do the RPiOS Legacy (Buster) it would be http://ip_address:8765 instead of just http://ip_address.
If you are connecting in via SSH or via monitor and keyboard, you can run the command:
> netstat -tulpn

and look for the line:
tcp        0      0 0.0.0.0:8765            0.0.0.0:*               LISTEN      -
This confirms motioneye service is running and listening for connections. If you do not see that line run the command:
> sudo systemctl status motioneye

You should get something like:
>pi@camera-server:~$ sudo systemctl status motioneye
[sudo] password for pi:
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-10-09 14:39:42 EDT; 2h 55min ago
   Main PID: 480 (meyectl)
      Tasks: 40 (limit: 6865)
     Memory: 2.3G
     CGroup: /system.slice/motioneye.service
             ├─ 480 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             ├─ 755 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             └─3439 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5

and a number of motions listed

closed 2022-10-10 by soapy75