motioneye-project/motioneyeos #2951: impossible to log in (read-only file system)

axelsimon, 2022-10-14:
After booting and resizing the rest of the SD card to a data partition, MotionEyeOS loads and runs into a series of errors explained by:
`Read-only file system`.

Most crucially, this means that the startup action to create an empty password for the root user fails and it's impossible to log in with a keyboard + screen:
`Setting empty root password: touch: /data/etc/shadow: No such file or directory`

The device also never runs wpa_supplicant from what i can tell and therefore never connects to the network. It just sits there, impossible to interact with.

This is a second SD card i'm trying, and about to try a third.

### motionEyeOS Version

motioneyeos-raspberrypi-20200606 flashed from the archive with Balena Etcher 1.7.9

### Board Model

Raspberry Pi Zero W v1.1

### Camera

never got that far

### Network Connection

WiFi, provisioned with `/boot/wpa_supplicant.conf` (not working)

### Peripherals

none

### Log Files

non (the OS can't write them ¯\\\_(ツ)\_/¯ )

axelsimon, 2022-10-14:
Ok, works fine with a different (and brand new) micro SD card.
Looks like even though the one i was using is recent and hasn't been used much, it still went to read-only mode because of errors or some sorts, preventing MotionEyeOS from booting properly.

Conclusion: SD cards are crap :)

closed 2022-10-14 by axelsimon

starbasessd, 2022-10-14:
Also, mEOS creates it's own 3rd partition and expands by itself (your original issue implies YOU created and expanded the 3rd partition)
If you want, there are many sets of instructions out there for running mEOS on USB sticks instead of SDCards (I agree, SDCards SUCK, and I only use them with pre-Pi2B V2 Pis.) and if asked nicely, I could post again here...