motioneye-project/motioneyeos #2953: Doesn't work for Pi4

Laz2516, 2022-10-18:
Doesn't work for Pi4.
Checked two images, 20200606 and 20201026. None starts on the new Pi4. Help, what should I do? (((

starbasessd, 2022-10-18:
The last known good bootloader for dev20201026 for RPi4 was 04/29/2021.
You can _try_ to downgrade the bootloader, but it is hit or miss.
Boot to RPiOS Buster Legacy
run the command:
> sudo rpi-eeprom-update -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2021-04-29.bin

When completed (after a reboot) the command 
> vcgencmd bootloader_version

should now show
- - - - -
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1645126804
- - - - -
I don't have a Pi4-2GB, but it works with the Pi4-4GB and Pi4-8GB from an upgrade, but still sometimes fails when downgraded from a newer version...

starbasessd, 2022-10-18:
There are a number of CSI (ribbon cable) cameras that would benefit from a update to the kernel, too (especially the ultra-HD / Ultra High Rez that cams out around and just after 10/2020...

Laz2516, 2022-10-19:
Thank you
Then please give a link to the correct repository with updates. I can wait and not update the bootloader to the old image file.
(sorry, I'm from Russia and I have to use a translator, I'm also still a little experienced at Raspberry....))))

Junaid0411, 2022-10-19:
I'm getting the same issue, both dev20201026 and 20200606 won't boot up for Pi 4 Model B 2GB

starbasessd, 2022-10-19:
It is a known issue (see my comments [above](https://github.com/motioneye-project/motioneyeos/issues/2953#issuecomment-1282475393)) and the current solution is to downgrade the boot loader (also see instructions [above](https://github.com/motioneye-project/motioneyeos/issues/2953#issuecomment-1282475393)). Downgrading to boot loader April 29 2021 should (but not guaranteed) allow dev20201026 to boot, at least until the new Devs can get a new updated version of motionEyeOS compiled.

Laz2516, 2022-10-19:
We are waiting for a change in the image, which will include a change in the bootloader (if possible). Still, the hardware is updated, and the image has not changed for 2 years....

starbasessd, 2022-10-19:
@Laz2516:
Unfortunately, real life interfered with CCrisan's ability to continue with the project, which is why he handed it over to some new Developers. 
As you where you should be able to find the new release, it should appear [here](https://github.com/ccrisan/motioneyeos/releases)

Junaid0411, 2022-10-19:
@starbasessd Hello, I loaded up legacy OS and followed what you said, I installed 
`sudo` rpi-eeprom-update -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2021-04-29.bin``

My version lines up to yours now, what should I do after?

starbasessd, 2022-10-19:
Now you should try using dev20201026 for the Pi4 from [here](https://github.com/ccrisan/motioneyeos/releases)

Junaid0411, 2022-10-19:
Sorry, I'm quite new, just to clarify, I need to flash the image back onto the SD card correct and then bootup again

starbasessd, 2022-10-19:
Yes, if you used the same SDCard to downgrade the boot loader...

Laz2516, 2022-10-19:
Forgive me too..) I'm not a very experienced Raspberry owner either...))
If I understand correctly, then I upload the image 20200606 to the map, launch it on the Pi4. I get a never-ending error. Next, I go to the SSH Pi4 (what is the login\password in this case?).
I update the bootloader to the required version. Take the card out of the Pi4 without turning it off?
Recording the image 20201026 on the map, inserting it into Pi4..... overloading the Pi4 and should it start normally motioneyeos?

starbasessd, 2022-10-19:
You want to use dev20201026 on any RPi4.
You want to image the SDCard with the image [here](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz )using Etcher or Win32DiskImager (Raspberry Pi Imager breaks the file system)
ALWAYS turn off the Pi before pulling the SD Card. ALWAYS.
SSH User is "admin" and password is just enter.
If you get a continuous reboot where the last line says 'panic' then that means you don't have an active network connection.
There are ways around it but that involves a lot more work for a new user.
After re-imaging the SDCard, you then put in in the turned OFF Pi4, then turn it on. It should boot to motionEyeOS.

Laz2516, 2022-10-19:
Yes. I understand about you.
Indeed, I have a continuous update of messages on the screen, but the LAN indicator is not active.
And I don't see Pi4 in the IP scanner. So I will not be able to enter it via SSH (I forgot about it when I wrote the text here).
But the board itself is healthy, from the other SD card I run MagicMirror2 on this board.

So I will not be able to boot from any motioneyeos image and install a new bootloader, then boot from the desired image and work... ((
How do I start the LAN?

starbasessd, 2022-10-19:
The process is:
Install RaspberryPiOS preferably Buster Legacy, on an SDCard
Boot the Pi 4 from the RPiOS SDCard
Log in and run the following command:
> sudo rpi-eeprom-update -d -f /usr/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2021-04-29.bin

When it is complete, reboot with the RPiOS SDCard, and run the following command:
> vcgencmd bootloader_version

should now show
- - - - -
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1645126804
- - - - -
Then you would image a SDCard with motionEyeOS [here](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz) but do NOT use Raspberry Pi Imager, it breaks it,
Use Etcher or Win32Disk Imager. 
You can use several methods to disable networking, or if you have WiFi you can use the same wpa_supplicant.conf file in the /boot partition that any other Pi uses. 

If there is someone else who wants to assist with this, feel free to jump in, I am going to be a bit busy to get too involved today or tomorrow...

Junaid0411, 2022-10-19:
I've tried these steps but I'm getting the same error again...

Junaid0411, 2022-10-22:
@Laz2516 
Hello. I got it to work in the end 😄 

Basically, what you have to do is 
1. Go to https://github.com/raspberrypi/firmware/tree/master/boot 
2. Then download and copy these files to the bootloader: 
start.elf
start4.elf
start_x.elf
start4x.elf
fixup.dat
fixup4.dat
fixup_x.dat
fixup4x.dat

4. Add "start_x=1" to the config.txt file.

Source: https://github.com/motioneye-project/motioneyeos/issues/2892

Hope this helps!

starbasessd, 2022-10-22:
You may (probably will) have issues with USB attached or CSI attached cameras, though as you are not updating the kernal and dtb files that are needed.

Junaid0411, 2022-10-22:
I'm having issues with connecting to WiFi at the moment, it keeps giving me the panic error...

starbasessd, 2022-10-22:
Yep. Need a full update, not just the boot files. According to Michal Ng it's coming...

GamingTechRides, 2022-10-22:
> @Laz2516 Hello. I got it to work in the end 😄
> 
> Basically, what you have to do is
> 
> 1. Go to https://github.com/raspberrypi/firmware/tree/master/boot
> 2. Then download and copy these files to the bootloader:
>    start.elf
>    start4.elf
>    start_x.elf
>    start4x.elf
>    fixup.dat
>    fixup4.dat
>    fixup_x.dat
>    fixup4x.dat
> 3. Add "start_x=1" to the config.txt file.
> 
> Source: #2892
> 
> Hope this helps!

Hi, when I follow these steps (including using the older bootloader version)- I still get an error stating "Invalid ELF header" for files start4x.elf and start_x.elf.  I am on a 8gb Pi 4 plugged into ethernet cable for networking, and using a USB mouse/keyboard.     Do we have any idea when motioneyeOS will become compatible for the Pi 4 8gb?  Thank you.

Junaid0411, 2022-10-23:
> Hi, when I follow these steps (including using the older bootloader version)- I still get an error stating "Invalid ELF header" for files start4x.elf and start_x.elf. I am on a 8gb Pi 4 plugged into [...]

Did you replace all the old start and fixup files with the latest ones? And did you edit the config file?

Laz2516, 2022-10-28:
Good afternoon!
I performed the specified steps (wrote the necessary new files to the root of the map and added a line to the config.txt file.
Everything works on the Raspberry Pi 4, 2 Gb. In a browser on a PC at raspberry, I launched motionEye.
I did not check and configure the work of wifi, I have enough LAN connection.

I still have a question now. I need to display the image of IP cameras not on a PC or tablet in a browser, but need to display the cameras on a monitor connected directly to Raspberry via HDMI.
How to do this? Can I install motionEye on Raspberry Pi OS Full (with desktop and applications) so that I can run motionEye in a browser on the same Raspberry???

Laz2516, 2022-10-29:
Thanks a bunch!
Everything worked out on Pi4\2 Gb. It works

I have two questions, help or tell me where to find or ask a question about help-
1. How to make the browser start automatically (so that when the Pi reboots with the connected monitor to it via HDMI, the browser with the Raspberry IP address will immediately start (to automatically launch on the motionEye monitor).
2. How to connect and configure the operation of the PIR sensor in Raspberry so that the monitor turns off when there is no movement on the sensor (the monitor is connected to Raspberry via HDMI)?
Please help....))))!

starbasessd, 2022-10-29:
Last time I did  your #1, I used the instructions from [here](https://pimylifeup.com/raspberry-pi-kiosk/) and had it rotate through the direct streams: http://ip_address:8081,  http://ip_address:8082, http://ip_address:8083, and so forth. Worked well enough.
As to your #2, it is 'problematic', as RPi doesn't "turn off" the monitor as such. You can put the Pi to sleep, but then the motionEye functionality ends until the Pi wakes up again... You can, with some monitors, turn off the back-light via command, but the best way I've found is to control the power via a motion detector switch like [this](https://www.amazon.com/Activated-Occupancy-Adjustable-Electrical-Triggering/dp/B0B5WYQDBK) or something similar. You could probably use HomeAssistant or similar to do the PIR motion to control a WiFi outlet, but motionEye wasn't written with that functionality built in. You can and do have access to Python 2.7 and 3.x (if you are using motionEye on Raspbian) and can find scripts out in the Interwebs to read the GPIO pins. I did find [this](https://raspberrypi.stackexchange.com/questions/67991/raspbian-put-monitor-to-sleep-at-night-and-wake-in-morning) and you might be able to modify the trigger (they use a cron job) to turn the monitor off and on via the GPIO. I'd be willing to work on it if I had a GPIO sensor, but since I don't...

Laz2516, 2022-10-29:
OK.
I'll try to look for something. The monitor can simply be extinguished so that there is no image. I found such a module used in MagicMirror. But I don't know how to use it without the MagicMirror software itself. There it works as a module.

starbasessd, 2022-10-29:
You could ask on their forum or support, maybe someone could point you at the module and how to put it where you could use it for motionEye...

nickadian, 2022-11-03:
You can try to check out https://github.com/jawsper/motioneyeos/releases/tag/20220119-dev

Laz2516, 2022-11-06:
К сожалению данный образ совсем не запускается на Pi4\ 2 Gb. Светодиод
указывает ошибку. На экране совсем пусто.

starbasessd, 2022-11-06:
English please, or which language is it, as it didn't auto-translate.

Laz2516, 2022-11-06:
;))))) Sorry....))

starbasessd, 2022-11-06:
That is because the images are very specific for each CPU/GPU/extra hardware on each board, and the PiZero2W is different than the Pi4. They are not generic like the RaspberryPiOS images. That particular image was 'updated' for the PiZero2W by Jawsper...

Laz2516, 2022-11-06:
> You can, instructions [here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) Use Buster (legacy) Desktop.

Unfortunately, when installing on this instruction on the old OS, the program itself works well, but the motion sensor does not work well. False positives are generated.
When checking the operation of the motion sensor on the new OS, the sensor works perfectly.
Can I install the program on the new OS Pi4 according to this instructions?

starbasessd, 2022-11-06:
The issue between the old OS (Buster Legacy) and the New OS (Bullseye) is Python2.7 is no longer fully supported and they changed out the old PiCam software. If you do not use a CSI (ribbon cable) and only use IP cameras or USB csmeras, it works (and I wrote those instructions to help out).

Laz2516, 2022-11-06:
I only have IP cameras. So I can safely install according to the instructions from the Wiki? Shouldn't there be any problems?)

starbasessd, 2022-11-06:
That is correct.

Laz2516, 2022-11-13:
Yes, that's right. Installed on the latest version of the Raspberry Pi 32 Full image. Everything works in a Raspberry browser using its own ip: 8765.