motioneye-project/motioneyeos #2954: New bug report: Google Drive no longer accepts connections to get new key in Files, Uploads

starbasessd, 2022-10-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue. Yes

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue. Yes

### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: Any
### Camera

I am using the following type of camera: N/A

My camera model is: N/A
### Network Connection

My motionEyeOS unit is connected to the network via: Either Ethernet or WiFi
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

When attempting to set up Files, Uploads, Google Drive, Obtain Key, before you can do anything else you get:

[image]

Their report suggestion [Related Developer documentation](https://developers.google.com/identity/protocols/oauth2/resources/oob-migration)

soapy75, 2022-10-30:
Ive just encountered the error, is there any manual way to obtain the key?

starbasessd, 2022-10-30:
Not that I am aware of, you could check with google...

starbasessd, 2022-10-30:
In the developer notes from google, it says you can add something to the request to 'bypass' the Oauth error (not recommended), however it would entail decompiling the .pyc files to find where the request is made, and isn't a real fix.

iamapez, 2022-11-01:
I am encountering the same issue running **dev20201026**

xjon685, 2022-11-04:
same issue here,  using motioneye in Home Assistant , using th HACS community addon.   Was working before with the same setup, without issues.  Had to reinstall home assistant due to unrelated issue and now the the  "Access blocked: motionEye’s request is invalid" message when clicking "obtain Key"

atreeon, 2022-11-05:
I have one camera (camera1) that can upload files to google drive.  Adding another camera requires me to obtain a new key from google drive.  Is there a way to use the camera1 key for new cameras?

xjon685, 2022-11-05:
> I have one camera (camera1) that can upload files to google drive. Adding another camera requires me to obtain a new key from google drive. Is there a way to use the camera1 key for new cameras?

Im not sure, from what i understood, all currently working auth keys will stop working at some point anyway. https://developers.google.com/identity/protocols/oauth2/resources/oob-migration

Hopefully the devs will have a solution soon , I really can't afford to pay Dropbox prices, even with Black Friday around the corner.  Maybe this is a good time to consider implementing Amazon Drive, AWS S3 bucket, Backblaze B2, or self hosted solution like nextcloud.  I'll keep my fingers crossed.

starbasessd, 2022-11-05:
I don't know. The key should be in the camera1.conf file, and you could try to use it in the camera2 setup (don't bother with trying the get key function).
I doubt if the new devs will add functionality for the AWS B2 or other options as they aren't trying very hard to get the new release out there for motionEye let alone motionEyeOS. They aren't even acknowledging new issues here.

xjon685, 2022-11-05:
I understand you used to be the dev (great work btw).  But looks like you still spending a lot of your time here.  I think the google Oauth OOB migration is going to make all google drive users of Motioneye and motioneyeOS fail at some point in the next month or so (or maybe they all stopped working on 5th October), most people won't even notice until they check.  I don't know how much contact you have with the new devs, but this is probably going to effect a lot of users.   If they are not going to deal with the issue, should we consider Motioneye to be slow for updates and mOS to be an unmaintained project?

starbasessd, 2022-11-05:
@xjon685 I am not, nor have I ever been a dev, and CCrisan was the originator. I have been here for a number of years, originally helping CCrisan with tech support (my specialty).
As to current dev work, the new devs are almost exclusively focused on motionEye (not mEOS) converting from Python2 to Python3....
I created tickets for both mE and mEOS concerning the issue.
I really hate to think it's been abandoned by the new devs, but it is a possibility. (depressing).

Weedefix, 2022-11-10:
looking forward to having this sorted out. my motioneye is not usable anymore without google photo, or at least drive upload.

xjon685, 2022-11-10:
I wouldn't hold your breath,   I'm considering this to be a dead project now which is sad.  

I'm looking at work arounds after concluding this won't be fixed,  realistically ever.  im now saving videos to my local NAS  (openmediavault 6)  then using rclone to move it to google drive, or nextcloud.  same result as  " clean cloud " option is working by using the local "save videos for X time" and using the rclone sync option.

penguinjeff, 2023-01-01:
A possible work around is mounting google drive https://forums.raspberrypi.com/viewtopic.php?t=109587 and changing the default path. Not a good solution but it should work for some.

willba4, 2023-01-05:
i'm also experiencing this issue. its a pitty

o0LINNY0o, 2023-01-13:
Yep , also just added another camera and found this ,,, was about to delete the container and start again .... (lucky i didnt) but unless a statement gets released , it does look like this motioneyeOS is dead..... Can anyone recommend a alternative ?

benitocalcanho, 2023-01-13:
Same here. Please fix

KhanDoors, 2023-01-13:
same issue 1/13/2023

mkn025, 2023-07-17:
i was able to slove it by writing a python script that copies the fil to my local server, then syncing that folder to googe drive

fatchild, 2023-09-09:
> mkn025

Could you share this code please?

mkn025, 2023-09-09:
> > mkn025
> 
> 
> 
> Could you share this code please?

Sure. It is quite simple really, I it just copies the newest file from where the camera stores it to the server using scp. I am at my cabin right now, but I will share the code when I get back home on Monday.

fatchild, 2023-09-11:
Much appreciated 👍

mkn025, 2023-09-11:
> > > > mkn025
> > > 
> > > 
> > > Could you share this code, please?
> > 
> > 
> > Sure. It is quite simple really, it just copies the newest file from where the camera stores it to the server using scp. I am at my cabin right now, but I will share the code when I get back home on Monday.
> 
> Much appreciated 👍

I am just trying this notion thing
[https://nitramknutsen.notion.site/Transfer-files-to-the-server-from-Motion-Eye-using-SCP-12753569ad4f47fb913453acabbf3eb4?pvs=4](url)

mkn025, 2023-09-11:
> > > > mkn025
> > > 
> > > 
> > > Could you share this code please?
> > 
> > 
> > Sure. It is quite simple really, it just copies the newest file from where the camera stores it to the server using scp. I am at my cabin right now, but I will share the code when I get back home on Monday.
> 
> Much appreciated 👍

i am just trying out this notion thing
https://nitramknutsen.notion.site/Transfer-files-to-the-server-from-Motion-Eye-using-SCP-12753569ad4f47fb913453acabbf3eb4?pvs=4

PeterGamma, 2024-08-30:
There is a lot to try out, and I have also this issue. The most straight forward solution is for me personally create a samba shared folder and let motioneye store the videos in this folder. And then share the folder with a windows PC where it easy to sync it to google drive with to google sync software. But it would be highly desirable if someone has more straight forward solution which not has be be tried out anymore, since it works.

PeterGamma, 2024-09-01:
Google Drive sync does not allow to sync a network folder, so samba share to a Windows PC is too complicated. But NextCloud has a Linux NextCloud Desktop app in the Ubuntu app store. I have a Rasbperry Pi with NextCloudPi on it and it works great. NextCloud has apps for Linux, Windows and Android. I think this is a good replacement for Google Drive for motioneye. You can open it up to access it over the internet:

https://petergamma.org/nextcloudpi-as-a-private-cloud-with-access-over-the-web-as-motioneye-security-camera-solution/

And according to my own little experience synchronizing the local folder of motioneye is better than have a motioneye internal interface. You do not depend on it that the cloud software changes the protocol.

To install NextCloudPi on a Raspberry Pi takes a bit time. Apfelkast has a good instruction. And if it works, I think NextCloudPi with motioneye is a good fit.