motioneye-project/motioneyeos #2955: Unable to pull RTSP stream

nickadian, 2022-11-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20201026

### Board Model

I am using the following board/model: Pi4

### Camera

I am using the following type of camera: V4L2 Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Log Files

Hi Friends! I am hoping someone can help me out here. I am trying to pull an RTSP stream from my pi4 running motioneyeos. It is not working. What am I missing here?

What is the RTSP stream yall usually use?

starbasessd, 2022-11-03:
You don't say specifically which camera you are using.
Camera URLs can vary, even within the same manufacturer.
The database at [iSpy](https://www.ispyconnect.com/cameras) has a great listing of URLs.
Where a URL lists user or password, enter those into the appropriate boxes, and do not include them in the URL itself.

nickadian, 2022-11-03:
Thanks! I noticed afterwards that the device must be in `Fast Network Camera` mode

closed 2022-11-03 by nickadian

starbasessd, 2022-11-03:
You didn't say that you were using a motionEye or motionEyeOS setup as a network camera, either...