motioneye-project/motioneyeos #2958: Usb Camera feed over HDMI and other issues

hackintoshlover, 2022-11-11:
**Preliminary Docs**
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

**motionEyeOS Version**
I am running motionEyeOS version: 20200606

**Board Model**
I am using the following board/model: Raspberry pi 4 model b (2gb ram)

**Camera**
I am using the following type of camera: V4L2

My camera model is:Logitech C270

**Network Connection**
My motionEyeOS unit is connected to the network via: Ethernet

**Peripherals**
I am using the following peripherals that I consider relevant to this issue:
* USB webcam

Issue:
Hi, first of all huge thanks for the project.
1) Click pictures manually while recording video continuously.
When movie setting is changed to continuous, image setting automatically changes to all frames despite being set to manual mode.
How do i fix image setting to manual mode while recording movie continuously?
Attached is a video for better explanation.
[](https://drive.google.com/file/d/16Z7_Gepge0WQckEQCGvA_dRPW-Fgdbpi/view?usp=drivesdk)

2) Access usb camera feed over hdmi
When i switch to fast network camera it displays a video feed over hdmi only if CSI camera is connected, how to view usb camera feed over hdmi?

3) Overlay text on video feed over hdmi.
When i connect CSI camera and enable fast network camera, video is displayed over hdmi but no details such as camera name and timestamp is overlaid on the video, how do i achieve the same?

Thanks.

starbasessd, 2022-11-11:
Unfortunately, all 3 issues are limitations of the software, and would have to be requested as 'feature requests'.

hackintoshlover, 2022-11-12:
Are there scripts/ steps which i can modify to get the above functionalities? so i can get started with those and contribute.

Does motion support two way audio streaming?

starbasessd, 2022-11-12:
Yes, there are some .pyc (pre-compiled Python2 scripts) that have tools available to de-compile them available elsewhere, and modify (but to do so with motionEyeOS is very involved). They are located in a Read-Only part of the OS in the /usr/lib/python2.7/site-packages/motioneye tree but the FastNetworkCamera functionality would need to be re-written altogether. It uses functionality found [here](https://github.com/motioneye-project/motioneyeos/wiki/Fast-Network-Camera)
The source code for the Python is available [here](https://github.com/motioneye-project/motioneye/tree/python2/motioneye)
The current Devs are working on moving motionEye off of Python2 to all Python3 code, but has been difficult over the past 8 - 10 months or so with little advancement.
Motion functionality does not currently support two way (or even one way) audio, and is a separate package found [here](https://github.com/Motion-Project/motion) and not written or developed by the same team that does motionEye/motionEyeOS... Excellent documentation for motion exists [here](https://motion-project.github.io/motion_config.html)

hackintoshlover, 2022-11-12:
Thanks for the detailed description and links.
I will go through all of them and try adapting them further.
Can you suggest me a good library for audio streaming as well?

starbasessd, 2022-11-12:
As I have stated in many times and many places, I am not a developer, or programmer, I don't even play one on TV. I do some old school scripting in bash, sh, dos, etc, and can sometimes follow the logic of Perl & Python, but that is about as far as it goes. If you look here and in the motionEye githubs, there have been many issues created about the lack of sound control, but most suggest going over to the motion-project and querying there...