motioneye-project/motioneyeos #2959: Fixed ip address on eth0 and wlan0

steinhorstbr, 2022-11-11:
Hello, I have a raspberry pi running motioneyeOs, and I'm having trouble putting the ips addresses, I'm not able to locate the config files in /etc/network/interfaces

I need different addresses in wlan0 and eth0
example
eth0
192.168.1.2/24

wlan0
192.168.5.2/24

Would you help me ?

starbasessd, 2022-11-11:
The easiest way is to have a static address handed out by the dhcp service on both the Ethernet network and WiFi network. 
In motionEyeOS, most of the system is Read-Only and requires a few hoops to be jumped through to edit files there. 
MotionEyeOS is not a pure Debian linux, but a BuildRoot/ThingOS IoT system meant to minimize the size.

steinhorstbr, 2022-11-11:
what complicates, is that this network does not have a dhcp server :(

starbasessd, 2022-11-11:
Could easily put a Pi3, Pi3B, Pi3B+, Pi4, or a Pi1 or Pi2 with a USB WiFi dongle on to provide DHCP services. 
motionEyeOS did not consider your use case (it normally only uses one or the other), and the static setup in the WebGUI does one or the other (looks to see if WiFi is active first and assigns the static setup to it, if not assigns the static to the Ethernet side), but not both.

steinhorstbr, 2022-11-11:
Got it, thanks a lot for the answers :)
taking advantage of the open topic, is it possible for this motioneye to transfer the recorded images to another one?

for example my motioneye does the recording during the day and at night it transfers to another one.

starbasessd, 2022-11-11:
Another option would be to install [Raspbian (Buster Legacy)](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) and motionEye on it. This would allow you to set both addresses locally as static...

starbasessd, 2022-11-11:
Not easily to change where it stores by time of day, but you can upload to a NAS as well as locally (duplicates all the time), or you could write a script to save a specific times worth of files (say at 7am copy all the files from 7p previous night to 7a to another location.) SFTP and SCP and SMB services are available...

steinhorstbr, 2022-11-11:
I will do this.
Thank you so much for the replies, your work is amazing.

starbasessd, 2022-11-11:
BTW, I only do support for motionEye/motionEyeOS, I didn't write any of it (except some of the documentation) I am not a developer...