motioneye-project/motioneyeos #2961: Access blocked: Third-party app is using an unsupported OAuth method Google has safe ways for users to sign in and share their Google Account data with third-party applications. To help protect your account, Google sign-in requests are blocked from apps with less secure authorization methods because they’re more vulnerable to phishing and app impersonation attacks.  Learn more about how your data is shared with third-party apps  App sent an invalid request If the app is blocked because it uses an unsupported authorization method, you may be directed to an “Error 400” page that says “Access blocked: App sent an invalid request

mouse2477, 2022-11-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

coder0107git, 2022-11-22:
Possibly related: #2945.

starbasessd, 2022-11-22:
Bug report already listed: #2954 #2945 and others.

starbasessd, 2022-11-22:
Also applies to motionEye bug reported as [#2625](https://github.com/motioneye-project/motioneye/issues/2625)

mouse2477, 2022-11-22:
looks like you guys have it coverd

mouse2477, 2022-12-06:
so guys what is going on with this fault  its been 14 day and still the same or is motioneyeos now not being updated with a fix 
or should i start looking for a new OS

mouse2477, 2022-12-06:
may have to do the same as you 
but google was the easy way to go as my works wifi is locked down and no 3g or 4g  my work station (nhs) and cant open the main app for the camera's
its a great shame its being left and sounds like its been put out to glass

xjon685, 2022-12-29:
Im using rclone to upload to s3 compatible bucket with idrivee2.com now.   They have 1TB for $4 for the first year , or $0.004 per GB.  works out cheaper than Google drive and I can pick a closer datacenter .   just 20ms ping.   So maybe this has a silver lining )

myscho15, 2023-04-01:
> idrivee2.com

> Im using rclone to upload to s3 compatible bucket with idrivee2.com now. They have 1TB for $4 for the first year , or $0.004 per GB. works out cheaper than Google drive and I can pick a closer datacenter . just 20ms ping. So maybe this has a silver lining )

Hallo, can you please send here how to make this work? I used google drive to monitor my cabin (4G router with rpi an motioneye os) and google drive on my phone... now when google drive dont work i use email but is not so good as drive.

Thank you for help