motioneye-project/motioneyeos #2963: How to record videos with a timer

ManuVanLinden, 2022-12-08:
I would like to record a 10 second video every 10 minutes with 4 different cameras, would it be possible without adding a script or do i need to write one myself ? Thanks

starbasessd, 2022-12-08:
That is not currently a feature available in movies. 
I would ask in the [motion](https://github.com/Motion-Project/motion) (not controlled by the same people as this) support group to see if anyone has written a similar script.
If you wanted to take a single snapshot with all 4 cameras, every 10 minutes, that IS a feature that exists in motionEye and motionEyeOS.

ManuVanLinden, 2023-01-10:
No i just want videos but it doesn't seem to exist...

starbasessd, 2023-01-10:
motionEye and motionEyeOS is a python script wrapper for the motion-project. The ability to do what you want doesn't exist within motionEye or motionEyeOS, and is not a trigger-able function in motion-project. Best option is to do a feature-request to [motion-project](https://github.com/Motion-Project/motion/wiki).

starbasessd, 2023-01-10:
That is one reason the motionEye/motionEyeOS is open source on github. Enjoy.

ManuVanLinden, 2023-01-10:
Thanks ! Do you know which file handles the movies options ?