motioneye-project/motioneyeos #2966: Choppy image with Artifacts with Higher resolution

Finalshare12, 2022-12-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

MotionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Ubuntu 22.04

### Board Model   

Bqeel Z83 

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: V4L2 ELP Webcam 1080p Infrarot nightvision Kamera USB
https://www.amazon.de/gp/product/B0194ZILNY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 

i had  a raspberry Pi 4 B   running but  it can only handle low quality resolution,  

i now switched to the Bqeel Z83 mini PC with ubuntu   but now everytime i switch to higher Resolution thatn    800x600  i get extremly choppy videos   with green and red Stripes trough it   also it dosnt react to the FPS limiter,  if i Set both cameras to 10 FPS

The Information window in the Web interface still shows around 29-33 FPS.

Tha Cameras Ran Stable on Raspberry  but the Raspi couldnt handle better resolution    but now on ubuntu it cant really handle it at all

starbasessd, 2022-12-18:
There are some limitations with USB and bandwidth for cameras. If you can make sure your cameras are on separate internal USB hubs, you may find it working better. Do you get the same issue only running a single camera?
I've never tried to run it on an Atom processor. I do not know how well supported it is.
I've had Ubuntu 22.04 have issues on Ryzen and Intel x64 CPUs. Ubuntu 20.04 worked without issues, can you check Ubuntu 20.04?
I run Debian Server Bullseye on my production system, do you have other OS options?

Finalshare12, 2022-12-18:
Hi, It also does it with okay 1 camera connected, and I was able to run 720p on the raspberry without problems. I can try Ubuntu 22.04I dont care which OS it runs, Ubuntu was just the easiest to install. I dont really know much about Linux --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.